CTSRD-CHERI/cheribsd #113: CHERIBSD refuses to try to start dynamically linked CHERIABI binaries

davidchisnall, 2016-02-01:
I'm investigating `rtld` support for CHERIABI, but CHERIBSD refuses to run binaries that are dynamically linked.  I tried this patch, but it does not do what I wanted (it breaks all dynamically linked binaries):

```diff
--- a/sys/mips/cheri/cheriabi_machdep.c
+++ b/sys/mips/cheri/cheriabi_machdep.c
@@ -128,7 +128,7 @@ static Elf64_Brandinfo freebsd_cheriabi_brand_info = {
        .interp_path    = "/libexec/ld-elf.so.1",
        .sysvec         = &elf_freebsd_cheriabi_sysvec,
        .interp_newpath = NULL,
-       .flags          = 0
+       .flags          = BI_CAN_EXEC_DYN | BI_BRAND_NOTE
 };
 
 SYSINIT(cheriabi, SI_SUB_EXEC, SI_ORDER_ANY,
```

brooksdavis, 2016-02-01:
I'm guessing that the introduction of `BI_BRAND_NOTE` is causing `elf64_get_brandinfo()` (http://fxr.watson.org/fxr/source/kern/imgact_elf.c#L253) to pick up CheriABI for mips64 binaries.  I don't think it should, but that's the best guess I've come up with (I'm currently trying to figure out how to create a dynamic binary so haven't tested yet).

On other thing that will need to change one way or another is the loader.  Either we'll want to link passing something other than `/libexec/ld-elf.so.1` (e.g. `/libexec/ld-elf-cheri.so.0`) and change `.interp_path` appropriately or we'll need to set `.interp_newpath` to point to it.  It's probably safe to assume that a native loader and a cheri loader will coexist for quite a while so just giving the cheri one a new name and always using it probably make sense.

davidchisnall, 2016-02-02:
Initially, I'm going to be invoking rtld directly - I imagine it will take a while to get it into a state where it makes sense to try to run things.

brooksdavis, 2016-05-20:
Works as expected.  Dynamic binaries run fine.  rtld is weird and not expected to run.  Here's the mips64 rtld failing:
```
root@beri1:~ # /libexec/ld-elf.so.1 
Cannot execute shared object
/libexec/ld-elf.so.1: Exec format error. Binary file not executable.
```

closed 2016-05-20 by brooksdavis