CTSRD-CHERI/cheribsd #115: libjpeg build contains Makefile directive that works improperly with -OX instead of -O

rwatson, 2016-02-13:
The libjpeg Makefile contains the following directives setting or transforming the optimisation level:
```
CFLAGS:=       ${CFLAGS:S/-O/-O2/}
CFLAGS.rdgif.c=         -O0
CFLAGS.rdswitch.c=      -O0
```

When used with an optimisation argument other than ```-O```, it will generate surprising arguments, such as ```-O20```. Currently, this triggers a Clang failure:
```
/home/rnw24/sdk/sdk/bin/clang -g -integrated-as --target=cheri-unknown-freebsd  -msoft-float --sysroot=/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp -mabi=sandbox -mxgot -cheri-linker  -O2 -pipe -I/home/rnw24/git/cheribsd/lib/libjpeg -DNO_GETENV  -G0  -O20 -Wno-error  -std=gnu99 -Wsystem-headers -Werror -Wall -Wno-format-y2k -W -Wno-unused-parameter -Wstrict-prototypes -Wmissing-prototypes -Wpointer-arith -Wreturn-type -Wcast-qual -Wwrite-strings -Wswitch -Wshadow -Wunused-parameter -Wcast-align -Wchar-subscripts -Winline -Wnested-externs -Wredundant-decls -Wold-style-definition -Wno-pointer-sign -Wmissing-variable-declarations -Wthread-safety -Wno-empty-body -Wno-string-plus-int -Wno-unused-const-variable  -Qunused-arguments -Wno-unused-parameter -Wno-cast-qual -c /home/rnw24/git/cheribsd/lib/libjpeg/../../contrib/jpeg/jdmarker.c -o jdmarker.o
warning: optimization level '-O20' is not supported; using '-O3' instead
clang-3.8: error: unable to execute command: Segmentation fault (core dumped)
clang-3.8: error: clang frontend command failed due to signal (use -v to see invocation)
clang version 3.8.0 (https://github.com/CTSRD-CHERI/clang 0a5302b5d63318045129da0e929f9114eba0264a) (http://github.com/CTSRD-CHERI/llvm 37973f7b9c8cc62bf4c06d4cfbb5c038c8d78618)
Target: cheri-unknown-freebsd
Thread model: posix
InstalledDir: /home/rnw24/sdk/sdk/bin
clang-3.8: note: diagnostic msg: PLEASE submit a bug report to http://llvm.org/bugs/ and include the crash backtrace, preprocessed source, and associated run script.
clang-3.8: note: diagnostic msg: 
********************

PLEASE ATTACH THE FOLLOWING FILES TO THE BUG REPORT:
Preprocessed source(s) and associated run script(s) are located at:
clang-3.8: note: diagnostic msg: /tmp/jdmarker-33eb7e.c
clang-3.8: note: diagnostic msg: /tmp/jdmarker-33eb7e.sh
clang-3.8: note: diagnostic msg: 

********************
*** Error code 254
```

rwatson, 2016-02-13:
This relates to CHERI Clang/LLVM bug https://github.com/CTSRD-CHERI/clang/issues/86

brooksdavis, 2016-02-14:
I've fixed the Makefile bug in 8ba9dc1bc45999bed6bc0162ba7606b9bb426c58.

closed 2016-02-14 by brooksdavis