CTSRD-CHERI/cheribsd #116: Missing mac(2) system calls prevent CheriABI ls(1) from compiling

rwatson, 2016-02-13:
As of https://github.com/CTSRD-CHERI/cheribsd/commit/d5a6a2b505ba41d27ebe87ab4dfe3621ec9d2e23, the ```__mac``` system calls are no longer exposed by CheriABI's ```libc```. As a result, a CheriABI ```/bin/ls``` will no longer compile:
```
index e57bf35..728ada4 100644
--- a/bin/ls/Makefile
+++ b/bin/ls/Makefile
@@ -3,6 +3,9 @@
 
 .include <src.opts.mk>
 
+WANT_CHERI=pure
+WANT_DUMP=yes
+
 PROG=  ls
 SRCS=  cmp.c ls.c print.c util.c
 LIBADD=        xo util
```

With the following failure mode:
```
===> bin/ls (all)
/home/rnw24/sdk/sdk/bin/clang -g -integrated-as --target=cheri-unknown-freebsd -msoft-float --sysroot=/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp -mabi=sandbox -mxgot -cheri-linker -O -pipe -G0 -DCOLORLS -O0 -Wno-error -g -std=gnu99 -Wsystem-headers -Werror -Wall -Wno-format-y2k -W -Wno-unused-parameter -Wstrict-prototypes -Wmissing-prototypes -Wpointer-arith -Wreturn-type -Wcast-qual -Wwrite-strings -Wswitch -Wshadow -Wunused-parameter -Wcast-align -Wchar-subscripts -Winline -Wnested-externs -Wredundant-decls -Wold-style-definition -Wno-pointer-sign -Wmissing-variable-declarations -Wthread-safety -Wno-empty-body -Wno-string-plus-int -Wno-unused-const-variable -Qunused-arguments  -Wl,--no-warn-mismatch -static -o ls.full cmp.o ls.o print.o util.o   -lxo  -lutil  -lutil  -ltermcapw   -lcompiler_rt  -lssp_nonshared  -lc -lmalloc_simple
/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp/usr/libcheri/libc.a(mac_get.o): In function `mac_get_fd':
/home/rnw24/git/cheribsd/lib/libc/posix1e/mac_get.c:43: undefined reference to `__mac_get_fd'
/home/rnw24/git/cheribsd/lib/libc/posix1e/mac_get.c:43: undefined reference to `__mac_get_fd'
/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp/usr/libcheri/libc.a(mac_get.o): In function `mac_get_file':
/home/rnw24/git/cheribsd/lib/libc/posix1e/mac_get.c:50: undefined reference to `__mac_get_file'
/home/rnw24/git/cheribsd/lib/libc/posix1e/mac_get.c:50: undefined reference to `__mac_get_file'
/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp/usr/libcheri/libc.a(mac_get.o): In function `mac_get_link':
/home/rnw24/git/cheribsd/lib/libc/posix1e/mac_get.c:57: undefined reference to `__mac_get_link'
/home/rnw24/git/cheribsd/lib/libc/posix1e/mac_get.c:57: undefined reference to `__mac_get_link'
/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp/usr/libcheri/libc.a(mac_get.o): In function `mac_get_pid':
/home/rnw24/git/cheribsd/lib/libc/posix1e/mac_get.c:73: undefined reference to `__mac_get_pid'
/home/rnw24/git/cheribsd/lib/libc/posix1e/mac_get.c:73: undefined reference to `__mac_get_pid'
/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp/usr/libcheri/libc.a(mac_get.o): In function `mac_get_proc':
/home/rnw24/git/cheribsd/lib/libc/posix1e/mac_get.c:80: undefined reference to `__mac_get_proc'
/home/rnw24/git/cheribsd/lib/libc/posix1e/mac_get.c:80: undefined reference to `__mac_get_proc'
clang-3.8: error: linker command failed with exit code 1 (use -v to see invocation)
*** Error code 1

Stop.
make[4]: stopped in /home/rnw24/git/cheribsd/bin/ls
*** Error code 1

Stop.
make[3]: stopped in /home/rnw24/git/cheribsd/bin
*** Error code 1

Stop.
make[2]: stopped in /home/rnw24/git/cheribsd
*** Error code 1

Stop.
make[1]: stopped in /home/rnw24/git/cheribsd
*** Error code 1

Stop.
make: stopped in /home/rnw24/git/cheribsd
```

These system calls will need to be implemented by CheriABI to allow userspace to build (and, presumably, for ```ls -lZ``` to work).

rwatson, 2016-02-14:
Also affects ```id(1)``` and ```inetd(8)```.

brooksdavis, 2016-02-16:
Hmm, this presumably prevents `ls -lZ` from working correctly under `freebsd32` since the lack of an implement is copied from there.

commit 98e422b4 (2016-02-16): Declare more syscalls.

Add declerations for several NOSTD marked system calls.

Use the base implementation of seemingly compatible system calls that
freebsd32 mysteriously marked UNIMPL.

Add a stub MAC implementation.  It looks straightforward to implement
properly, but some refactoring needs to be done (that would also let
freebsd32 have an implementation).

Fixes #116.
closed 2016-02-16 by brooksdavis