CTSRD-CHERI/cheribsd #118: cheri-c-tests don't build for 128-bit capabilities

staceyson, 2016-02-22:
After adding the cheri-c-tests code to contrib/cheri-c-tests I get the following error when trying to build CheriBSD (buildworld) for 128-bit capabilities:

```
===> bin/cheriabitest (all)
/home/sson/src/llvm/Build/bin/clang -g -integrated-as --target=cheri-unknown-freebsd  -msoft-float --sysroot=/var/tmp/obj/mips.mips64/home/sson/src/cheribsd/tmp -mabi=sandbox -mxgot -cheri-linker -mllvm -cheri128 -O -pipe -DCHERI_C_TESTS  -I/home/sson/src/cheribsd/contrib/cheri-c-tests -DTEST_CUSTOM_FRAMEWORK -I/home/sson/src/cheribsd/bin/cheriabitest/../cheritest -G0   -O0 -Wno-error -Qunused-arguments -g -MD -MP -MF.depend.test_runtime.o -MTtest_runtime.o -std=gnu99 -Wsystem-headers -Wall -Wno-format-y2k -W -Wno-unused-parameter -Wstrict-prototypes -Wmissing-prototypes -Wpointer-arith -Wreturn-type -Wcast-qual -Wwrite-strings -Wswitch -Wshadow -Wunused-parameter -Wcast-align -Wchar-subscripts -Winline -Wnested-externs -Wredundant-decls -Wold-style-definition -Wno-pointer-sign -Wmissing-variable-declarations -Wthread-safety -Wno-empty-body -Wno-string-plus-int -Wno-unused-const-variable  -Qunused-arguments -c /home/sson/src/cheribsd/contrib/cheri-c-tests/test_runtime.c -o test_runtime.o
In file included from /home/sson/src/cheribsd/contrib/cheri-c-tests/test_runtime.c:40:
In file included from /home/sson/src/cheribsd/contrib/cheri-c-tests/cheri_c_test.h:64:
/home/sson/src/cheribsd/bin/cheriabitest/../cheritest/cheri_c_test_framework.h:7:9: warning: 
      'assert' macro redefined [-Wmacro-redefined]
#define assert(x)                                                       \
        ^
/var/tmp/obj/mips.mips64/home/sson/src/cheribsd/tmp/usr/include/assert.h:54:9: note: 
      previous definition is here
#define assert(e)       ((e) ? (void)0 : __assert(__func__, __FILE__, \
        ^
/home/sson/src/cheribsd/contrib/cheri-c-tests/test_runtime.c:43:27: warning: unused
      parameter 'capreg' [-Wunused-parameter]
static void handler(void *capreg, int cause)
                          ^
/home/sson/src/cheribsd/contrib/cheri-c-tests/test_runtime.c:43:39: warning: unused
      parameter 'cause' [-Wunused-parameter]
static void handler(void *capreg, int cause)
                                      ^
/home/sson/src/cheribsd/contrib/cheri-c-tests/test_runtime.c:159:2: error: static_assert
      failed "Layout of struct cheri_frame has changed!"
        _Static_assert(offsetof(struct cheri_frame, cf_pcc) == 27*32,
        ^              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/sson/src/cheribsd/contrib/cheri-c-tests/test_runtime.c:171:12: warning: comparison of
      integers of different signs: '__attribute__((address_space(200))) register_t' (aka
      '__attribute__((address_space(200))) long') and 'unsigned long' [-Wsign-compare]
        assert(pc < __builtin_memcap_length_get(getCHERIFrame(context)->cf_pcc));
               ~~ ^                             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/sson/src/cheribsd/bin/cheriabitest/../cheritest/cheri_c_test_framework.h:9:8: note: 
      expanded from macro 'assert'
        if (!(x))                                                       \
              ^
/home/sson/src/cheribsd/contrib/cheri-c-tests/test_runtime.c:354:37: warning: unused
      parameter 'info' [-Wunused-parameter]
capsighandler(int signo, siginfo_t *info, ucontext_t *uap)
                                    ^
/home/sson/src/cheribsd/contrib/cheri-c-tests/test_runtime.c:385:6: warning: no previous
      prototype for function 'test_setup' [-Wmissing-prototypes]
void test_setup(void)
     ^
6 warnings and 1 error generated.
*** Error code 1
```

Building work 256-bit capabilities works fine.

davidchisnall, 2016-02-22:
The error should be fixed as of CTSRD-CHERI/cheri-c-tests@32324dbbfe5fa0433f32e4d10d51b69d3567a294.

It would help if cheri-c-test issues could be filed in the cheri-c-test issue tracker.

The warnings appear to be being caused by @brooksdavis' changes to integrate the C tests with cheritest.

davidchisnall, 2016-02-22:
It looks as if this case is only hit when you define `TEST_CUSTOM_FRAMEWORK`, which should not be defined if building cheri-c-tests as stand-alone binaries, but only when building them as part of a larger test program.

staceyson, 2016-02-22:
Confirmed.  Fixed with CTSRD-CHERI/cheri-c-tests@32324dbbfe5fa0433f32e4d10d51b69d3567a294.

closed 2016-02-22 by staceyson