CTSRD-CHERI/cheribsd #119: cheriabi_sigaltstack(9) may not adequately align stack to hold capabilities

rwatson, 2016-02-24:
The function ```cheriabi_sendsig(9)``` attempts to store capabilities in the kernel stack, but it looks like the structure in the stack intended to contain the capability is insufficiently aligned in some cases:
```
static void
cheriabi_sendsig(sig_t catcher, ksiginfo_t *ksi, sigset_t *mask)
{
        struct proc *p;
        struct thread *td;
        struct trapframe *regs;
        struct cheri_frame *capreg;
        struct sigacts *psp;
        struct sigframe_c sf, *sfp;
        uintptr_t stackbase;
        vm_offset_t sp;
        int cheri_is_sandboxed;
        int sig;
        int oonstack;
...
        sf.sf_uc.uc_mcontext.mc_pc = regs->pc;
        sf.sf_uc.uc_mcontext.mullo = regs->mullo;
        sf.sf_uc.uc_mcontext.mulhi = regs->mulhi;
        cheri_capability_copy(&sf.sf_uc.uc_mcontext.mc_tls,
            &td->td_md.md_tls_cap);
        sf.sf_uc.uc_mcontext.mc_regs[0] = UCONTEXT_MAGIC;  /* magic number */
...
```

Console output:
```
cpu:0-Trap cause = 4 (address error (load or I-fetch) - kernel mode)
[ thread pid 954 tid 100044 ]
Stopped at      cheri_capability_copy:  clc     c11, a1, 0(kdc)

db> bt
Tracing pid 954 tid 100044 td 0x980000000179e550
db_trace_thread+40 (?,?,?,?) ra ffffffff801602cc sp c00000001edaa1e0 sz 32
ffffffff80160140+18c (0,?,ffffffff,?) ra ffffffff8015f3a0 sp c00000001edaa200 sz 64
ffffffff8015ef18+488 (?,?,?,?) ra ffffffff8015f66c sp c00000001edaa240 sz 224
db_command_loop+f4 (?,?,?,?) ra ffffffff80163200 sp c00000001edaa320 sz 32
ffffffff80163058+1a8 (?,?,?,?) ra ffffffff80359760 sp c00000001edaa340 sz 832
kdb_trap+1f0 (?,?,?,?) ra ffffffff80599a5c sp c00000001edaa680 sz 64
trap+1a54 (?,?,?,?) ra ffffffff80581e98 sp c00000001edaa6c0 sz 288
MipsKernGenException+168 (1edaac20,179e9d0,8066e948,1edab240) ra ffffffff805b3bc0 sp c00000001edaa7e0 sz 448
cheri_capability_copy+0 (?,?,?,?) ra 0 sp c00000001edaa9a0 sz 0
pid 954
db> show registers
at                           0
v0                        0x23  _DYNAMIC_LINKING+0x22
v1                           0
a0          0xc00000001edaac20
a1          0x980000000179e9d0
a2          0xffffffff8066e948
a3          0xc00000001edab240
a4          0xc000000000a17ca8
a5                        0x10  _DYNAMIC_LINKING+0xf
a6                        0x30  _DYNAMIC_LINKING+0x2f
a7          0xc00000001edab0a1
t0          0xffffffff805eee68  hex2ascii_data
t1                           0
t2                           0
t3                           0
s0          0x980000000179e8c0
s1          0x980000000179e550
s2          0xc00000001edab460
s3                           0
s4                           0
s5          0xc000000000a17000
s6                        0x22  _DYNAMIC_LINKING+0x21
s7          0x9800000001a24a70
t8                         0x8  _DYNAMIC_LINKING+0x7
t9                         0x4  _DYNAMIC_LINKING+0x3
k0          0x980000003eefc160
k1                       0x201  _DYNAMIC_LINKING+0x200
gp          0xffffffff8078ae00  _gp
sp          0xc00000001edaa9a0
s8                0x7fff7ff000
ra          0xffffffff805b6a18  cheriabi_get_signal_stack_capability+0x230
sr                  0x4000fce3
lo                         0x1  _DYNAMIC_LINKING
hi                     0x2f027
bad         0x980000000179e9d0
cs                       0x110  _DYNAMIC_LINKING+0x10f
pc          0xffffffff805b3bc0  cheri_capability_copy
cheri_capability_copy:  clc     c11, a1, 0(kdc)
CHERI cause: ExcCode: 0x09 RegNum: $c02 (TLB prohibits store capability)
$c00: v:1 s:0 p:7fffffff b:0000000000000000 l:ffffffffffffffff o:0 t:0
$c01: v:1 s:0 p:7fff807d b:0000007fffffa480 l:0000000000000020 o:0 t:0
$c02: v:1 s:0 p:7fff807f b:00000001601e1000 l:0000000000001000 o:0 t:0
$c03: v:1 s:0 p:7fff807d b:0000007fffffa460 l:0000000000000004 o:0 t:0
$c04: v:1 s:0 p:7fff807d b:000000012017b590 l:0000000000000004 o:0 t:0
$c05: v:1 s:0 p:7fff807d b:0000007fffff9758 l:0000000000000004 o:0 t:0
$c06: v:1 s:0 p:7fff807d b:0000007fffffa060 l:0000000000000020 o:0 t:0
$c07: v:1 s:0 p:7fff807d b:0000007fffffa468 l:0000000000000018 o:17 t:0
$c08: v:1 s:0 p:7fff807d b:0000007fffffa0a0 l:0000000000000020 o:0 t:0
$c09: v:1 s:0 p:7fff807d b:0000007fffffa080 l:0000000000000020 o:0 t:0
$c10: v:1 s:0 p:7fff807d b:0000007fffffa060 l:0000000000000020 o:0 t:0
$c11: v:1 s:0 p:7fff8017 b:0000000000000000 l:0000010000000000 o:120022bec t:0
$c12: v:1 s:0 p:7fff807d b:0000007fffffa4a0 l:0000000000000020 o:0 t:0
$c13: v:1 s:0 p:00008055 b:0000007fffff9800 l:0000000000000008 o:0 t:0
$c14: v:1 s:0 p:7fff807d b:0000007fffffa020 l:0000000000000020 o:0 t:0
$c15: v:1 s:0 p:7fff807d b:0000007fffff9fa8 l:0000000000000078 o:0 t:0
$c16: v:1 s:0 p:7fff807d b:0000007fffff9fa4 l:0000000000000004 o:0 t:0
$c17: v:1 s:0 p:7fff8017 b:0000000000000000 l:0000010000000000 o:120022bd0 t:0
$c18: v:0 s:0 p:00000000 b:0000000000000000 l:ffffffffffffffff o:0 t:0
$c19: v:0 s:0 p:00000000 b:0000000000000000 l:ffffffffffffffff o:0 t:0
$c20: v:0 s:0 p:00000000 b:0000000000000000 l:ffffffffffffffff o:0 t:0
$c21: v:0 s:0 p:00000000 b:0000000000000000 l:ffffffffffffffff o:0 t:0
$c22: v:0 s:0 p:00000000 b:0000000000000000 l:ffffffffffffffff o:0 t:0
$c23: v:0 s:0 p:00000000 b:0000000000000000 l:ffffffffffffffff o:0 t:0
$c24: v:0 s:0 p:00000000 b:0000000000000000 l:ffffffffffffffff o:0 t:0
$c25: v:0 s:0 p:00000000 b:0000000000000000 l:ffffffffffffffff o:0 t:0
$c26: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:0 t:0
$c27: v:1 s:0 p:7fffffff b:0000000000000000 l:ffffffffffffffff o:0 t:0
$c28: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:0 t:0
$c29: v:1 s:0 p:7fffffff b:0000000000000000 l:ffffffffffffffff o:0 t:0
$c30: v:1 s:0 p:7fffffff b:0000000000000000 l:ffffffffffffffff o:0 t:0
$c31: v:1 s:0 p:7fffffff b:0000000000000000 l:ffffffffffffffff o:ffffffff801e6154 t:0
```

Disassembly around $pc (0xffffffff805b3bc0):
```
ffffffff805b3bc0 <cheri_capability_copy>:
ffffffff805b3bc0:       d97e2800        clc     c11,a1,0(c30)

static inline void
cheri_capability_store(u_int crn_from, struct chericap *cp)
{

        CHERI_CSC(crn_from, CHERI_CR_KDC, cp, 0);
ffffffff805b3bc4:       f97e2000        csc     c11,a0,0(c30)
cheri_capability_copy(struct chericap *cp_to, struct chericap *cp_from)
{

        cheri_capability_load(CHERI_CR_CTEMP0, cp_from);
        cheri_capability_store(CHERI_CR_CTEMP0, cp_to);
}
ffffffff805b3bc8:       03e00008        jr      ra
ffffffff805b3bcc:       00000000        nop
```

Disassembly around $ra (0xffffffff805b6a18):
```
ffffffff805b6a04:       de420110        ld      v0,272(s2)
ffffffff805b6a08:       ffa20160        sd      v0,352(sp)
        cheri_capability_copy(&sf.sf_uc.uc_mcontext.mc_tls,
ffffffff805b6a0c:       67a40280        daddiu  a0,sp,640
ffffffff805b6a10:       0c16cef0        jal     ffffffff805b3bc0 <cheri_capability_copy>
ffffffff805b6a14:       66250480        daddiu  a1,s1,1152 
            &td->td_md.md_tls_cap);
        sf.sf_uc.uc_mcontext.mc_regs[0] = UCONTEXT_MAGIC;  /* magic number */
ffffffff805b6a18:       3c025676        lui     v0,0x5676  
ffffffff805b6a1c:       3442dd6f        ori     v0,v0,0xdd6f
ffffffff805b6a20:       00021078        dsll    v0,v0,0x1  
ffffffff805b6a24:       ffa20050        sd      v0,80(sp)
```

It looks like ```td->td_md.md_tls_cap``` may be insufficiently aligned -- which is a bit surprising. (I expected, looking at the trace, that it would be the store to the stack that was insufficiently aligned... but the fault is on the load.) Indeed, $a1 (0x980000000179e9d0) is insufficiently aligned. What's odd is that this is the first time we've encountered this. Perhaps we're looking at a case where most of the time, this unaligned access is within a cache line, but today, on this run, for whatever reason, it wasn't?

brooksdavis, 2016-02-24:
It would be interesting to see how `td` and `td->td_md` are aligned.  They should be capability aligned owing to the embedded capability, but there could be a gcc bug.  The code that uses `md_tls_cap` is new as of last week.  If it's causing trouble, we could comment it back out for now

brooksdavis, 2016-02-28:
I've increased the alignment of lots of UMA zones include the `thread_zone` so this should be fixed. with 70f6f8b48060419c0714f44aa908bdffcc9dbcb2.

closed 2016-02-28 by brooksdavis