CTSRD-CHERI/cheribsd #122: libcheri should use better alignment heuristics for sealed capabilities on 128-bit CHERI

rwatson, 2016-03-09:
In changes e1cf72b0cfbb762f3cc9904014589b890816d6a8, 9cad20f444503d17234c1a9f92ecb707eaa4a958, and ac63f54c2ad2a5ab5bf4231bf58b1b619083ef8e, ```libcheri``` is adjusted to use better aligned lengths in order to ensure that bounds are representable. For code capabilities, a 4K roundup occurs, and for data capabilities, a 1M roundup. Really, this should be done with greater awareness of compression semantics, to minimise memory fragmentation (and ensure that suitable bounds can be used).

rwatson, 2016-03-12:
7ed51f1f4ff2ea4c7ba1cebc101d9dd6e26f3844 fixes a kernel bug in which userspace requests to memory map data to be sealed at superpage alignment were ignored. However, the alignment and rounding we've requested remains (likely) unnecessarily strong for data, and unnecessarily weak for code.

rwatson, 2016-04-04:
@brooksdavis may have resolved this, so assigning to him to confirm.

brooksdavis, 2016-04-04:
Fixed for data memory in 60293b647af9c4d4eb04c3bedc7e6e67b99f12f8, but code memory needs resolving and we should switch to using MAP_CHERI_NOSETBOUNDS for sub-region maps.

jrtc27, 2022-01-21:
libcheri was removed in #1056