CTSRD-CHERI/cheribsd #130: TLS access fails in first use for MIPS ABI

rwatson, 2016-04-19:
When run in the MIPS ABI on CheriBSD, the following test code (```tlstest.c```) will crash the first time it is run, yet succeed for future runs:
```
#include <err.h>
#include <pthread.h>
#include <stdio.h>
#include <sysexits.h>

static void *
do_nothing_thread(void *arg __unused)
{

        return (NULL);
}

int
main(int argc __unused, char *argv[] __unused)
{
        pthread_t thread;
        int ret;

        ret = pthread_create(&thread, NULL, do_nothing_thread, NULL);
        if (ret < 0)
                err(EX_OSERR, "pthread_create");

        ret = pthread_join(thread, NULL);
        if (ret < 0)
                err(EX_OSERR, "pthread_join");

        return (0);
}
```

Makefile:
```
# $FreeBSD$

PROG=   tlstest
SRCS=   tlstest.c

MAN=
WANT_DUMP=      yes
NO_SHARED=      yes

LIBADD= pthread

.include <bsd.prog.mk>
```

The first run sees this console output:
```
seroot@qemu-cheri:~ # tlstest
$c00: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:0 t:0
$c01: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:120099160 t:0
$c02: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c03: v:1 s:0 p:7fff807d b:000000016040d080 l:0000000000000050 o:0 t:0
$c04: v:1 s:0 p:7fff807d b:0000000120099160 l:0000000000000050 o:0 t:0
$c05: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c06: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c07: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c08: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c09: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c10: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c11: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:0 t:0
$c12: v:1 s:0 p:7fff8017 b:0000000000000000 l:0000010000000000 o:120003160 t:0
$c13: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c14: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c15: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c16: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c17: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c18: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c19: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c20: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c21: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c22: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c23: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c24: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c26: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:0 t:0
$c31: v:1 s:0 p:7fff8017 b:0000000000000000 l:0000010000000000 o:120008d0c t:0
Apr 19 12:17:16 qemu-cheri kernel: BAD_PAGE_FAULT: pid 546 tid 100047 (tlstest), uid 0: pc 0x120008d0c got a read fault (type 0x2) at 0xe4
Apr 19 12:17:16 qemu-cheri kernel: Trapframe Register Dump:
Apr 19 12:17:16 qemu-cheri kernel: zero: 0      at: 0   v0: 0x16040d070 v1: 0x200000
Apr 19 12:17:16 qemu-cheri kernel: a0: 0        a1: 0   a2: 0   a3: 0x7fffffff
Apr 19 12:17:16 qemu-cheri kernel: a4: 0x1      a5: 0   a6: 0   a6: 0
Apr 19 12:17:16 qemu-cheri kernel: t0: 0        t1: 0   t2: 0   t3: 0
Apr 19 12:17:16 qemu-cheri kernel: t8: 0        t9: 0x120008cc8 s0: 0   s1: 0
Apr 19 12:17:16 qemu-cheri kernel: s2: 0        s3: 0   s4: 0   s5: 0
Apr 19 12:17:16 qemu-cheri kernel: s6: 0        s7: 0   k0: 0   k1: 0
Apr 19 12:17:16 qemu-cheri kernel: gp: 0x1200a6c20      sp: 0x7ffbffdec0        s8: 0   ra: 0x120008ea0
Apr 19 12:17:16 qemu-cheri kernel: sr: 0x408084b3       mullo: 0        mulhi: 0        badvaddr: 0xe4
Apr 19 12:17:16 qemu-cheri kernel: cause: 0x8   pc: 0x120008d0c
Apr 19 12:17:16 qemu-cheri kernel: Page table info for bad address 0xe4: pde = 0, pte = 0
Segmentation fault (core dumped)
```

Further runs work fine:
```
root@qemu-cheri:~ # tlstest
root@qemu-cheri:~ # tlstest
root@qemu-cheri:~ # tlstest
root@qemu-cheri:~ # 
```

The crash appears to come around access to TLS data, which on this run appears to involve an undesirable NULL pointer dereference:
```
   120008d00:   0000802d        move    s0,zero
   120008d04:   7c02e83b        0x7c02e83b                  <- rdhwr   $2, $29
   120008d08:   dc508ff8        ld      s0,-28680(v0)
   120008d0c:   8e0200e4        lw      v0,228(s0)           <--- boom here
```

Also tagging @brooksdavis and @colinrothwell -- the latter in particular as he encountered this (or a very similar) bug when trying to run qemu on BERI. This reduced test case seems to do the trick adequately, with the problem arising when ```libpthread``` tries to access per-thread data via ```curthread``` (e.g., cancelation-point state when performing a sleep, etc). This makes me think that the hardware register might not always be getting set correctly...?

(This test also fails if the binary is dynamically linked.)

brooksdavis, 2016-04-19:
This is almost certainly something wrong with dc56e810bc7d9840928e16ad3bbbcc70eff517af and thus my fault, but if @staceyson has time to look at it that would be great since I'll be traveling for the next couple days.

staceyson, 2016-04-19:
@booksdavis Would dc56e810bc7d9840928e16ad3bbbcc70eff517af even be used for a MIPS ABI binary?

brooksdavis, 2016-04-19:
@staceyson I reworked the way the TLS offset is handled to hang the value off the thread in all cases and I fear I make have broken the initialization model for the userlocal register.  I thought I had it right, but this feels like a bug I encountered on the CheriABI side.

staceyson, 2016-04-19:
If libthr/libpthread is compiled with TLS_USE_SYSARCH defined I can't get it to fail.  This uses the sysarch() system call instead of the rdhwr instruction to get the TCB pointer in lib/libthr/arch/mips/include/pthread_md.h:

```
/*
 * Get the current tcb.
 */
#ifdef TLS_USE_SYSARCH
static __inline struct tcb *
_tcb_get(void)
{
        struct tcb *tcb;

        sysarch(MIPS_GET_TLS, &tcb);
        return tcb;
}

#else /* ! TLS_USE_SYSARCH */

#  if defined(__mips_n64)
static __inline struct tcb *
_tcb_get(void)
{
        uint64_t _rv;

        __asm__ __volatile__ (
            ".set\tpush\n\t"
            ".set\tmips64r2\n\t"
            "rdhwr\t%0, $29\n\t"
            ".set\tpop"
            : "=r" (_rv));
[...]
```

This makes me wonder if the UserLocal register needs to be initialized for a new process in http://fxr.watson.org/fxr/source/mips/mips/vm_machdep.c?im=3#L92

Also, saved and restored in get_mcontext() and set_mcontext():  http://fxr.watson.org/fxr/source/mips/mips/pm_machdep.c?im=3#L321

brooksdavis, 2016-04-19:
If seems odd that the state of the UserLocal register wouldn't follow fork.  If it doesn't then we definitely need to bring it along.

staceyson, 2016-04-20:
Yeah, after looking at it some more I don't believe that is the problem.  The UserLocal register gets restored in cpu_switch().

The offset in the dissembled code Robert included above doesn't look right.  That is:
```
120008d08:   dc508ff8        ld      s0,-28680(v0)
```

28680 = 0x7008

From sys/mips/include/tls.h
```
#define TLS_TP_OFFSET   0x7000
#define TLS_DTP_OFFSET  0x8000

#ifdef __mips_n64
#define TLS_TCB_SIZE    16
#ifdef COMPAT_FREEBSD32
#define TLS_TCB_SIZE32  8
#endif
#ifdef COMPAT_CHERIABI
#define TLS_TCB_SIZE_C  (2*sizeof(struct chericap))
#endif
#else
#define TLS_TCB_SIZE    8
#endif
```
Therefore, the offset should be TLS_TP_OFFSET + TLS_TCB_SIZE = 0x7000 + 0x10 = 0x7010 = 28688

staceyson, 2016-04-20:
The problem seems to be with this code change in sys/mips/mips/swtch.S:

```
   * remove_userlocal_code() in cpu.c.
   */
  	PTR_L	t0, TD_MDTLS(a1)		# Get TLS pointer
 -	PTR_ADDIU v0, t0, TLS_TCB_OFFSET	# Add TLS/TCB offset
 +	PTR_L	t1, TD_MDTLS_TCB_OFFSET(a1)	# Get TLS/TCB offset
 +	PTR_ADDU v0, t0, t1
  	MTC0	v0, MIPS_COP_0_USERLOCAL, 2	# write it to ULR for rdhwr
  
  	j	ra
```

Rolling back to adding a fixed offset seems to fix the problem.  Of course, this would break CheriABI binaries using thread local storage.

staceyson, 2016-04-20:
So the problem is that `td_md.md_tls_tcb_offset` field in the thread structure needs to be initialized to the offset value in cpu_thread_alloc().  The following change fixes the problem but maybe there is a better way that this should be done:

```
--- a/sys/mips/mips/vm_machdep.c
+++ b/sys/mips/mips/vm_machdep.c
@@ -328,6 +328,18 @@ cpu_thread_alloc(struct thread *td)
            td->td_kstack_pages * PAGE_SIZE) - 1;
        td->td_frame = &td->td_pcb->pcb_regs;
 
+#if defined(__mips_n64) && defined(COMPAT_FREEBSD32)
+       if (td->td_proc && SV_PROC_FLAG(td->td_proc, SV_ILP32)
+               td->td_md.md_tls_tcb_offset = TLS_TP_OFFSET + TLS_TCB_SIZE32;
+       else
+#endif
+#if defined (COMPAT_CHERIABI)
+       if (td->td_proc && SV_PROC_FLAG(td->td_proc, SV_CHERI))
+               td->td_md.md_tls_tcb_offset = TLS_TP_OFFSET + TLS_TCB_SIZE_C;
+       else
+#endif
+       td->td_md.md_tls_tcb_offset = TLS_TP_OFFSET + TLS_TCB_SIZE;
+
 #ifdef KSTACK_LARGE_PAGE
        /* Just one entry for one large kernel page. */
        pte = pmap_pte(kernel_pmap, td->td_kstack);
```

staceyson, 2016-04-20:
Fixed with d6a0191f75c0b46712dc2239423cf326d819627a

closed 2016-04-20 by staceyson