CTSRD-CHERI/cheribsd #131: We should provide a typedef for capability permissions

brooksdavis, 2016-04-25:
We should typedef `uint64_t` or `uint32_t` to something like `cheri_perm_t` and adjust code accordingly.

rwatson, 2016-04-25:
I'd lean towards 64-bit on the general principle that things always get bigger, but also because, architecturally, we pass a 64-bit register in and out of the corresponding permission-related instructions.

It would be nice for us to gradually generalise our cheri.h/cheric.h/cherireg.h files to make them suitable for reuse on other systems. The compiler-based distinction between "memory capabilities" and "CHERI capabilities" seems a bit moot where operating systems and applications are overtly aware of CHERI-specific semantics.

brooksdavis, 2016-04-25:
64-bit works for me.  I'm mostly tired of trying to decide how to spell it every time I write code using permissions.

davidchisnall, 2016-04-25:
`memcap.h` includes `memcap_perms_t`, which is currently a `short`.  I agree that it should be a 64-bit quantity.