CTSRD-CHERI/cheribsd #132: Kernel trap output sometimes prints the wrong pc

RichardsonAlex, 2016-05-09:
This was happening in `atomic_cmpset_32` due to a null dereference in the `cscw` instruction, but the pc of the previous `move` was being printed.

```
$c00: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:0 t:0
$c01: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:0 t:0
$c02: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c03: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:16044efc0 t:0
$c04: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c05: v:1 s:0 p:7fff8017 b:0000000000000000 l:0000010000000000 o:120003dd8 t:0
$c06: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:16041e420 t:0
$c07: v:1 s:0 p:7fff807d b:0000000120136860 l:0000000000000020 o:0 t:0
$c08: v:1 s:0 p:7fff807d b:0000000120136880 l:0000000000000004 o:0 t:0
$c09: v:1 s:0 p:7fff807d b:00000001201368a0 l:0000000000000020 o:0 t:0
$c10: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c11: v:1 s:0 p:7fff807d b:0000007fff7ff000 l:00000000007ff3c0 o:0 t:0
$c12: v:1 s:0 p:7fff8017 b:0000000000000000 l:0000010000000000 o:120020fe8 t:0
$c13: v:1 s:0 p:00008055 b:0000007fffffdb40 l:0000000000000028 o:0 t:0
$c14: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c15: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c16: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c17: v:1 s:0 p:7fff8017 b:0000000000000000 l:0000010000000000 o:120020fac t:0
$c18: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:16044efc0 t:0
$c19: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:16044efc0 t:0
$c20: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c21: v:1 s:0 p:7fff807d b:0000000120132440 l:0000000000000008 o:0 t:0
$c22: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c23: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c24: v:1 s:0 p:7fff8017 b:0000000000000000 l:0000010000000000 o:120003dd8 t:0
$c26: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:0 t:0
$c31: v:1 s:0 p:7fff8017 b:0000000000000000 l:0000010000000000 o:120021000 t:0
 0: pc 0x120021000 got a write fault (type 0x3) at 0
May  5 14:57:06 qemu-cheri-root kernel: Trapframe Register Dump:
May  5 14:57:06 qemu-cheri-root kernel: zero: 0 at: 0x120020fe8 v0: 0x186ca     v1: 0x12013a6c0
May  5 14:57:06 qemu-cheri-root kernel: a0: 0   a1: 0x186ca     a2: 0x7fe8d0    a3: 0x7fe9b8
May  5 14:57:06 qemu-cheri-root kernel: a4: 0   a5: 0xffffe00000000000  a6: 0x2d2d2d2820181008  a6: 0
May  5 14:57:06 qemu-cheri-root kernel: t0: 0   t1: 0   t2: 0   t3: 0
May  5 14:57:06 qemu-cheri-root kernel: t8: 0   t9: 0x120020f68 s0: 0   s1: 0x12011e890
May  5 14:57:06 qemu-cheri-root kernel: s2: 0x12011e890 s3: 0x1ba0      s4: 0   s5: 0
May  5 14:57:06 qemu-cheri-root kernel: s6: 0   s7: 0   k0: 0   k1: 0
May  5 14:57:06 qemu-cheri-root kernel: gp: 0x12011e890 sp: 0x7fe560    s8: 0x7fe560    ra: 0
May  5 14:57:06 qemu-cheri-root kernel: sr: 0x408084b3  mullo: 0x160    mulhi: 0        badvaddr: 0
May  5 14:57:06 qemu-cheri-root kernel: cause: 0xc      pc: 0x120021000
May  5 14:57:06 qemu-cheri-root kernel: Page table info for bad address 0: pde = 0, pte = 0
```

Here is a trace of the relevant instructions:
```
0x0000000120020ff4:  cllwu      v0,c3
    Write v0 = 0000000000000000
    Write LLAddr = 00000000004aaefc
0x0000000120020ff8:  bne        v0,a0,0x120021018
0x0000000120020ffc:  nop
0x0000000120021000:  move       v0,a1
    Write v0 = 00000000000186ca
0x0000000120021004:  cscw       v0,v0,c1
    Write Context = 0000000000000000
    Write BadVAddr = 0000000000000000
    Write EntryHi = 0000000000000097
    Write XContext = 0000000000000000
--- Exception #3: , vector ffffffff80000080
    Write C31|v:1 s:0 p:7fff8017 b:0000000000000000 l:0000010000000000
             |o:0000000120021000 t:0
    Write Status = 00000000408084b3
    Write Cause = 000000000000000c
    Write EPC = 0000000120021000
```

Seems like the PC is not incremented yet when the trap is handled, so the trap handler should take this into account and increment before printing.
Happens on QEMU, not sure if it also happens on FPGA.

rwatson, 2016-05-09:
Hmm. I'm not quite sure I understand the sequence. Could you paste in the objdump -dS output of the userspace code, +/- a few instructions around the faulting $pc?

rwatson, 2016-05-09:
(In general, however, if an instruction triggers an exception, $epc / $epcc should be set to the address of the instruction that triggered the exception, not the prior exception.)

RichardsonAlex, 2016-05-09:
I've recompiled so the pc is no longer the same. `0000000120021000` In the previous trace is now
`120025a00` (the `move	v0,a1`)
```
static __inline uint32_t
atomic_cmpset_32(__volatile uint32_t* p, uint32_t cmpval, uint32_t newval)
{
   1200259e8:	67bdffe0 	daddiu	sp,sp,-32
   1200259ec:	ebcbe8c3 	csd	s8,sp,24(c11)
   1200259f0:	03a0f025 	move	s8,sp
/build_output/cheribsd-obj-256/mips.mips64/mnt/sources/cheribsd/tmp/usr/include/machine/atomic.h:528
		"3:\n"
		: "=&r" (ret), "=m" (*p)
		: "r" (cmpval), "r" (newval), "m" (*p)
		: "memory");
#else
	__asm __volatile (
   1200259f4:	4a02180e 	cllw	v0,c3
   1200259f8:	14440007 	bne	v0,a0,120025a18 <atomic_cmpset_32+0x30>
   1200259fc:	00000000 	nop
   120025a00:	00a01025 	move	v0,a1
   120025a04:	4a021882 	cscw	v0,v0,c1
   120025a08:	1040fffa 	beqz	v0,1200259f4 <atomic_cmpset_32+0xc>
   120025a0c:	00000000 	nop
   120025a10:	08009687 	j	120025a1c <atomic_cmpset_32+0x34>
   120025a14:	00000000 	nop
   120025a18:	24020000 	li	v0,0
/build_output/cheribsd-obj-256/mips.mips64/mnt/sources/cheribsd/tmp/usr/include/machine/atomic.h:544
		: "=&r" (ret), "+C" (p)
		: "r" (cmpval), "r" (newval)
		: "memory");
#endif

	return ret;
   120025a1c:	03c0e825 	move	sp,s8
   120025a20:	cbcbe8c3 	cld	s8,sp,24(c11)
   120025a24:	49008800 	cjr	c17
   120025a28:	67bd0020 	daddiu	sp,sp,32
   120025a2c:	00000000 	nop
```

RichardsonAlex, 2016-05-09:
(This particular exception has been fixed in 4078b30d42fccb)

RichardsonAlex, 2016-05-11:
Looks like this could be CTSRD-CHERI/qemu#4

rwatson, 2016-05-27:
It would be useful to confirm whether this is resolved with more recent Qemu?

rwatson, 2017-03-19:
I'm assuming this is closed with recent Qemu; if not, please re-open!

closed 2017-03-19 by rwatson