CTSRD-CHERI/cheribsd #147: Reentrant and threading-safe system objects using libcheri

rwatson, 2016-05-27:
Currently, there is a single global stack used on entering CHERI system objects (```CHERI_ENTER_STACK_SIZE``` size for ```__cheri_enter_stack```). To properly handle reentrance of the system object, and multithreading, a better solution is required.

davidchisnall, 2016-07-19:
We're not currently using this for any of the eval, but it's a bit of an omission (other sandboxing things can do it quite happily).  Currently, the JVM manages clearing stacks and it would be good to move that responsibility into libcheri, so it manages a pool of stacks and clears them on return (ideally in a background thread).

jrtc27, 2022-01-21:
libcheri was removed in #1056

closed 2022-01-21 by jrtc27