CTSRD-CHERI/cheribsd #150: New threads have incorrect register sets

davidchisnall, 2016-06-01:
Simple test case:

```C
#include <pthread.h>

void *thr(void *a)
{
        return 0;
}

int main(void)
{
        pthread_t t;
        int a = 42;
        pthread_create(&t, NULL, thr, &a);
        void *val;
        pthread_join(t, &val);
}
```

Compile with:

```
$ sdk/bin/clang thread.c -lpthread
```

On creation of the first thread, there is an immediate fault because the registers are not set up correctly:

```
USER_CHERI_EXCEPTION: pid 458 tid 100058 (thread-mips), uid 2999: CP2 fault (type 0x32)
Trapframe Register Dump:
        zero: 0 at: 0   v0: 0   v1: 0
        a0: 0x160616400 a1: 0   a2: 0   a3: 0
        a4: 0   a5: 0   a6: 0   a6: 0
        t0: 0   t1: 0   t2: 0   t3: 0
        t8: 0   t9: 0x16004c7d0 s0: 0   s1: 0
        s2: 0   s3: 0   s4: 0   s5: 0
        s6: 0   s7: 0   k0: 0   k1: 0
        gp: 0   sp: 0x7ffbffdfe0        s8: 0   ra: 0
        sr: 0x408084b3  mullo: 0        mulhi: 0        badvaddr: 0x16004c7d0
        cause: 0x48     pc: 0x16004c7d0
CHERI cause: ExcCode: 0x02 RegNum: $pcc (tag violation)
$c00: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c01: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c02: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c03: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c04: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c05: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c06: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c07: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c08: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c09: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c10: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c11: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c12: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c13: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c14: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c15: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c16: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c17: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c18: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c19: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c20: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c21: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c22: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c23: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c24: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c26: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c31: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:16004c7d0 t:0
```

Neither PCC or DDC have valid tag bits.

rwatson, 2016-06-01:
Believed fixed in the MIPS ABI with 33309edf0114521da57cd6f3529cc7c015daf270, but a similar (but more capability-aware) change is also required in CheriABI's corresponding code.

rwatson, 2016-06-01:
It looks like the thread creation code in CheriABI sets up the necessary capability registers .. Except DDC. I wonder what it should set DDC to...?

brooksdavis, 2016-06-01:
The obvious first cut would be for DDC to be whatever it was for the caller of `pthread_create()`.  Off hand, I think that still holds even with a DDC per library model since it will need to be set up on function entry.

davidchisnall, 2016-06-02:
I believe that $c2 (the argument capability) is not being propagated correctly in CheriABI, causing breakage there.

I can confirm that 33309ed fixes this for the MIPS ABI.

davidchisnall, 2016-06-02:
Ideally for CheriABI we'd have a `pthread_create_np` that would specify the DDC as an extra pointer argument.

rwatson, 2016-06-02:
A CheriABI adaptation of the ```thread''' test program suggests that DDC is the first obstacle:
```
root@:~ # thread
CHERI cause: ExcCode: 0x02 RegNum: $c00 (tag violation)
$c00: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c01: v:1 s:0 p:7fff807d b:000000016043fe00 l:0000000000000e00 o:8 t:0
$c02: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c03: v:1 s:0 p:7fff807d b:000000016043fe00 l:0000000000000e00 o:0 t:0
$c04: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c05: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c06: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c07: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c08: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c09: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c10: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c11: v:1 s:0 p:7fff807f b:0000000160e00000 l:0000000000800000 o:0 t:0
$c12: v:1 s:0 p:7fff8017 b:0000000000000000 l:0000010000000000 o:120007ef0 t:0
$c13: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c14: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c15: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c16: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c17: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c18: v:1 s:0 p:7fff807d b:000000016043fe00 l:0000000000000e00 o:0 t:0
$c19: v:1 s:0 p:7fff807f b:00000001615fff50 l:0000000000000010 o:0 t:0
$c20: v:1 s:0 p:7fff807d b:000000016043fe00 l:0000000000000e00 o:8 t:0
$c21: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c22: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c23: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c24: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c26: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c31: v:1 s:0 p:7fff8017 b:0000000000000000 l:0000010000000000 o:12000810c t:0
Jun  1 20:51:09  kernel: USER_CHERI_EXCEPTION: pid 551 tid 100047 (thread), uid 0: CP2 fault (type 0x32)
Jun  1 20:51:09  kernel: Trapframe Register Dump:
Jun  1 20:51:09  kernel: zero: 0        at: 0x120109710 v0: 0x186cf     v1: 0
Jun  1 20:51:09  kernel: a0: 0x7fffffff a1: 0   a2: 0x186cf     a3: 0x1
Jun  1 20:51:09  kernel: a4: 0  a5: 0   a6: 0   a6: 0
Jun  1 20:51:09  kernel: t0: 0  t1: 0   t2: 0   t3: 0
Jun  1 20:51:09  kernel: t8: 0  t9: 0x120007ef0 s0: 0   s1: 0
Jun  1 20:51:09  kernel: s2: 0  s3: 0   s4: 0   s5: 0
Jun  1 20:51:09  kernel: s6: 0  s7: 0   k0: 0   k1: 0
Jun  1 20:51:09  kernel: gp: 0x120109710        sp: 0x7fff40    s8: 0x7fff40    ra: 0
Jun  1 20:51:09  kernel: sr: 0x408084b3 mullo: 0        mulhi: 0        badvaddr: 0x12000810c
Jun  1 20:51:09  kernel: cause: 0x48    pc: 0x12000810c
Signal 34 (core dumped)
```

It may also be that argument register $c3 (not $c2?) is not being propagated.

rwatson, 2016-06-02:
It looks like $c3 is explicitly propagated already. I assume $c2 in your comment above is a typo...?

davidchisnall, 2016-06-02:
Hmm, that's not what I see:

```
# USER_CHERI_EXCEPTION: pid 102 tid 100034 (thread-cheri), uid 2999: CP2 fault (type 0x32)
Trapframe Register Dump:
        zero: 0 at: 0   v0: 0x120123aa0 v1: 0
        a0: 0x1200f27c0 a1: 0   a2: 0   a3: 0
        a4: 0   a5: 0   a6: 0   a6: 0
        t0: 0   t1: 0   t2: 0   t3: 0
        t8: 0   t9: 0x120002f50 s0: 0   s1: 0x1
        s2: 0   s3: 0   s4: 0   s5: 0
        s6: 0   s7: 0   k0: 0   k1: 0
        gp: 0x1201097d0 sp: 0x7ff360    s8: 0x7ff360    ra: 0
        sr: 0x408084b3  mullo: 0        mulhi: 0        badvaddr: 0x120003060
        cause: 0x48     pc: 0x120003060
CHERI cause: ExcCode: 0x01 RegNum: $c02 (length violation)
$c00: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:0 t:0
$c01: v:1 s:0 p:7fff807d b:0000007fffffe700 l:0000000000000600 o:0 t:0
$c02: v:1 s:0 p:7fff807d b:0000000120123aa0 l:0000000000000000 o:0 t:0
$c03: v:1 s:0 p:7fff807d b:0000007fffffe480 l:00000000000000a0 o:0 t:0
$c04: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c05: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:0 t:0
$c06: v:1 s:0 p:7fff8017 b:0000000000000000 l:0000010000000000 o:120002fc4 t:0
$c07: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:1200f27c0 t:0
$c08: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:1200ba0e0 t:0
$c09: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c10: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c11: v:1 s:0 p:7fff807d b:0000007fff7ff000 l:00000000007ff480 o:0 t:0
$c12: v:1 s:0 p:7fff8017 b:0000000000000000 l:0000010000000000 o:120002f50 t:0
$c13: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c14: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c15: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c16: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c17: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c18: v:1 s:0 p:7fff807d b:0000007fffffe520 l:0000000000000040 o:0 t:0
$c19: v:1 s:0 p:7fff807d b:0000007fffffe560 l:00000000000001a0 o:0 t:0
$c20: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c21: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c22: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c23: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c24: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c26: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:0 t:0
$c31: v:1 s:0 p:7fff8017 b:0000000000000000 l:0000010000000000 o:120003060 t:0
pid 102 (thread-cheri), uid 2999: exited on signal 34
```

I guess that we're both seeing symptoms of something else.

rwatson, 2016-06-02:
I assume that your ```thread-cheri``` is statically linked...?

rwatson, 2016-06-02:
For CheriABI, it looks like we probably want to extend ```thr_param_c``` to include a DDC value that is NULL by default, and if NULL, then gets filled in as the caller DDC in ```cheriabi_thr_new_initthr()```. If the caller provides a DDC via the ```thr_param_c``` argument, it will be used, and otherwise we inherit from the parent thread.

davidchisnall, 2016-06-02:
Never mind, user error (I forgot to add `-cheri-linker`).  I need to make that the default...

michael-roe, 2016-06-02:
What if you want to create a thread that has a null DDC (e.g. It a pure
capability thread)? Null meanung inherit from creator may not be the best
api.

rwatson, 2016-06-02:
Agreed -- I think we must be able to express the policy that a thread is created from inception with a NULL DDC. What to do as the default might be something we want to evolve over time, though, since currently compiler-generated code continues to depend on DDC even in pure-capability mode.

rwatson, 2016-06-02:
I've committed with the NULL behaviour for now to get CheriABI threaded applications working again -- but we will want to adjust the pthread_create() API in the future to allow NULL DDC values to be installed.

closed 2016-06-21 by rwatson