CTSRD-CHERI/cheribsd #154: Investigate cheaper stack reset

davidchisnall, 2016-06-10:
Libcheri resets the stack by remapping the memory.  If only a small amount of the stack has been touched, it might be cheaper to `bzero` it.  We can use `mincore` to determine whether more than a page has been touched (though this will also need to acquire some VM locks, so might not be much faster).  We should run some benchmarks to determine when it's cheaper to `bzero` than to remap.

rmn30, 2016-06-10:
Worth reading up about  hardware features PICA had to aid with this:
http://ieeexplore.ieee.org/xpl/login.jsp?tp=&arnumber=546488&url=http%3A%2F%2Fieeexplore.ieee.org%2Fxpls%2Fabs_all.jsp%3Farnumber%3D546488

rmn30, 2016-06-10:
Actually maybe this is the paper I was thinking of:
http://ieeexplore.ieee.org/xpl/login.jsp?tp=&arnumber=634463&url=http%3A%2F%2Fieeexplore.ieee.org%2Fstamp%2Fstamp.jsp%3Ftp%3D%26arnumber%3D634463

davidchisnall, 2016-06-10:
If I'm reading that correctly, they have a 'zero cache line' instruction that is used for an efficient `bzero` for the stack.

rmn30, 2016-06-10:
Something like that, and a stack limit register which records the maximum
extent of stack (although I suppose this assumes hw support for push/pop).

davidchisnall, 2016-06-10:
It sounds difficult to implement a stack limit register.  It's entirely valid to take the address of a stack location, perform some arbitrary arithmetic, and then use the result in a load or store.  I can't think of an ISA that's capable of supporting C that doesn't have this property.  On CHERI, there is no guarantee that code (particularly malicious code) will do stack accesses via `$c11`, we derive capabilities to stack allocations by subsetting `$c11`.

That's not to say that we couldn't do the same thing in software, by having the kernel record the lowest page for each MAP_STACK allocation and providing a system call for accessing it.

davidchisnall, 2016-06-13:
It appears to be cheaper to bzero one page than to re-mmap it (this is surprisingly consistent across operating systems and architectures and even ways of measuring time), but above one page it is faster to bzero.  Part of this may be to do with the fact that we do not account the time spent by the kernel zeroing pages to the process, but on a multicore system this can typically be done by the idle loop anyway.

We might be able to improve latency by using a pool of stacks for an object that we zero in the background, though it wouldn't take much lock contention to push the cost of this above the cost of doing the mmap (which is very fast if the kernel has spare zero'd pages available).  A cheaper way of clearing the tag bits on a load of memory might still be nice though, for when we don't care too much about data left on stacks, only about tags left on stacks.

closed 2016-07-19 by davidchisnall