CTSRD-CHERI/cheribsd #155: Cheaper sandbox reset

davidchisnall, 2016-06-10:
We should be able to snapshot the state of a sandbox after it has been linked (required method variables have been reset and constructors have run) and then re-map it.  The easiest way to implement this might be to use anonymous shared memory objects and provide a `shm_clone` system call to create CoW snapshots of them.  This would have some extra overhead for the first sandbox, because we'd need to copy, rather than map, the data segment.  We'd then do the following sequence:

1. Create a shared memory object.
2. Copy the data segment into it.
3. Resolve all of the required methods and perform any run-time linking that does not create new capabilities.  At this point, we have a zygote sandbox.
4. Clone the shm object to create a new sandbox instance.
5. Run the global constructors.
6. Clone the object again.
7. On sandbox reset, simply re-clone and re-map.

davidchisnall, 2016-07-19:
This is not vital, but currently the costs for sandboxes that are reset on every invocation is quite high and it would be quite nice if we could bring this down.  In particular, if we can have before and after numbers then it would be a quite nice claim for the paper.

jrtc27, 2022-01-21:
libcheri was removed in #1056

closed 2022-01-21 by jrtc27