CTSRD-CHERI/cheribsd #157: memcpy crashes when the source doesn't have load-capability permissions

davidchisnall, 2016-06-17:
Our capability `memcpy` implementation performs `clc` and `csc` instructions in a loop.  This causes traps if we don't have the correct permissions, which means that the `memcpy` can't be used for pure-data copies.  I believe that we made the `clc` trap even if the data does not contain capabilities to avoid the need for data-dependent exceptions.  As a temporary work-around, we could branch to a version that used `cld` / `csd` instructions if we don't have load-capability on the source or store-capability on the destination.

The behaviour for `memcpy` should be to work as if we'd done a type-aware assignment of every field within an object, so if copying the data did not require that we have a permission that we don't have then it shouldn't matter.

This is currently causing a problem for the JNI work, as all buffers delegated to sandboxes have the load/store capability permissions stripped to avoid sandboxes leaking pointers into them.

rwatson, 2016-06-17:
I was pretty sure clc had the same tag-stripping behaviour our MMU bits have, for this reason? Normally, the memcpy() should trap only if the source had tags, and the capability through which we loaded permitted tag load, and we tried to store through a capability that doesn’t permit tagged stores, or via MMU pages that don’t permit tagged stores. A data-only memcpy() should simply need to clear the load-capability permission from the source capability, causing all tags to be stripped. If this is not the behaviour, this is surprising.

Robert

davidchisnall, 2016-06-17:
It's possible that this is a qemu bug.  I'll check the spec...

davidchisnall, 2016-06-17:
No, the pseudocode for `clc` contains this:
```
else if not cb.perms.Permit Load Capability then
    raise c2 exception(exceptionPermitLoadCapability, cb)
end if
```

So we are raising an exception if we don't have permission to read capabilities and we issue this instruction, even if the loaded value is not actually a capability.

rwatson, 2016-06-17:
On 17 Jun 2016, at 14:05, davidchisnall <notifications@github.com> wrote:
> No, the pseudocode for clc contains this:
>
> else if not cb.perms.Permit Load Capability then
>     raise c2 exception(exceptionPermitLoadCapability, cb)
> end if
> So we are raising an exception if we [...]
I think that I think this is a specification bug, given the above-indicated intention.

Robert

davidchisnall, 2016-07-19:
I have a work-around for this in the cherijni branch, which isn't hurting performance notably.

closed 2016-07-19 by davidchisnall