CTSRD-CHERI/cheribsd #159: FreeBSD/beri build broken out of CheriBSD due to memcpy_c

rwatson, 2016-07-03:
In attempting to build a FreeBSD/beri userspace to try to debug multicore boot issues on the L3 model, I encountered the following build failure:
```
===> lib/libc/tests/c063 (all)
(cd /home/rnw24/git/cheribsd/lib/libc/tests/c063 &&  DEPENDFILE=.depend.faccessat_test  NO_SUBDIR=1 make -f /home/rnw24/git/cheribsd/lib/libc/tests/c063/Makefile _RECURSING_PROGS=t  PROG=faccessat_test )
cc  -O -pipe -D_INCOMPLETE_XOPEN_C063 -G0   -I/home/rnw24/git/cheribsd/lib/libnetbsd -I/home/rnw24/git/cheribsd/contrib/netbsd-tests -g -MD -MP -MF.depend.faccessat_test.t_faccessat.o -MTt_faccessat.o -std=gnu99 -Wsystem-headers -Werror -Wall -Wno-format-y2k -Wno-uninitialized -Wno-pointer-sign    -c /home/rnw24/git/cheribsd/contrib/netbsd-tests/lib/libc/c063/t_faccessat.c -o t_faccessat.o
cc -O -pipe -D_INCOMPLETE_XOPEN_C063 -G0 -I/home/rnw24/git/cheribsd/lib/libnetbsd -I/home/rnw24/git/cheribsd/contrib/netbsd-tests -g -std=gnu99 -Wsystem-headers -Werror -Wall -Wno-format-y2k -Wno-uninitialized -Wno-pointer-sign  -Wl,--no-warn-mismatch -L/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/lib/libnetbsd -o faccessat_test.full t_faccessat.o  -lprivateatf-c -L/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/lib/libnetbsd -lnetbsd
/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp/lib/libc.so.7: undefined reference to `memcpy_c'
*** Error code 1

Stop.
make[7]: stopped in /home/rnw24/git/cheribsd/lib/libc/tests/c063
*** Error code 1
```

brooksdavis, 2016-07-05:
What is your build command?  It's building in jenkins and I've verified that tests are not disabled.

brooksdavis, 2016-07-05:
I'm getting a different failure (missing `__je*` symbols) in the same test with:
```
make  TARGET=mips TARGET_ARCH=mips64 buildworld
```

brooksdavis, 2016-07-05:
Rerunning at tip of tree, I don't see this at all.  The `_je*` symbol issue appears to have been due to an in-progress experiment.

rwatson, 2016-07-07:
I'm no longer seeing this, and am not sure why I was. Closing but will re-open if I I encounter this again.

closed 2016-07-07 by rwatson