CTSRD-CHERI/cheribsd #168: CheriABI should not support the obsolete mount(2) syscall

brooksdavis, 2016-11-07:
There is no simple way to sanity check the `data` argument to `mount()`.  The `nmount()` interface corrects this using an `iovec` to pass all non-flag arguments as strings.  The last consumers of mount in the base system (`df` and `tunefs`) should be fixed to use `nmount()` and the syscall should be removed.

trasz, 2017-09-08:
FWIW, df and tunefs no longer use mount(2); the only consumer left seems to be amd(8).

brooksdavis, 2017-09-08:
In an ideal world I think we'd make mount() a wrapper around nmount().  The thing I've not been able to figure out is which filesystems have a `data` element that needs to be translated (other than UFS).  I think we could even get away with removing the syscall with a two-major-revision delay.