CTSRD-CHERI/cheribsd #172: MIPS libraries are a mixture of MIPS-III and MIPS-IV

RichardsonAlex, 2016-12-07:
When attempting to use the latest binutils LD with elf64btsmip_cheri_fbsd emulation added ([RichardsonAlex/binutils-gdb](https://github.com/RichardsonAlex/binutils-gdb/tree/cheribsd)) the build will fail when it attempts to link libpng because the libpng object files are all MIPS-III
```
/home/alr48/cheri/build/cheribsd-obj-256/mips.mips64/home/alr48/cheri/sources/cheribsd/lib/libpng/png.o:              ELF 64-bit MSB relocatable, MIPS, MIPS-IV version 1 (FreeBSD), not stripped
/home/alr48/cheri/build/cheribsd-obj-256/mips.mips64/home/alr48/cheri/sources/cheribsd/lib/libpng/png.pico:           ELF 64-bit MSB relocatable, MIPS, MIPS-IV version 1 (FreeBSD), not stripped
/home/alr48/cheri/build/cheribsd-obj-256/mips.mips64/home/alr48/cheri/sources/cheribsd/lib/libpng/png.po:             ELF 64-bit MSB relocatable, MIPS, MIPS-IV version 1 (FreeBSD), not stripped
```
And libz.so is MIPS-III:
```
/home/alr48/cheri/build/cheribsd-obj-256/mips.mips64/home/alr48/cheri/sources/cheribsd/tmp/lib/libz.so.6: ELF 64-bit MSB shared object, MIPS, MIPS-III version 1 (FreeBSD), dynamically linked, not stripped
```

In a tree built with the CheriBSD ld we get libraries that are MIPS-III vs MIPS-IV:
```
/home/alr48/cheri/output/rootfs256/lib/libz.so.6: ELF 64-bit MSB shared object, MIPS, MIPS-III version 1 (FreeBSD), dynamically linked, not stripped
/home/alr48/cheri/output/rootfs256/usr/lib/libpng.so.15: ELF 64-bit MSB shared object, MIPS, MIPS-IV version 1 (FreeBSD), dynamically linked, not stripped
```

Should we attempt to make this consistent to allow linking with newer BFD or just wait until we can compile everything with clang?

davidchisnall, 2016-12-07:
Can you use readelf or similar to dump the full set of ELF headers for both?  We've linked MIPS III and MIPS IV files together previously without issues: it's expected to work even with an old MIPS linker and the result should be a MIPS IV binary.

RichardsonAlex, 2016-12-07:
This is only a problem when using newer binutils (2.27+git)

```
/home/alr48/cheri/output/sdk256/bin/cheri-unknown-freebsd-clang -shared -o /tmp/foo.so /home/alr48/cheri/build/cheribsd-obj-256/mips.mips64/home/alr48/cheri/sources/cheribsd/lib/libpng/*.pico /home/alr48/cheri/build/cheribsd-obj-256/mips.mips64/home/alr48/cheri/sources/cheribsd/tmp/lib/libz.so.6 -v -fuse-ld=bfd                                                                         
clang version 3.8.0 (https://github.com/CTSRD-CHERI/clang.git e66a25a5347f4739c87515f86495d98e1dd7b54b) (https://github.com/CTSRD-CHERI/llvm.git 6a7ed62359b6c4129499b8a9735fef977f71b6a1)
Target: cheri-unknown-freebsd
Thread model: posix
InstalledDir: /home/alr48/cheri/output/sdk256/bin
 "/home/alr48/cheri/output/sdk256/bin/cheri-unknown-freebsd-ld.bfd" --sysroot=/home/alr48/cheri/output/sdk256/sysroot --eh-frame-hdr -Bshareable --enable-new-dtags -o /tmp/foo.so /home/alr48/cheri/output/sdk256/sysroot/usr/lib/crti.o /home/alr48/cheri/output/sdk256/sysroot/usr/lib/crtbeginS.o -L/home/alr48/cheri/output/sdk256/sysroot/usr/lib /home/alr48/cheri/build/cheribsd-obj-256/mips.mips64/home/alr48/cheri/sources/cheribsd/lib/libpng/png.pico /home/alr48/cheri/build/cheribsd-obj-256/mips.mips64/home/alr48/cheri/sources/cheribsd/lib/libpng/pngerror.pico /home/alr48/cheri/build/cheribsd-obj-256/mips.mips64/home/alr48/cheri/sources/cheribsd/lib/libpng/pngget.pico /home/alr48/cheri/build/cheribsd-obj-256/mips.mips64/home/alr48/cheri/sources/cheribsd/lib/libpng/pngmem.pico /home/alr48/cheri/build/cheribsd-obj-256/mips.mips64/home/alr48/cheri/sources/cheribsd/lib/libpng/pngpread.pico /home/alr48/cheri/build/cheribsd-obj-256/mips.mips64/home/alr48/cheri/sources/cheribsd/lib/libpng/pngread.pico /home/alr48/cheri/build/cheribsd-obj-256/mips.mips64/home/alr48/cheri/sources/cheribsd/lib/libpng/pngrio.pico /home/alr48/cheri/build/cheribsd-obj-256/mips.mips64/home/alr48/cheri/sources/cheribsd/lib/libpng/pngrtran.pico /home/alr48/cheri/build/cheribsd-obj-256/mips.mips64/home/alr48/cheri/sources/cheribsd/lib/libpng/pngrutil.pico /home/alr48/cheri/build/cheribsd-obj-256/mips.mips64/home/alr48/cheri/sources/cheribsd/lib/libpng/pngset.pico /home/alr48/cheri/build/cheribsd-obj-256/mips.mips64/home/alr48/cheri/sources/cheribsd/lib/libpng/pngtrans.pico /home/alr48/cheri/build/cheribsd-obj-256/mips.mips64/home/alr48/cheri/sources/cheribsd/lib/libpng/pngwio.pico /home/alr48/cheri/build/cheribsd-obj-256/mips.mips64/home/alr48/cheri/sources/cheribsd/lib/libpng/pngwrite.pico /home/alr48/cheri/build/cheribsd-obj-256/mips.mips64/home/alr48/cheri/sources/cheribsd/lib/libpng/pngwtran.pico /home/alr48/cheri/build/cheribsd-obj-256/mips.mips64/home/alr48/cheri/sources/cheribsd/lib/libpng/pngwutil.pico /home/alr48/cheri/build/cheribsd-obj-256/mips.mips64/home/alr48/cheri/sources/cheribsd/tmp/lib/libz.so.6 -lgcc --as-needed -lgcc_s --no-as-needed -lc -lgcc --as-needed -lgcc_s --no-as-needed /home/alr48/cheri/output/sdk256/sysroot/usr/lib/crtendS.o /home/alr48/cheri/output/sdk256/sysroot/usr/lib/crtn.o
/home/alr48/cheri/build/cheribsd-obj-256/mips.mips64/home/alr48/cheri/sources/cheribsd/tmp/lib/libz.so.6: error adding symbols: File in wrong format
clang-3.8: error: linker command failed with exit code 1 (use -v to see invocation)
```

Whereas when using lld it works fine (with fine not meaning that the resulting library is actually usable):
```
/home/alr48/cheri/output/sdk256/bin/cheri-unknown-freebsd-clang -shared -o /tmp/foo.so /home/alr48/cheri/build/cheribsd-obj-256/mips.mips64/home/alr48/cheri/sources/cheribsd/lib/libpng/*.pico /home/alr48/cheri/build/cheribsd-obj-256/mips.mips64/home/alr48/cheri/sources/cheribsd/tmp/lib/libz.so.6 -v -fuse-ld=lld
clang version 3.8.0 (https://github.com/CTSRD-CHERI/clang.git e66a25a5347f4739c87515f86495d98e1dd7b54b) (https://github.com/CTSRD-CHERI/llvm.git 6a7ed62359b6c4129499b8a9735fef977f71b6a1)
Target: cheri-unknown-freebsd
Thread model: posix
InstalledDir: /home/alr48/cheri/output/sdk256/bin
 "/home/alr48/cheri/output/sdk256/bin/cheri-unknown-freebsd-ld.lld" --sysroot=/home/alr48/cheri/output/sdk256/sysroot --eh-frame-hdr -Bshareable --enable-new-dtags -o /tmp/foo.so /home/alr48/cheri/output/sdk256/sysroot/usr/lib/crti.o /home/alr48/cheri/output/sdk256/sysroot/usr/lib/crtbeginS.o -L/home/alr48/cheri/output/sdk256/sysroot/usr/lib /home/alr48/cheri/build/cheribsd-obj-256/mips.mips64/home/alr48/cheri/sources/cheribsd/lib/libpng/png.pico /home/alr48/cheri/build/cheribsd-obj-256/mips.mips64/home/alr48/cheri/sources/cheribsd/lib/libpng/pngerror.pico /home/alr48/cheri/build/cheribsd-obj-256/mips.mips64/home/alr48/cheri/sources/cheribsd/lib/libpng/pngget.pico /home/alr48/cheri/build/cheribsd-obj-256/mips.mips64/home/alr48/cheri/sources/cheribsd/lib/libpng/pngmem.pico /home/alr48/cheri/build/cheribsd-obj-256/mips.mips64/home/alr48/cheri/sources/cheribsd/lib/libpng/pngpread.pico /home/alr48/cheri/build/cheribsd-obj-256/mips.mips64/home/alr48/cheri/sources/cheribsd/lib/libpng/pngread.pico /home/alr48/cheri/build/cheribsd-obj-256/mips.mips64/home/alr48/cheri/sources/cheribsd/lib/libpng/pngrio.pico /home/alr48/cheri/build/cheribsd-obj-256/mips.mips64/home/alr48/cheri/sources/cheribsd/lib/libpng/pngrtran.pico /home/alr48/cheri/build/cheribsd-obj-256/mips.mips64/home/alr48/cheri/sources/cheribsd/lib/libpng/pngrutil.pico /home/alr48/cheri/build/cheribsd-obj-256/mips.mips64/home/alr48/cheri/sources/cheribsd/lib/libpng/pngset.pico /home/alr48/cheri/build/cheribsd-obj-256/mips.mips [... cut ...] /build/cheribsd-obj-256/mips.mips64/home/alr48/cheri/sources/cheribsd/tmp/lib/libz.so.6 -lgcc --as-needed -lgcc_s --no-as-needed -lc -lgcc --as-needed -lgcc_s --no-as-needed /home/alr48/cheri/output/sdk256/sysroot/usr/lib/crtendS.o /home/alr48/cheri/output/sdk256/sysroot/usr/lib/crtn.o
Relocating:.rela.dyn type=4, flags=2
Relocating:.rela.plt type=4, flags=2
Relocating:.rela.dyn type=4, flags=2
Relocating:.rela.plt type=4, flags=2
 "/home/alr48/cheri/output/sdk256/bin/capsizefix" /tmp/foo.so
```

davidchisnall, 2016-12-07:
Hmm, that seems to not include anything useful.  Please can you look at the ELF headers (e_flags field, in particular)?

RichardsonAlex, 2016-12-07:
```
cheri-freebsd-readelf -h /home/alr48/cheri/build/cheribsd-obj-256/mips.mips64/home/alr48/cheri/sources/cheribsd/tmp/lib/libz.so.6                                                                                                                           
ELF Header:
  Magic:   7f 45 4c 46 02 02 01 09 00 00 00 00 00 00 00 00 
  Class:                             ELF64
  Data:                              2's complement, big endian
  Version:                           1 (current)
  OS/ABI:                            UNIX - FreeBSD
  ABI Version:                       0
  Type:                              DYN (Shared object file)
  Machine:                           MIPS R3000
  Version:                           0x1
  Entry point address:               0x24f0
  Start of program headers:          64 (bytes into file)
  Start of section headers:          102520 (bytes into file)
  Flags:                             0x20000007, noreorder, pic, cpic, mips3
  Size of this header:               64 (bytes)
  Size of program headers:           56 (bytes)
  Number of program headers:         4
  Size of section headers:           64 (bytes)
  Number of section headers:         30
  Section header string table index: 27
```

```
cheri-freebsd-readelf -h /home/alr48/cheri/build/cheribsd-obj-256/mips.mips64/home/alr48/cheri/sources/cheribsd/lib/lib
png/png.pico 
ELF Header:
  Magic:   7f 45 4c 46 02 02 01 09 00 00 00 00 00 00 00 00 
  Class:                             ELF64
  Data:                              2's complement, big endian
  Version:                           1 (current)
  OS/ABI:                            UNIX - FreeBSD
  ABI Version:                       0
  Type:                              REL (Relocatable file)
  Machine:                           MIPS R3000
  Version:                           0x1
  Entry point address:               0x0
  Start of program headers:          0 (bytes into file)
  Start of section headers:          118272 (bytes into file)
  Flags:                             0x30000007, noreorder, pic, cpic, mips4
  Size of this header:               64 (bytes)
  Size of program headers:           0 (bytes)
  Number of program headers:         0
  Size of section headers:           64 (bytes)
  Number of section headers:         34
  Section header string table index: 1
```

I think the e_flags is the following line from the objdump output?
png: `private flags = 30000007: [abi=64] [mips4] [not 32bitmode] [noreorder] [PIC] [CPIC]`
libz: `private flags = 20000007: [abi=64] [mips3] [not 32bitmode] [noreorder] [PIC] [CPIC]`

davidchisnall, 2016-12-07:
This definitely looks like a new binutils bug.  Those files ought to be linkable.

michael-roe, 2016-12-11:
I think we're expected to encounter this case when linking .o files that
have been compiled with a mixture of gcc and clang/llvm, The CHERI MIPS
backend in LLVM is MIPS-IV, but gcc is building libraries for MIPS-III. It
clearly should work, as MIPS IV is backwards compatible with MIPS III.

Mike

RichardsonAlex, 2017-01-18:
Problem only affects the latest binutils which won't work properly with cheriBSD anyway so should not be a problem

closed 2017-01-18 by RichardsonAlex