CTSRD-CHERI/cheribsd #174: mergesort() is unable to sort ints

brooksdavis, 2016-12-07:
FreeBSD's `mergesort()` implementation requires that the size of the object being sorted by at least `sizeof(void *)/2`.  As `sizeof(void *)` is 16 or 32 bytes, this means that neither implementation can sort `int`s and CHERI256 can't sort `long`s.

A new implementation is likely required.

davidchisnall, 2016-12-07:
Does this mean that it's broken upstream if you try to sort a char array or a short array on 64-bit systems?

brooksdavis, 2016-12-07:
Yes.  This limitation is documented in the manpage, but is quite surprising and rather gratuitous.

I think there's an argument for reimplementing all our `*sort()` functions to generally modernize them.  If https://github.com/swenson/sort could emit a non-specialized version I'd be tempted to replace our current implementations with a dispatch function calling fast versions of primitive types and the generic one otherwise.  On "small" systems an option to only implement the slow version might make sense.

davidchisnall, 2016-12-07:
It would be interesting to see how libc++'s `std::sort` performs in comparison to some of these as well...

brooksdavis, 2016-12-07:
My current planed workaround/fix is to import https://github.com/patperry/timsort and use it to implement `mergesort` in CheriABI.

brooksdavis, 2016-12-14:
Fix in cheri by replacing with timsort.

closed 2016-12-14 by brooksdavis