CTSRD-CHERI/cheribsd #181: Conditionally included tests in cheritest

michael-roe, 2016-12-21:
In cheribsd/bin/cheritest/cheritest.c, some tests are conditionally compiled in with #ifdef __CHERI_PURE_CAPABILITY.

Unfortunately, cheritest is run both on the target (to run the test) and on the host (to find out which tests are expected to be in the test output), and if these two don't agree then the Jenkins job will report a test failure.

Something needs to be fixed here.
a) (svn repo)/cheribsd/trunk/simboot/test_boot.py could be a bit smarter about handling cases when it encounters a test it didn't expect.
b) We could add a command line argument to cheritest, that asks it to leave out the pure capability tests from the list of tests.

brooksdavis, 2017-02-08:
`test_boot.py` should be altered to not need to know what tests will be run and just care that the number is >0.

jrtc27, 2022-01-21:
The ability to use cheritest in this way was removed in a24dc8fa08fdb5c674164097c8ea11262e49e0f8

closed 2022-01-21 by jrtc27