CTSRD-CHERI/cheribsd #183: Dynamic linking broken

RichardsonAlex, 2017-01-11:
It seems that dynamically linked binaries no longer work with latest git HEAD. Not sure if it was the merge or is related to the linker script change.

When running /bin/helloworld with rtld debug enabled I get the following output:

```
root@qemu-cheri-alr48:~ # /bin/helloworld
/libexec/ld-cheri-elf.so.1 is initialized, base address = 0x161000000
RTLD dynamic = 0x1610026d8
RTLD pltgot  = 0x1610b5a00
initializing thread locks
_rtld_thread_init: done
processing main program's program header
note osrel 1200019
note crt_no_init
AT_EXECPATH 0x7fffffee90 /bin/helloworld
obj_main path /bin/helloworld
Ignoring d_tag 21 = 0x15
Ignoring d_tag 1879048193 = 0x70000001
Ignoring d_tag 1879048197 = 0x70000005
Ignoring d_tag 1879048198 = 0x70000006
Ignoring d_tag 1879048210 = 0x70000012
/bin/helloworld valid_hash_sysv 1 valid_hash_gnu 0 dynsymcount 41
lm_init("(null)")
loading LD_PRELOAD libraries
loading needed objects
 Searching for "libgcc_s.so.1"
search_library_pathfds('libgcc_s.so.1', '(null)', fdp)
lm_find("(null)", "/libcheri")
lmp_find("$DEFAULT$")
  Trying "/libcheri/libgcc_s.so.1"
lm_find("(null)", "/usr/libcheri")
lmp_find("$DEFAULT$")
  Trying "/usr/libcheri/libgcc_s.so.1"
loading "/usr/libcheri/libgcc_s.so.1"
Ignoring d_tag 1879048193 = 0x70000001
Ignoring d_tag 1879048197 = 0x70000005
Ignoring d_tag 1879048198 = 0x70000006
Ignoring d_tag 1879048210 = 0x70000012
/usr/libcheri/libgcc_s.so.1 valid_hash_sysv 1 valid_hash_gnu 0 dynsymcount 20
  0xc000 .. 0x1cfff: /usr/libcheri/libgcc_s.so.1
 Searching for "libc.so.7"
search_library_pathfds('libc.so.7', '(null)', fdp)
lm_find("(null)", "/libcheri")
lmp_find("$DEFAULT$")
  Trying "/libcheri/libc.so.7"
lm_find("(null)", "/usr/libcheri")
lmp_find("$DEFAULT$")
  Trying "/usr/libcheri/libc.so.7"
loading "/usr/libcheri/libc.so.7"
Ignoring d_tag 1879048193 = 0x70000001
Ignoring d_tag 1879048197 = 0x70000005
Ignoring d_tag 1879048198 = 0x70000006
Ignoring d_tag 1879048210 = 0x70000012
/usr/libcheri/libc.so.7 valid_hash_sysv 1 valid_hash_gnu 0 dynsymcount 3106
  0x1d000 .. 0x347fff: /usr/libcheri/libc.so.7
  WARNING: /usr/libcheri/libc.so.7 has impure text
checking for required versions
initializing initial thread local storage offsets
relocating "/bin/helloworld"
got:0x120011598 for 17 entries adding 0
got:0x120011618 for 41 entries
relocating "/usr/libcheri/libgcc_s.so.1"
got:0x1c988 for 8 entries adding 0xc000
got:0x1c9c0 for 20 entries
looking for .size.__CTOR_END__ in /usr/libcheri/libgcc_s.so.1 and /bin/helloworld
found .size.__CTOR_END__ in /usr/libcheri/libgcc_s.so.1 and /bin/helloworld
looking for .size.__start___cap_relocs in /usr/libcheri/libgcc_s.so.1 and /bin/helloworld
looking for .size.__stop___cap_relocs in /usr/libcheri/libgcc_s.so.1 and /bin/helloworld
looking for .size._int in /usr/libcheri/libgcc_s.so.1 and /bin/helloworld
found .size._int in /usr/libcheri/libgcc_s.so.1 and /bin/helloworld
relocating "/usr/libcheri/libc.so.7"
got:0x315688 for 5077 entries adding 0x1d000
got:0x31f528 for 3106 entries
looking for .size.__pthread_cxa_finalize in /usr/libcheri/libc.so.7 and /libexec/ld-cheri-elf.so.1
looking for .size.environ in /usr/libcheri/libc.so.7 and /bin/helloworld
found .size.environ in /usr/libcheri/libc.so.7 and /bin/helloworld
0 -> 20
looking for .size.__progname in /usr/libcheri/libc.so.7 and /bin/helloworld
found .size.__progname in /usr/libcheri/libc.so.7 and /bin/helloworld
0 -> 20
looking for .size._Block_copy in /usr/libcheri/libc.so.7 and /bin/helloworld
doing copy relocations
initializing initial thread local storage
initializing key program variables
"__progname": *0x1200114c0 <-- 0x7fffffed05
"environ": *0x120011740 <-- 0x7fffffe4a0
"__elf_aux_vector": *0x347000 <-- 0x7fffffe700
resolving ifuncs
calling init function for /usr/libcheri/libc.so.7 at 0x71fd0
calling init function for /usr/libcheri/libgcc_s.so.1 at 0xc7a0
loading filtees
transferring control to program entry point = 0x120000c00
$c00: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:0 t:0
$c01: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:40 t:0
$c02: v:1 s:0 p:7fff807d b:00000001200114c0 l:0000000000000020 o:0 t:0
$c03: v:1 s:0 p:7fff807d b:0000007fffffdc58 l:0000000000000308 o:8 t:0
$c04: v:1 s:0 p:7fff807d b:0000007fffffdc58 l:0000000000000308 o:0 t:0
$c05: v:1 s:0 p:7fff807d b:0000000120011700 l:0000000000000020 o:0 t:0
$c06: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:120011740 t:0
$c07: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:1200114c0 t:0
$c08: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:1200013a0 t:0
$c09: v:1 s:0 p:7fff807d b:0000007fff7ff000 l:00000000007ff3c0 o:7fe258 t:0
$c10: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c11: v:1 s:0 p:7fff807d b:0000007fff7ff000 l:00000000007ff3c0 o:0 t:0
$c12: v:1 s:0 p:7ffe8017 b:0000000000000000 l:0000010000000000 o:70bf0 t:0
$c13: v:1 s:0 p:00008055 b:0000007fffffd500 l:0000000000000020 o:0 t:0
$c14: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c15: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c16: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c17: v:1 s:0 p:7ffe8017 b:0000000000000000 l:0000010000000000 o:161016554 t:0
$c18: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:40 t:0
$c19: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:5020 t:0
$c20: v:1 s:0 p:7fff807d b:0000007fffffdf60 l:0000000000000020 o:0 t:0
$c21: v:1 s:0 p:7fff807d b:00000001610a4980 l:0000000000000020 o:0 t:0
$c22: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:120011590 t:0
$c23: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c24: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c26: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:0 t:0
$c31: v:1 s:0 p:7ffe8017 b:0000000000000000 l:0000010000000000 o:71f6c t:0
Jan 11 16:52:05 qemu-cheri-alr48 kernel: ALIGNMENT_FIX_ERR: pid 592 tid 100040 (helloworld), uid 0: pc 0x71f6c got a read fault (type 0x4) at 0x1e0000001f
Jan 11 16:52:05 qemu-cheri-alr48 kernel: Trapframe Register Dump:
Jan 11 16:52:05 qemu-cheri-alr48 kernel: zero: 0        at: 0x26f6d80001d000    v0: 0x1e0000001f        v1: 0
Jan 11 16:52:05 qemu-cheri-alr48 kernel: a0: 0  a1: 0x7ff058    a2: 0x9 a3: 0
Jan 11 16:52:05 qemu-cheri-alr48 kernel: a4: 0  a5: 0   a6: 0   a6: 0
Jan 11 16:52:05 qemu-cheri-alr48 kernel: t0: 0x38560    t1: 0x161038560 t2: 0x20        t3: 0
Jan 11 16:52:05 qemu-cheri-alr48 kernel: t8: 0x22       t9: 0x70bf0     s0: 0x22        s1: 0x1
Jan 11 16:52:05 qemu-cheri-alr48 kernel: s2: 0  s3: 0   s4: 0   s5: 0
Jan 11 16:52:05 qemu-cheri-alr48 kernel: s6: 0  s7: 0   k0: 0   k1: 0
Jan 11 16:52:05 qemu-cheri-alr48 kernel: gp: 0x1610bd9f0        sp: 0x7feb20    s8: 0x7feb20    ra: 0x120001360
Jan 11 16:52:05 qemu-cheri-alr48 kernel: sr: 0x408084b3 mullo: 0        mulhi: 0x1      badvaddr: 0x1e0000001f
Jan 11 16:52:05 qemu-cheri-alr48 kernel: cause: 0x10    pc: 0x71f6c
Jan 11 16:52:05 qemu-cheri-alr48 kernel: Page table info for bad address 0x1e0000001f: pde = 0, pte = 0
Bus error (core dumped)
```

RichardsonAlex, 2017-01-11:
Reverting a05d36083186cf4f42f2a5ba0e028d45361f2cbf makes it work again

bsdjhb, 2017-01-11:
After some debugging I found that comment out 'sigsetjmp' "fixed" rtld.  I used instruction tracing and found a very odd bug.  sigsetjmp() is written in assembly and uses the second argument to decide whether to call setjmp() or _setjmp():

```
LEAF(sigsetjmp)
	PIC_PROLOGUE(sigsetjmp)

	bne	a1, zero, 1f			# do saving of signal mask?
	PIC_TAILCALL(_setjmp)

1:	PIC_TAILCALL(setjmp)
END(sigsetjmp)
```

However, after the `bne` instruction in the trace I noted that `cgetpcc` was invoked next.  Looking at PIC_TAILCALL for CHERI:

```
# define PIC_TAILCALL(l)			\
	dla		t0, 1f;		\
1:	cgetpcc		$c12;			\
	cgetoffset	t1, $c12;		\
	dsub		t0, t1, t0;		\
	csetoffset	$c12, $c12, t0;		\
	dla		t9, _C_LABEL(l);	\
	cincoffset	$c12, $c12, t9;		\
	cjr		$c12;
```

There is a nested '1' label inside that macro.  So the effect was that the `bne` that was supposed to jump to the tailcall of `setjmp` instead jumped to the nested '1' in the tailcall of `_setjmp` at which point it jumped to whatever garbage was in t0.  I'm testing a workaround of changing `PIC_TAILCALL` to use a local label of '100' instead of '1' (hopefully that is big enough to be unique), but that requires a new world (not just a new rtld) so will take a bit to test.

RichardsonAlex, 2017-01-11:
I also have some patches which will remove the use of the local label in PIC_TAILCALL by using pc-relative relocations. Until that works, changing the local label sounds like a good solution.

Although I feel like clang should actually be warning about duplicate labels here.

bsdjhb, 2017-01-12:
I think rtld is now fixed, (at least simple tests pass for me) after my last two fixes for sigsetjmp.

closed 2017-01-12 by bsdjhb