CTSRD-CHERI/cheribsd #197: cannot build disk image from METALOG unless -DWITHOUT_TESTS is passed

RichardsonAlex, 2017-03-16:
I get the following error:

```
makefs: ./usr/tests: missing directory in specification
makefs: failed at line 1063 of the specification
```

brooksdavis, 2017-03-16:
It builds in jenkins with tests so you'll need to provide a lot more information

RichardsonAlex, 2017-03-16:
My script runs the following commands:
```
cd /exports/users/alr48/sources/cheribsd && nice make CHERI=256 CHERI_CC=/home/alr48/cheri/output-debug/sdk256/bin/clang CHERI_CXX=/home/alr48/cheri/output-debug/sdk256/bin/clang++ -DDB_FROM_SRC -DNO_WERROR -DNO_CLEAN -DNO_ROOT -DWITHOUT_GDB KERNCONF=CHERI_MALTA64 -s DEBUG_FLAGS=-g CHERI_OPTFLAGS=-O2 -DWITH_CHERI_SHARED -DWITHOUT_CHERI_USE_LLD -DWITH_CHERI_PURE -j16 buildworld
cd /exports/users/alr48/sources/cheribsd && nice make CHERI=256 CHERI_CC=/home/alr48/cheri/output-debug/sdk256/bin/clang CHERI_CXX=/home/alr48/cheri/output-debug/sdk256/bin/clang++ -DDB_FROM_SRC -DNO_WERROR -DNO_CLEAN -DNO_ROOT -DWITHOUT_GDB KERNCONF=CHERI_MALTA64 -s DEBUG_FLAGS=-g CHERI_OPTFLAGS=-O2 -DWITH_CHERI_SHARED -DWITHOUT_CHERI_USE_LLD -DWITH_CHERI_PURE -j16 buildkernel
cd /exports/users/alr48/sources/cheribsd && nice make CHERI=256 CHERI_CC=/home/alr48/cheri/output-debug/sdk256/bin/clang CHERI_CXX=/home/alr48/cheri/output-debug/sdk256/bin/clang++ -DDB_FROM_SRC -DNO_WERROR -DNO_CLEAN -DNO_ROOT -DWITHOUT_GDB KERNCONF=CHERI_MALTA64 -s DEBUG_FLAGS=-g CHERI_OPTFLAGS=-O2 -DWITH_CHERI_SHARED -DWITHOUT_CHERI_USE_LLD -DWITH_CHERI_PURE DESTDIR=/exports/users/alr48/rootfs-debug-256 installkernel
cd /exports/users/alr48/sources/cheribsd && nice make CHERI=256 CHERI_CC=/home/alr48/cheri/output-debug/sdk256/bin/clang CHERI_CXX=/home/alr48/cheri/output-debug/sdk256/bin/clang++ -DDB_FROM_SRC -DNO_WERROR -DNO_CLEAN -DNO_ROOT -DWITHOUT_GDB KERNCONF=CHERI_MALTA64 -s DEBUG_FLAGS=-g CHERI_OPTFLAGS=-O2 -DWITH_CHERI_SHARED -DWITHOUT_CHERI_USE_LLD -DWITH_CHERI_PURE DESTDIR=/exports/users/alr48/rootfs-debug-256 installworld
cd /exports/users/alr48/sources/cheribsd && nice make CHERI=256 CHERI_CC=/home/alr48/cheri/output-debug/sdk256/bin/clang CHERI_CXX=/home/alr48/cheri/output-debug/sdk256/bin/clang++ -DDB_FROM_SRC -DNO_WERROR -DNO_CLEAN -DNO_ROOT -DWITHOUT_GDB KERNCONF=CHERI_MALTA64 -s DEBUG_FLAGS=-g CHERI_OPTFLAGS=-O2 -DWITH_CHERI_SHARED -DWITHOUT_CHERI_USE_LLD -DWITH_CHERI_PURE DESTDIR=/exports/users/alr48/rootfs-debug-256 distribution

makefs -b 30% -f 30% -R 128m -M 256m -B be -F /tmp/tmpj68_vvz_/METALOG -N /exports/users/alr48/sources/cheribsd/etc /home/alr48/cheri/output-debug/cheri256-disk.img /exports/users/alr48/rootfs-debug-256
```

And `/tmp/tmpj68_vvz_/METALOG` is /exports/users/alr48/rootfs-debug-256 with a few extra lines added for ssh keys

brooksdavis, 2017-03-16:
`makefs -F` is almost never useful.  Instead, you probably want:
```
cd /exports/users/alr48/rootfs-debug-256
makefs -b 30% -f 30% -R 128m -M 256m -B be -N /exports/users/alr48/sources/cheribsd/etc /home/alr48/cheri/output-debug/cheri256-disk.img /tmp/tmpj68_vvz_/METALOG
```
You may need to ensure that the added lines in the generated `METALOG` have `contents=` tags pointing to the source.

brooksdavis, 2017-03-16:
That being said, the order of the entries in `METALOG` is odd and I'll try to figure out why it is happening.

commit d68a2dc4 (2017-03-16): Add CheriABI test directories after base test directories.

This likely fixes #197.
closed 2017-03-16 by brooksdavis