CTSRD-CHERI/cheribsd #198: CheriABI csh(1) crashes with tag violation if running sub-process is terminated using ^C - longjmp(3) bug?

rwatson, 2017-03-19:
When building a CheriABI world, hitting ^C to terminate a running ```file(1)``` session leads to ```csh``` itself terminating with a tag violation in what appears to relate to ```longjmp(3)```:
```
root@qemu-cheri:/bin # file auxargs
auxargs: ELF 64-bit MSB executable, MIPS, MIPS-IV with CHERI-256 (CheriABI) version 1 (FreeBSD), statically linked, for FreeBSD 12.0 (1200024), FreeBSD-style, not stripped
root@qemu-cheri:/bin # file auxargs.dump
^C
CHERI cause: ExcCode: 0x02 RegNum: $c17 (tag violation)
$c00: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:0 t:0
$c01: v:1 s:0 p:7fff807d b:00000001202ee540 l:0000000000000520 o:0 t:0
$c02: v:1 s:0 p:7fff807d b:000000012035c040 l:0000000000000008 o:0 t:0
$c03: v:1 s:0 p:7fff807d b:00000001203330e8 l:0000000000000300 o:0 t:0
$c04: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c05: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:400000 t:0
$c06: v:1 s:0 p:7fff807d b:00000000004331d8 l:0000000000000008 o:4 t:0
$c07: v:1 s:0 p:7fff807d b:00000001202e8080 l:0000000000000200 o:0 t:0
$c08: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:1202e8fb0 t:0
$c09: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c10: v:1 s:0 p:7fff807d b:000000000053df80 l:0000000000000180 o:a0 t:0
$c11: v:0 s:0 p:7fff807d b:0000007fff7ff000 l:00000000007ff580 o:0 t:0
$c12: v:0 s:0 p:7ffe8017 b:0000000000000000 l:0000010000000000 o:120118130 t:0
$c13: v:0 s:0 p:00008055 b:0000007fffffd140 l:0000000000000020 o:0 t:0
$c14: v:0 s:0 p:7fff807d b:0000000120332b00 l:0000000000000008 o:0 t:0
$c15: v:0 s:0 p:7fff807d b:0000000120332b80 l:0000000000000008 o:0 t:0
$c16: v:0 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:120332a80 t:0
$c17: v:0 s:0 p:7ffe8017 b:0000000000000000 l:0000010000000000 o:12000d8ec t:0
$c18: v:0 s:0 p:7fff807d b:0000000120300d8c l:0000000000000004 o:0 t:0
$c19: v:0 s:0 p:7fff807d b:0000007fffffda1c l:0000000000000004 o:0 t:0
$c20: v:0 s:0 p:7fff807d b:00000001203330e8 l:0000000000000300 o:0 t:0
$c21: v:0 s:0 p:7fff807d b:00000001203435cc l:0000000000000004 o:0 t:0
$c22: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c23: v:0 s:0 p:7fff807d b:00000001203010c0 l:0000000000000020 o:0 t:0
$c24: v:0 s:0 p:7fff807d b:0000000000442200 l:00000000000000a0 o:0 t:0
$c26: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:0 t:0
$c31: v:1 s:0 p:7ffe8017 b:0000000000000000 l:0000010000000000 o:1200edd9c t:0
Mar 19 19:20:45 qemu-cheri kernel: USER_CHERI_EXCEPTION: pid 553 tid 100055 (csh), uid 0: CP2 fault (type 0x32)
Mar 19 19:20:45 qemu-cheri kernel: Trapframe Register Dump:
Mar 19 19:20:45 qemu-cheri kernel: zero: 0      at: 0x120118130 v0: 0x1 v1: 0
Mar 19 19:20:45 qemu-cheri kernel: a0: 0x1      a1: 0x4 a2: 0x4 a3: 0
Mar 19 19:20:45 qemu-cheri kernel: a4: 0x60     a5: 0x3 a6: 0x2e        a6: 0x2f
Mar 19 19:20:45 qemu-cheri kernel: t0: 0x1203331e0      t1: 0xf8        t2: 0x1203331e0 t3: 0
Mar 19 19:20:45 qemu-cheri kernel: t8: 0x31     t9: 0x120117bf8 s0: 0x1 s1: 0x1203009d0
Mar 19 19:20:45 qemu-cheri kernel: s2: 0        s3: 0x1203009d0 s4: 0x1203009d0 s5: 0x1203009d0
Mar 19 19:20:45 qemu-cheri kernel: s6: 0x1      s7: 0x120294f20 k0: 0   k1: 0
Mar 19 19:20:45 qemu-cheri kernel: gp: 0x1203009d0      sp: 0x7fe300    s8: 0x7fe300    ra: 0x1203009d0
Mar 19 19:20:45 qemu-cheri kernel: sr: 0x408084b3       mullo: 0        mulhi: 0        badvaddr: 0x1200edd9c
Mar 19 19:20:45 qemu-cheri kernel: cause: 0x48  pc: 0x1200edd9c

FreeBSD/mips (qemu-cheri) (ttyu0)

login:
```

This appears to be an attempt to jump to $c17 at the end of ```longjmp(3)```:
```
00000001200edca0 <longjmp>:
#define LONGJMP_FRAME_SIZE      (CALLFRAME_SIZ + _MIPS_SZCAP/8)

NESTED(longjmp, LONGJMP_FRAME_SIZE, ra)
        .protected _C_LABEL(longjmp)
        .mask   0xc0000000, (CALLFRAME_RA - CALLFRAME_SIZ)
        PTR_SUBU        sp, sp, LONGJMP_FRAME_SIZE      # allocate stack frame
   1200edca0:   67bdff80        daddiu  sp,sp,-128

        csd             ra, sp, CALLFRAME_RA($c11)
   1200edca4:   ebebe85b        csd     ra,sp,88(c11)
        csd             s8, sp, CALLFRAME_FP($c11)
   1200edca8:   ebcbe853        csd     s8,sp,80(c11)
        csd             a0, sp, CALLFRAME_S0($c11)      # save return value
   1200edcac:   e88be843        csd     a0,sp,64(c11)
        csc             $c17, sp, CALLFRAME_C17($c11)
[... 105 lines cut ...]
        clc             $c16, t1, (_JB_CHERI_C16 * _MIPS_SZCAP/8)($c3)
   1200edd74:   da03680a        clc     c16,t1,160(c3)
        clc             $c17, t1, (_JB_CHERI_C17 * _MIPS_SZCAP/8)($c3)
   1200edd78:   da23680c        clc     c17,t1,192(c3)
        clc             $c18, t1, (_JB_CHERI_C18 * _MIPS_SZCAP/8)($c3)
   1200edd7c:   da43680e        clc     c18,t1,224(c3)
        clc             $c19, t1, (_JB_CHERI_C19 * _MIPS_SZCAP/8)($c3)
   1200edd80:   da636810        clc     c19,t1,256(c3)
        clc             $c20, t1, (_JB_CHERI_C20 * _MIPS_SZCAP/8)($c3)
   1200edd84:   da836812        clc     c20,t1,288(c3)
        clc             $c21, t1, (_JB_CHERI_C21 * _MIPS_SZCAP/8)($c3)
   1200edd88:   daa36814        clc     c21,t1,320(c3)
        clc             $c22, t1, (_JB_CHERI_C22 * _MIPS_SZCAP/8)($c3)
   1200edd8c:   dac36816        clc     c22,t1,352(c3)
        clc             $c23, t1, (_JB_CHERI_C23 * _MIPS_SZCAP/8)($c3)
   1200edd90:   dae36818        clc     c23,t1,384(c3)
        clc             $c24, t1, (_JB_CHERI_C24 * _MIPS_SZCAP/8)($c3)
   1200edd94:   db03681a        clc     c24,t1,416(c3)

        move            v0, a0
   1200edd98:   00801025        move    v0,a0
        cjr             $c17
   1200edd9c:   49008800        cjr     c17                 <--- Boom here
        ...
```

trasz, 2017-09-08:
I can't reproduce it.  Does it still happen for you?

brooksdavis, 2017-12-13:
I'm still seeing this

jrtc27, 2018-05-03:
I suspect this is because the array inside `_jmp_buf` in `sys/mips/include/setjmp.h` has `long long` as its element type and not `uintcap_t` on CHERI, but tcsh copies `(sig)jmp_buf`s around by direct assignment: https://github.com/jrtc27/cheribsd/blob/mct-new/contrib/tcsh/sh.h#L695-L696

jrtc27, 2018-05-03:
Although the struct is large enough that `memcpy` is used (which in and of itself is not guaranteed), if the source and destination have different alignment, the `copy_caps` part of `memcpy` will never run, leaving all of the capabilities untagged in the destination. Note here that `$c3` (the jump buffer passed to `longjmp`) is `0x1203330e8`, which isn't 32-byte aligned. Of course, whilst the tag bits are dropped, when `longjmp` comes to align the pointer to find the caps, it'll also end up potentially pointing to a different place to where the caps were stored originally, though in this case since there's one global jump buffer that's the current one, old saved jump buffers may have capabilities in the wrong place, but when they get copied back into the global jump buffer, the capabilities will be back in the right place, so that's not an issue in this specific case.

commit 3dee3c07 (2018-05-08): Various setjmp/longjmp fixes (Closes: #198, #264)

Firstly, (sig)jmp_buf's contents is turned into a union on CHERI to
ensure the correct alignment, that tag bits are propagated when copied
around (needed for csh) and that capabilities are always at the right
aligned location within the buffer even when copied to a different
buffer.

In the process, the broken CHERI code for the hybrid versions was fixed
to use a0 for the incoming address rather than the garbage $c3, and the
purecap longjmp gained a missing DDC restore.
closed 2018-05-08 by brooksdavis