CTSRD-CHERI/cheribsd #199: CheriABI mdconfig(2) error message during boot

rwatson, 2017-03-19:
With a CheriABI world, ```mdconfig(8)``` prints an error message during boot:
```
Starting syslogd.
mdmfs: mdconfig (attach) exited with error code 1
Clearing /tmp (X related).
```

Pure-capability build of ```mdconfig(8)``` disabled in 26e18317ac9. In which case there is still a boot-time warning, likely relating to ```mount(8)```:
```
Starting syslogd.
mdmfs: mount exited with error code 1
Clearing /tmp (X related).
```

brooksdavis, 2017-03-20:
It looks like mdconfig is failing due to a lack of modules on my system:

```
root@beri1:~ # ktrace mdconfig -l
mdconfig: failed to load geom_md module: No such file or directory
root@beri1:~ # kdump -s
...
   786 mdconfig CALL  kldload(0x1200006aa)
   786 mdconfig NAMI  "/boot/kernel/linker.hints"
   786 mdconfig NAMI  "/boot/kernel/geom_md"
   786 mdconfig NAMI  "/boot/kernel/geom_md.ko"
   786 mdconfig NAMI  "/boot/modules/linker.hints"
   786 mdconfig NAMI  "/boot/modules/geom_md"
   786 mdconfig NAMI  "/boot/modules/geom_md.ko"
   786 mdconfig RET   kldload -1 errno 2 No such file or directory
...
```

brooksdavis, 2017-03-20:
My fs image has a CheriABI `mdconfig` so it looks like 26e18317ac9bd3f14a36ad1f9c0aba2b111797ac may not be the right workaround.

trasz, 2017-09-08:
This seems to work just fine without the workaround.  Should we perhaps revert it?

brooksdavis, 2019-02-08:
This seems to be resolved.

closed 2019-02-08 by brooksdavis