CTSRD-CHERI/cheribsd #200: CheriABI ping(8) crashes in nv-related code

rwatson, 2017-03-19:
Running a CheriABI ```ping(8)``` results in a crash:
```
root@qemu-cheri:~ # ping 127.0.0.1
CHERI cause: ExcCode: 0x02 RegNum: $c12 (tag violation)
$c00: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:0 t:0
$c01: v:1 s:0 p:7fff807d b:00000001201f5ac0 l:0000000000000520 o:0 t:0
$c02: v:1 s:0 p:7fff807d b:00000001202304e0 l:0000000000000008 o:0 t:0
$c03: v:1 s:0 p:7fff807d b:0000007ffffed15c l:0000000000000004 o:0 t:0
$c04: v:1 s:0 p:7fff807d b:0000000000445000 l:00000000000000a0 o:0 t:0
$c05: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:400000 t:0
$c06: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:41a2e0 t:0
$c07: v:1 s:0 p:7fff807d b:0000007ffffec77c l:0000000000000001 o:0 t:0
$c08: v:1 s:0 p:7fff807d b:0000000120227420 l:0000000000000004 o:0 t:0
$c09: v:1 s:0 p:7fff807d b:0000000120227440 l:0000000000000020 o:0 t:0
$c10: v:1 s:0 p:7fff807d b:0000007fffffdb3c l:0000000000000001 o:0 t:0
$c11: v:1 s:0 p:7fff807d b:0000007fff7ff000 l:00000000007ff3a0 o:0 t:0
$c12: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:1200107a8 t:0
$c13: v:1 s:0 p:00008055 b:0000007ffffecfa0 l:0000000000000020 o:0 t:0
$c14: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c15: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c16: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c17: v:1 s:0 p:7ffe8017 b:0000000000000000 l:0000010000000000 o:120011140 t:0
$c18: v:1 s:0 p:7fff807d b:0000007ffffed154 l:0000000000000008 o:0 t:0
$c19: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:1200107a8 t:0
$c20: v:1 s:0 p:7fff807d b:00000001201a9e7a l:0000000000000005 o:0 t:0
$c21: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c22: v:1 s:0 p:7fff807d b:00000001201a9dc8 l:0000000000000007 o:0 t:0
$c23: v:1 s:0 p:7fff807d b:00000001201aa0de l:0000000000000007 o:0 t:0
$c24: v:1 s:0 p:7fff807d b:0000000000445000 l:00000000000000a0 o:0 t:0
$c26: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:0 t:0
$c31: v:1 s:0 p:7ffe8017 b:0000000000000000 l:0000010000000000 o:120011148 t:0
Mar 19 19:55:08 qemu-cheri kernel: USER_CHERI_EXCEPTION: pid 560 tid 100056 (ping), uid 0: CP2 fault (type 0x32)
Mar 19 19:55:08 qemu-cheri kernel: Trapframe Register Dump:
Mar 19 19:55:08 qemu-cheri kernel: zero: 0      at: 0x1200e3470 v0: 0   v1: 0
Mar 19 19:55:08 qemu-cheri kernel: a0: 0x8      a1: 0x20000001  a2: 0   a3: 0
Mar 19 19:55:08 qemu-cheri kernel: a4: 0x1      a5: 0x30        a6: 0x30        a6: 0
Mar 19 19:55:08 qemu-cheri kernel: t0: 0        t1: 0   t2: 0   t3: 0
Mar 19 19:55:08 qemu-cheri kernel: t8: 0        t9: 0x1200550d8 s0: 0x7 s1: 0x120207ec0
Mar 19 19:55:08 qemu-cheri kernel: s2: 0x1200107a8      s3: 0x120207ec0 s4: 0x120207ec0 s5: 0x120207ec0
Mar 19 19:55:08 qemu-cheri kernel: s6: 0x120207ec0      s7: 0x120207ec0 k0: 0   k1: 0
Mar 19 19:55:08 qemu-cheri kernel: gp: 0x120207ec0      sp: 0x7ee0e0    s8: 0x7ee0e0    ra: 0xffffffffffffffff
Mar 19 19:55:08 qemu-cheri kernel: sr: 0x408084b3       mullo: 0x10     mulhi: 0        badvaddr: 0x120011148
Mar 19 19:55:08 qemu-cheri kernel: cause: 0x48  pc: 0x120011148
ping: unable to limit access to system.dns service: Socket is not connected
```

It appears to crash dereferencing an untagged function pointer that has likely been case through an integer, ```func```, in ```zygote_main``` due to it being returned by ```nvlist_get_number()```:
```
$c12: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:1200107a8 t:0
```

The code:
```
0000000120010d78 <zygote_main>:
 * descriptor of the sandbox and socket pair descriptor for communication
 * between sandbox and its owner.
 */
static void
zygote_main(int sock)
{
...
                func = (zygote_func_t *)(uintptr_t)nvlist_get_number(nvlin,
   120011110:   49b3ac81        csetoffset      c19,c21,s2
                        /* Failure. */
                        error = errno;
                        break;
                case 0:
                        /* Child. */
                        close(sock);
   120011114:   3c010000        lui     at,0x0
   120011118:   003c882d        daddu   s1,at,gp  
   12001111c:   de219320        ld      at,-27872(s1)
   120011120:   480c09ff        cgetpccsetoffset        c12,at
   120011124:   48f16000        cjalr   c12,c17
   120011128:   02002025        move    a0,s0
                        close(chanfd[0]);
   12001112c:   c8920006        clw     a0,zero,0(c18)
   120011130:   de219320        ld      at,-27872(s1)
   120011134:   480c09ff        cgetpccsetoffset        c12,at
   120011138:   48f16000        cjalr   c12,c17
   12001113c:   00000000        nop
                        func(chanfd[1]);
   120011140:   c892000e        clw     a0,zero,4(c18)
   120011144:   49ac9800        cmove   c12,c19
   120011148:   48f16000        cjalr   c12,c17
   12001114c:   00000000        nop
```

rwatson, 2017-03-19:
```traceroute(8)``` appears to suffer the same problem:
```
root@qemu-cheri:~ # traceroute 127.0.0.1
CHERI cause: ExcCode: 0x02 RegNum: $c12 (tag violation)
$c00: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:0 t:0
$c01: v:1 s:0 p:7fff807d b:00000001201f8460 l:0000000000000520 o:0 t:0
$c02: v:1 s:0 p:7fff807d b:0000000120228b00 l:0000000000000008 o:0 t:0
$c03: v:1 s:0 p:7fff807d b:0000007fffffd5dc l:0000000000000004 o:0 t:0
$c04: v:1 s:0 p:7fff807d b:000000000045b0a0 l:00000000000000a0 o:0 t:0
$c05: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:400000 t:0
$c06: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:41a2e0 t:0
$c07: v:1 s:0 p:7fff807d b:0000007fffffcbfc l:0000000000000001 o:0 t:0
$c08: v:1 s:0 p:7fff807d b:000000012022a0c0 l:0000000000000020 o:0 t:0
$c09: v:1 s:0 p:7fff807d b:000000012022a0e0 l:0000000000000008 o:0 t:0
$c10: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:1202028e0 t:0
$c11: v:1 s:0 p:7fff807d b:0000007fff7ff000 l:00000000007ff3a0 o:0 t:0
$c12: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:1200181c8 t:0
$c13: v:1 s:0 p:00008055 b:0000007fffffd420 l:0000000000000020 o:0 t:0
$c14: v:1 s:0 p:7fff807d b:00000001202279e0 l:0000000000000008 o:0 t:0
$c15: v:1 s:0 p:7fff807d b:0000000120227a60 l:0000000000000008 o:0 t:0
$c16: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:120227960 t:0
$c17: v:1 s:0 p:7ffe8017 b:0000000000000000 l:0000010000000000 o:120018b60 t:0
$c18: v:1 s:0 p:7fff807d b:0000007fffffd5d4 l:0000000000000008 o:0 t:0
$c19: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:1200181c8 t:0
$c20: v:1 s:0 p:7fff807d b:00000001201ad1bc l:0000000000000005 o:0 t:0
$c21: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c22: v:1 s:0 p:7fff807d b:00000001201ad10a l:0000000000000007 o:0 t:0
$c23: v:1 s:0 p:7fff807d b:00000001201ad420 l:0000000000000007 o:0 t:0
$c24: v:1 s:0 p:7fff807d b:000000000045b0a0 l:00000000000000a0 o:0 t:0
$c26: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:0 t:0
$c31: v:1 s:0 p:7ffe8017 b:0000000000000000 l:0000010000000000 o:120018b68 t:0
Mar 19 20:15:15 qemu-cheri kernel: USER_CHERI_EXCEPTION: pid 564 tid 100051 (traceroute), uid 0: CP2 fault (type 0x32)
Mar 19 20:15:15 qemu-cheri kernel: Trapframe Register Dump:
Mar 19 20:15:15 qemu-cheri kernel: zero: 0      at: 0x1200e7180 v0: 0   v1: 0
Mar 19 20:15:15 qemu-cheri kernel: a0: 0xb      a1: 0x20000001  a2: 0   a3: 0
Mar 19 20:15:15 qemu-cheri kernel: a4: 0x1      a5: 0x30        a6: 0x30        a6: 0
Mar 19 20:15:15 qemu-cheri kernel: t0: 0        t1: 0   t2: 0x400       t3: 0x120220000
Mar 19 20:15:15 qemu-cheri kernel: t8: 0        t9: 0x1200567f8 s0: 0xa s1: 0x12020a900
Mar 19 20:15:15 qemu-cheri kernel: s2: 0x1200181c8      s3: 0x12020a900 s4: 0x12020a900 s5: 0x12020a900
Mar 19 20:15:15 qemu-cheri kernel: s6: 0x12020a900      s7: 0x12020a900 k0: 0   k1: 0
Mar 19 20:15:15 qemu-cheri kernel: gp: 0x12020a900      sp: 0x7fe560    s8: 0x7fe560    ra: 0xffffffffffffffff
Mar 19 20:15:15 qemu-cheri kernel: sr: 0x408084b3       mullo: 0x10     mulhi: 0        badvaddr: 0x120018b68
Mar 19 20:15:15 qemu-cheri kernel: cause: 0x48  pc: 0x120018b68
traceroute: unable to limit access to system.dns service
```

In the same code:
```
                func = (zygote_func_t *)(uintptr_t)nvlist_get_number(nvlin,
   120018b30:   49b3ac81        csetoffset      c19,c21,s2
                        /* Failure. */
                        error = errno;
                        break;
                case 0:
                        /* Child. */
                        close(sock);
   120018b34:   3c010000        lui     at,0x0
   120018b38:   003c882d        daddu   s1,at,gp
   120018b3c:   de2192b8        ld      at,-27976(s1)
   120018b40:   480c09ff        cgetpccsetoffset        c12,at
   120018b44:   48f16000        cjalr   c12,c17
   120018b48:   02002025        move    a0,s0
                        close(chanfd[0]);
   120018b4c:   c8920006        clw     a0,zero,0(c18)
   120018b50:   de2192b8        ld      at,-27976(s1) 
   120018b54:   480c09ff        cgetpccsetoffset        c12,at
   120018b58:   48f16000        cjalr   c12,c17 
   120018b5c:   00000000        nop
                        func(chanfd[1]);
   120018b60:   c892000e        clw     a0,zero,4(c18)
   120018b64:   49ac9800        cmove   c12,c19
   120018b68:   48f16000        cjalr   c12,c17
   120018b6c:   00000000        nop
                        /* NOTREACHED */
                        exit(1);
   120018b70:   3c010000        lui     at,0x0  
   120018b74:   003c082d        daddu   at,at,gp
   120018b78:   dc219930        ld      at,-26320(at)
   120018b7c:   480c09ff        cgetpccsetoffset        c12,at
   120018b80:   48f16000        cjalr   c12,c17
   120018b84:   64040001        daddiu  a0,zero,1
        for (;;) {
                nvlin = nvlist_recv(sock, 0);   
                if (nvlin == NULL) {
                        if (errno == ENOTCONN) {
                                /* Casper exited. */
                                exit(0);
   120018b88:   3c010000        lui     at,0x0
```

rwatson, 2017-03-20:
It appears that new versions of `libcasper(3)` pass function pointers from a consumer to a zygote using integer nvlist types between `zygote_clone()` and  `zygote_main()`. `nvlist(3)` may need to grow support for pointer-type encoding.

rwatson, 2017-03-20:
Although `libcasper(3)` provides a general mechanism to remote call function pointers, it is currently used for precisely one function, suggesting a more CHERI-friendly route of providing a manual dispatch. This is also a parent-zygote-only ABI so may well be non-disruptive to change.

rwatson, 2017-03-20:
With change 9162b6e, I'm now running into:
```
root@qemu-cheri:~ # ping 127.0.0.1
Assertion failed: (service != NULL), function service_start, file /home/rnw24/git/cheribsd/lib/libcasper/libcasper/service.c, line 403.
ping: unable to limit access to system.dns service: Socket is not connected
```

This appears to be due to a similar issue relating to passing data pointers over IPC.

rwatson, 2017-03-20:
```ping(8)``` and ```traceroute(8)``` now appear to work in CheriABI. Leaving this issue open while we ponder whether these patches are appropriate to upstream to FreeBSD -- and if not these patches, then which patches.

brooksdavis, 2020-04-21:
This was upstreamed in r315862 (freebsd/freebsd@a78ff3f4f34f756dfc3e5dcf18361669af8a0ea0)

closed 2020-04-21 by brooksdavis