CTSRD-CHERI/cheribsd #201: CheriABI OpenSSH fails due to select returning EPROT

rwatson, 2017-03-20:
When running with a CheriABI OpenSSH client, I see the following:
```
root@qemu-cheri:~ # service sshd onestart
Generating RSA host key.
2048 SHA256:lpf4/Eg3v9nfTN3idt9dL1zGs1KqFt7K9Wdh4hjbAI4 root@qemu-cheri (RSA)
Generating ECDSA host key.
256 SHA256:983Wk2UogBcp5Obd25zI0WE7+PaphypbVqVWCEJq56Y root@qemu-cheri (ECDSA)
Generating ED25519 host key.
256 SHA256:wrROvx+PdRto7hvG89dLZnr0Gq1jZsElseLXfBse6t0 root@qemu-cheri (ED25519)
Performing sanity check on sshd configuration.
Starting sshd.
root@qemu-cheri:~ # slogin root@127.0.0.1
Fssh_ssh_dispatch_run_fatal: Connection to 127.0.0.1 port 22: Resource temporarily unavailable
```

Slightly confusingly, this is because ```select(2)``` is returning ```EPROT```, leading to an errant read (presumably due to a bug in OpenSSH) of an un-ready file descriptor:
```
   615 slogin   RET   write 1432/0x598
   615 slogin   CALL  select(0x4,0,0,0,0)
   615 slogin   RET   select -1 errno 97 Memory protection violation
   615 slogin   CALL  read(0x3,0x7fffffa37c,0x2000)
   615 slogin   RET   read -1 errno 35 Resource temporarily unavailable
   615 slogin   CALL  write(0x2,0x7fffffbf80,0x60)
   615 slogin   GIO   fd 2 wrote 96 bytes
       "Fssh_ssh_dispatch_run_fatal: Connection to 127.0.0.1 port 22: Resource\
         temporarily unavailable\r
       "
   615 slogin   RET   write 96/0x60
   615 slogin   CALL  exit(0xff)
```

brooksdavis, 2017-03-20:
Hmm, select is returning EPROT due to the `in` `fd_set` being shorter than it expects:
```
  607 slogin   UNKNOWN(18)  "cheriabi_cap_to_ptr: (length - offset) (8) < required (128)"
   607 slogin   UNKNOWN(18)  "CHERIABI_SYS_select_fill_uap: invalid 'in' pointer"
   607 slogin   CALL  select(0x4,0,0,0,0)
```
This appears to be due to OpenSSH allocating less than a full `struct fs_set`.  For example (from sshconnect.c):
```C
        fdset = xcalloc(howmany(sockfd + 1, NFDBITS),
            sizeof(fd_mask));
```
Is this pattern common?  It seems to relay on internals of `struct fd_set` which are not POSIX defined.

cc: @bsdjhb

brooksdavis, 2017-03-20:
Ah, it is correct(ish) practice since the application could override `FD_SETSIZE`.   This will require two more manual argument checkers since I can't express the mess that is the size the `fd_set` storage needs to be in SAL.

closed 2017-03-20 by brooksdavis

brooksdavis, 2017-03-20:
The fact that the `select` API provides no way to allocate an `fd_set` for a given set of descriptors presumably dates from a time when the set was a fixed size.  The fact that this hasn't been corrected and that non-POSIX macros are types are used in notionally portable code like OpenSSH is a problem.  The API should be revisited.

bsdjhb, 2017-03-21:
The API was revisited and it's now spelled poll(). :-/  No one is going to change select()'s API at this point.  select() generally assumes you are working with fixed-sizes at compile time as you can only override FD_SETSIZE at compile time, not at runtime.

commit 16e1307b (2017-03-20): Check the size of fd_set argument correctly.

struct fd_set contains an array of fd_mask objects (longs in practice),
sufficent to hold a bit for each descriptor.  By default, struct fd_set
holds 1024 bits on FreeBSD, but the user may define FD_SETSIZE to
another size, or may manually allocate a sufficent buffer.  OpenSSH uses
the latter pattern which broke CheriABI.

This interface is rather dubious C.  The failure to use an array or (if
C had them) a bitvector is made worse by there not being a single macro
defined in POSIX that hides implementation details like fd_mask.

This fixes #201