CTSRD-CHERI/cheribsd #202: objdump(1) can't decode rdhwr

trasz, 2017-03-21:
It looks like the objdump(1) utility can't disassemble the rdhwr opcode.  The result looks like this:

```
        if (unlikely(tsd->state != tsd_state_nominal)) {
   121235590:   3c010000        lui     at,0x0
   121235594:   7c03e83b        0x7c03e83b
   121235598:   64259000        daddiu  a1,at,-28672
   12123559c:   0065082d        daddu   at,v1,a1
   1212355a0:   8c230000        lw      v1,0(at)
   1212355a4:   10600011        beqz    v1,1212355ec <__je_malloc_tsd_boot0+0xe4>
   1212355a8:   00000000        nop

```

Looking at contrib/binutils/opcodes/mips-opc.c, it looks like it should, though:

`{"rdhwr",   "t,K",      0x7c00003b, 0xffe007ff, WR_t,                   0,              I33     },
`

davidchisnall, 2017-03-21:
`rdhwr` is difficult, because it's not technically a valid opcode on the processor model that we support (it's MIPS32r2).  The compiler will insert some push and pop directives around it when generating assembly.  In the LLVM disassembler, there's some special logic for this instruction to handle the fact that it's useable on targets that it shouldn't be.  You can see this if you use llvm-objdump.

trasz, 2017-03-22:
I didn't do the reading yet, so I don't quite understand the problem you're describing, but a trivial patch allows objdump to disassemble it, and the result looks reasonable:

```
        if (unlikely(tsd->state != tsd_state_nominal)) {
   121235590:   3c010000        lui     at,0x0
   121235594:   7c03e83b        rdhwr   v1,$29
   121235598:   64259000        daddiu  a1,at,-28672
   12123559c:   0065082d        daddu   at,v1,a1
   1212355a0:   8c230000        lw      v1,0(at)
   1212355a4:   10600011        beqz    v1,1212355ec <__je_malloc_tsd_boot0+0xe4>
   1212355a8:   00000000        nop
__je_malloc_tsd_boot0():
```

```
diff --git a/contrib/binutils/opcodes/mips-opc.c b/contrib/binutils/opcodes/mips-opc.c
index b692536173a..9d4624c2d83 100644
--- a/contrib/binutils/opcodes/mips-opc.c
+++ b/contrib/binutils/opcodes/mips-opc.c
@@ -1188,7 +1188,7 @@ const struct mips_opcode mips_builtin_opcodes[] =
 {"remu",    "z,s,t",    0x0000001b, 0xfc00ffff, RD_s|RD_t|WR_HILO,      0,             I1      },
 {"remu",    "d,v,t",   0,    (int) M_REMU_3,   INSN_MACRO,             0,              I1      },
 {"remu",    "d,v,I",   0,    (int) M_REMU_3I,  INSN_MACRO,             0,              I1      },
-{"rdhwr",   "t,K",     0x7c00003b, 0xffe007ff, WR_t,                   0,              I33     },
+{"rdhwr",   "t,K",     0x7c00003b, 0xffe007ff, WR_t,                   0,              I1      },
 {"rdpgpr",  "d,w",     0x41400000, 0xffe007ff, WR_d,                   0,              I33     },
 {"rfe",     "",                0x42000010, 0xffffffff, 0,                      0,              I1|T3   },
 {"rnas.qh", "X,Q",     0x78200025, 0xfc20f83f, WR_D|RD_T|FP_D,         RD_MACC,        MX      },
```

commit 8e99bc88 (2017-03-22): Make objdump(1) decode rdhwr instruction by default.

It's not quite correct, as the instruction is not valid for the machine
we're supposed to be using; the correct output - like the one produced
by llvm-objdump(1) - would look like this:

121235594:      7c 03 e8 3b     .set    push
    .set    mips32r2
    rdhwr   $3, $29
    .set    pop

Still, it doesn't introduce a regression, and for humans it's much easier
to parse than 0x7c03e83b.

Fixes #202
closed 2017-03-22 by trasz