CTSRD-CHERI/cheribsd #204: CheriABI top(1) crashes with capability length violation

rwatson, 2017-03-22:
CheriABI top(1) fails with a length violation:
```
root@qemu-cheri:~ # top
top: no termcap entry for a `vt100' terminal
CHERI cause: ExcCode: 0x01 RegNum: $c23 (length violation)
$c00: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:0 t:0
$c01: v:1 s:0 p:7fff807d b:0000000120275624 l:0000000000000020 o:0 t:0
$c02: v:1 s:0 p:7fff807d b:000000012033bc18 l:0000000000000008 o:0 t:0
$c03: v:1 s:0 p:7fff807d b:0000000120275624 l:0000000000000020 o:0 t:0
$c04: v:1 s:0 p:7fff807d b:000000012033bc28 l:0000000000000840 o:328 t:0
$c05: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:2023c0 t:0
$c06: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:41a360 t:0
$c07: v:1 s:0 p:7fff807d b:0000007fffffcf5c l:0000000000000001 o:0 t:0
$c08: v:1 s:0 p:7fff807d b:00000001202466b4 l:0000000000000004 o:0 t:0
$c09: v:1 s:0 p:7fff807d b:00000001202466bc l:0000000000000004 o:0 t:0
$c10: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:1202525e0 t:0
$c11: v:1 s:0 p:7fff807d b:0000007fff7ff000 l:00000000007ff3c0 o:0 t:0
$c12: v:1 s:0 p:7ffe8017 b:0000000000000000 l:0000010000000000 o:1200ed5b0 t:0
$c13: v:1 s:0 p:00008055 b:0000007fffffd600 l:0000000000000060 o:0 t:0
$c14: v:1 s:0 p:7fff807d b:000000012026e500 l:0000000000000008 o:0 t:0
$c15: v:1 s:0 p:7fff807d b:000000012026e580 l:0000000000000008 o:0 t:0
$c16: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:12026e480 t:0
$c17: v:1 s:0 p:7ffe8017 b:0000000000000000 l:0000010000000000 o:120010ca4 t:0
$c18: v:1 s:0 p:7fff807d b:0000000120258c20 l:0000000000000004 o:0 t:0
$c19: v:1 s:0 p:7fff807d b:0000000120246e64 l:0000000000000004 o:0 t:0
$c20: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c21: v:1 s:0 p:7fff807d b:0000007fffffd470 l:0000000000000010 o:0 t:0
$c22: v:1 s:0 p:7fff807d b:0000007fffffd458 l:0000000000000008 o:0 t:0
$c23: v:1 s:0 p:7fff807d b:0000007fffffe0e0 l:0000000000000100 o:e0 t:0
$c24: v:1 s:0 p:7fff807d b:0000007fffffd470 l:0000000000000010 o:0 t:0
$c26: v:1 s:0 p:7fff807d b:0000000000000000 l:0000010000000000 o:0 t:0
$c31: v:1 s:0 p:7ffe8017 b:0000000000000000 l:0000010000000000 o:120010dd4 t:0
Mar 22 13:24:44 qemu-cheri kernel: USER_CHERI_EXCEPTION: pid 556 tid 100056 (top), uid 0: CP2 fault (type 0x32)
Mar 22 13:24:45 qemu-cheri kernel: Trapframe Register Dump:
Mar 22 13:24:45 qemu-cheri kernel: zero: 0      at: 0x20        v0: 0x120275624v1: 0x1201e97c0
Mar 22 13:24:45 qemu-cheri kernel: a0: 0x12033bc18      a1: 0   a2: 0   a3: 0xffffffffffffffff
Mar 22 13:24:45 qemu-cheri kernel: a4: 0x1      a5: 0x30        a6: 0x30       a6: 0x100000000
Mar 22 13:24:45 qemu-cheri kernel: t0: 0x10     t1: 0   t2: 0   t3: 0x120270000
Mar 22 13:24:45 qemu-cheri kernel: t8: 0x10     t9: 0x12013e810 s0: 0x120260000s1: 0x12
Mar 22 13:24:45 qemu-cheri kernel: s2: 0x33     s3: 0x120260000 s4: 0   s5: 0x120260000
Mar 22 13:24:45 qemu-cheri kernel: s6: 0x4      s7: 0x20        k0: 0   k1: 0
Mar 22 13:24:45 qemu-cheri kernel: gp: 0x12025a6c0      sp: 0x7fe0a0    s8: 0x7fe0a0    ra: 0x120260000
Mar 22 13:24:45 qemu-cheri kernel: sr: 0x408084b3       mullo: 0x6e40   mulhi: 0 badvaddr: 0x120010dd4
Mar 22 13:24:45 qemu-cheri kernel: cause: 0x48  pc: 0x120010dd4
Signal 34 (core dumped)
```

The code that crashes is in ```get_process_info()```:
```
        /* get a pointer to the states summary array */
        si->procstates = process_states;
   120010da8:   3c010000        lui     at,0x0
   120010dac:   003c082d        daddu   at,at,gp 
   120010db0:   dc21acf8        ld      at,-21256(at)   
   120010db4:   3c020000        lui     v0,0x0
   120010db8:   005c102d        daddu   v0,v0,gp
   120010dbc:   dc42b128        ld      v0,-20184(v0)
   120010dc0:   dc210000        ld      at,0(at)
   120010dc4:   48810087        cfromptr        c1,c0,v0
   120010dc8:   48230840        csetbounds      c3,c1,at
   120010dcc:   49a11800        cmove   c1,c3
   120010dd0:   faebf018        csc     c23,s8,384(c11) 
   120010dd4:   f8370004        csc     c1,zero,64(c23)         <--- Crash here
```

Access is relative to:
```
$c23: v:1 s:0 p:7fff807d b:0000007fffffe0e0 l:0000000000000100 o:e0 t:0
```
The attempted access is at 0xe0 into the 0x100-length buffer, with a further immediate offset of 64 (0x40), which is out of bounds.

I will continue to investigate.

rwatson, 2017-03-22:
It appears this may be another case of a missing argument signature for a function, in this case ```get_process_info()```.

trasz, 2017-09-27:
Fixed by fc0ff8415ad2a436cc044fdecf305936ab976a92.

closed 2017-09-27 by trasz