CTSRD-CHERI/cheribsd #205: cheritest and cheriabitest should include threading-related tests

rwatson, 2017-03-22:
Currently, we don't check that various CHERI features work in the presence of multithreading. It would be useful to start testing threading more thoroughly. However, as threading affects a number of aspects of the run-time environment and we would not want to fail to test the non-threaded case. We're a bit on the wrong side of combinatorics, but it would be tempted to add "-threaded" compilations of the test suites, as well as specific tests for threading-related behaviours. (This is further complicated by a presumed future desire to have -dynamic versions as well..)

RichardsonAlex, 2017-03-22:
I should still have some simple pthread test programs somewhere from the initial pthreads bringup. I've been meaning to add them to cheritest but kept forgetting.

arichardson, 2019-06-28:
This was added a while back.

closed 2019-06-28 by arichardson