CTSRD-CHERI/cheribsd #207: Introduce memswap(3) API

rwatson, 2017-03-23:
Sundry sort-related utility functions end up implementing their own memory-swapping implementations that typically fail to preserve tags (```qsort()```, etc). This might be improved in the future by providing a ```memswap(3)``` API to C that provides a suitable implementation of swapping the contents of two memory locations -- e.g., ```memswap(void *dst, void *src, size_t len)``` that would preserve tags on suitably aligned and suitably sized chunks of memory. This would then ideally be used with uintptr_t when working with data structures that may contain pointers.

davidchisnall, 2017-03-23:
A few other issues that need to be considered:

The C++ `std::swap` template exists for this purpose, but is specialised by type and inlined, so can generate efficient code.  Most of the places where you'd need this in C are hot paths doing little work per swap, so the cost of the function call is likely to dominate.  As such, this would have to either be a macro, or be something with compiler support.  

Swapping can be a lot more efficient if we have a guarantee that `dst` and `src` don't alias.  Should the API require this, or should we have different variants with different guarantees?

rwatson, 2017-03-23:
Agreed on all points -- I'd imagined that ```memswap(3)``` (or whatever it ends up being) would be part of C rather than a library API per se, allowing the compiler to optimise it in much the same manner it currently does with ```memcpy(3)```, converting calls into more efficient inlined code, or vice versa.

On aliasing: If we think the primary consumer is sorting, then it probably makes sense to guarantee non-overlap. I wonder how much ```memmove(3)``` is used in practice...

davidchisnall, 2017-03-23:
> I wonder how much `memmove(3)` is used in practice...
LLVM re

brooksdavis, 2020-04-21:
Closing in favor of CTSRD-CHERI/llvm-project#395

closed 2020-04-21 by brooksdavis