CTSRD-CHERI/cheribsd #209: bsdtar failing in malloc_init due to spurious tags

brooksdavis, 2017-03-30:
I've got a bsdtar binary that is failing for me due to spurious tags appearing in the `.tdata` section of the mapped binary.  The all-zeros `tcache` memory of the `tsd_t` structure has tags that appear to have come from nowhere.  This results in a wrong NULL check and then a NULL pointer dereference.

Here's the failing sequence starting with a read of bad data from malloced TLS memory:

```
0x0000000120409ab0:  clc        c19,zero,32(c2)
    Cap Memory Read [0000000000418060] = v:1 tps:0000000000000000
    c:0000000000000000 b:0000000000000000 l:0000000000000000
    Write C19|v:1 s:0 p:00000000 b:0000000000000000 l:0000000000000000
             |o:0000000000000000 t:0
0x0000000120409ab4:  dsll       at,s4,0x3   
    Write at = 00000000000000c8
0x0000000120409ab8:  cld        s1,at,0(c1)
    Memory Read [0000000120054348] = 0000000000001000
    Write s1 = 0000000000001000
0x0000000120409abc:  cfromptr   c21,c0,zero
0x0000000120409ac0:  cne        at,c19,c21
    Write at = 0000000000000001
0x0000000120409ac4:  beqz       at,0x12040a488
0x0000000120409ac8:  nop
0x0000000120409acc:  sltiu      at,s0,14337
0x0000000120409ad0:  beqz       at,0x12040a198
0x0000000120409ad4:  nop
0x0000000120409ad8:  dsll       s5,s4,0x6
    Write s5 = 0000000000000640
0x0000000120409adc:  clw        v0,s5,112(c19)
mips_cpu_do_interrupt enter: PC 0000000120409adc EPC 000000012041a3bc precise co
processor 2 exception
    Write BadVAddr = 0000000120409adc
```

Tracking back to the source of the bogus data in `0000000000418060` we find this (in `memcpy`):

```
0x0000000120530934:  clc        c5,a1,0(c4)
    Cap Memory Read [0000000120560020] = v:1 tps:0000000000000000
    c:0000000000000000 b:0000000000000000 l:0000000000000000
0x0000000120530938:  daddi      a1,a1,32
    Write a1 = 0000000000000040
0x000000012053093c:  bne        a1,a2,0x120530934
0x0000000120530940:  csc        c5,a1,-32(c3)
    Cap Memory Write [0000000000418060] = v:1 tps:0000000000000000
    c:0000000000000000 b:0000000000000000 l:0000000000000000
```

There is no reference to `0000000120560020` anywhere else in the trace.  A few instructions up from the above snippet, we copy `0000000120560000` having just taken a TLB missed and filled the page.  It's possible the bug is in qemu, but it could also be a pmap bug.  I'm not sure how to begin debugging this.

I do wonder if it would make sense to have a mode in qemu that throws errors on non-sensical bit patterns that have tags attached.  I believe all-zeros is one such pattern.

bsdjhb, 2017-04-13:
I'm not sure.  Note that .tdata is TLS data, so it is supposed to be constructed by rtld for each thread.  Was this a static or dynamic binary, and was this with lld before Alex's recent fix to make __cap_relocs not be all-zero entries?  (You could maybe verify if you have all zero __cap_relocs via llivm-cheri-objdump -C if you still have the old bsdtar binary around.)

brooksdavis, 2017-04-13:
This was static and before Alex's fixes, but these locations were supposed to be NULL (they weren't in `.bss` due to being part of a struct with one non-zero default value).  The just ended up with bogus tags so all-zeros didn't end up equal to NULL due to the tag.

staceyson, 2017-04-13:
Do we know if bsdtar or some lib is doing something fancy to zero out the buffer we are finding these all-zero capabilities in?  I am wondering if it is getting zero'ed out using some non-standard store instruction that is missed in Qemu-CHERI for invalidating tags.

brooksdavis, 2017-04-13:
This is straight from a page mapped by the kernel in `execve()`.

staceyson, 2017-04-13:
Brooks, can you give 9c3b1a057984c60d4b566c5bedf49a7c2775f6e8 a try to see if it fixes this problem?   This is the old way I was invalidating tags for DMA writes which I am not convinced shouldn't work (see the change ab5220b7b0d5e42e17e46b4c91dc7f201b60d620).  It seems that all emulated DMA uses this code path.

jrtc27, 2017-05-01:
This sounds like https://github.com/CTSRD-CHERI/qemu/pull/43, so unless it reappears I suspect this can be closed?

brooksdavis, 2017-05-01:
Agreed.  Things are definitely better after CTSRD-CHERI/qemu#43.

closed 2017-05-01 by brooksdavis