CTSRD-CHERI/cheribsd #214: grep crashes under CHERI128 in hardware

jonwoodruff, 2017-04-06:
CHERI(256) seems unaffected.
Running CHERI128 in hardware with a statically-linked CHERIABI world yields page faults from grep.
Indeed, running grep manually produces the fault.
Very curious if the issue is reproducible in QEMU CHERI128.

toor@beri1:~ # grep TERM .profile
cpuid = 0
$c00: v:1 s:0 p:0007807d b:0000000000000000 l:0000010000000000 o:0 t:0
$c01: v:1 s:0 p:0007807d b:0000000000000000 l:0000010000000000 o:0 t:0
$c02: v:1 s:0 p:0007807d b:0000000120166b90 l:0000000000000008 o:0 t:0
$c03: v:1 s:0 p:0007807d b:0000000120166bc0 l:0000000000000004 o:0 t:0
$c04: v:1 s:0 p:0007807d b:0000000120166b90 l:0000000000000008 o:0 t:0
$c05: v:1 s:0 p:0007807d b:0000000000000000 l:0000010000000000 o:400000 t:0
$c06: v:1 s:0 p:0007807d b:0000000000000000 l:0000010000000000 o:429100 t:0
$c07: v:1 s:0 p:0007807d b:0000000000000000 l:0000010000000000 o:4001c0 t:0
$c08: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c09: v:1 s:0 p:0007807d b:0000007fffffe0e8 l:0000000000000008 o:0 t:0
$c10: v:1 s:0 p:0007807d b:0000007fffffe0e4 l:0000000000000001 o:0 t:0
$c11: v:1 s:0 p:0007807d b:0000007fff7ff000 l:00000000007ff8d0 o:0 t:0
$c12: v:1 s:0 p:00068017 b:0000000000000000 l:0000010000000000 o:12013bb80 t:0
$c13: v:1 s:0 p:00008055 b:0000007fffffdce0 l:0000000000000010 o:0 t:0
$c14: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c15: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c16: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c17: v:1 s:0 p:00068017 b:0000000000000000 l:0000010000000000 o:120092d94 t:0
$c18: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c19: v:1 s:0 p:0007807d b:0000000000000000 l:0000010000000000 o:413000 t:0
$c20: v:1 s:0 p:0007807d b:0000000000000000 l:0000010000000000 o:411020 t:0
$c21: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c22: v:1 s:0 p:0007807d b:0000000120165b10 l:0000000000000010 o:0 t:0
$c23: v:1 s:0 p:0007807d b:0000000120166308 l:0000000000000008 o:0 t:0
$c24: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c26: v:1 s:0 p:0007807d b:0000000000000000 l:0000010000000000 o:0 t:0
$c31: v:1 s:0 p:00068017 b:0000000000000000 l:0000010000000000 o:120092620 t:0
Apr  6 04:15:51 beri1 kernel: BAD_PAGE_FAULT: pid 971 tid 100067 (grep), uid 0: pc 0x120092620 got a read fault (type 0x2) at 0x78
Apr  6 04:15:51 beri1 kernel: Trapframe Register Dump:
Apr  6 04:15:51 beri1 kernel: $0: 0                  at: 0xffffffffffffff78 v0: 0                  v1: 0                 
Apr  6 04:15:51 beri1 kernel: a0: 0x120166b90        a1: 0x1                a2: 0x100              a3: 0x429100          
Apr  6 04:15:51 beri1 kernel: a4: 0                  a5: 0x120166340        a6: 0x278              a7: 0x120166b98       
Apr  6 04:15:51 beri1 kernel: t0: 0x16               t1: 0                  t2: 0xffffffffffffffff t3: 0                 
Apr  6 04:15:51 beri1 kernel: s0: 0x120165b30        s1: 0x140000000        s2: 0x1                s3: 0x6a              
Apr  6 04:15:51 beri1 kernel: s4: 0x6a               s5: 0x12014e590        s6: 0x12014e590        s7: 0x12014e590       
Apr  6 04:15:51 beri1 kernel: t8: 0                  t9: 0x12013bb80        k0: 0                  k1: 0                 
Apr  6 04:15:51 beri1 kernel: gp: 0x12014e590        sp: 0x7ff240           s8: 0x7ff240           ra: 0x1               
Apr  6 04:15:51 beri1 kernel: status: 0x4000fcf3 mullo: 0x808080808080808; mulhi: 0; badvaddr: 0x78
Apr  6 04:15:51 beri1 kernel: cause: 0x108; pc: 0x120092620
Apr  6 04:15:51 beri1 kernel: Page table info for bad address 0x78: pde = 0x9800000004ac3000, pte = 0
Segmentation fault

brooksdavis, 2017-12-08:
Is this still true?

arichardson, 2018-08-08:
Pretty sure this has been fixed, I have been running grep on 128 FPGA quite a bit recently. Feel free to reopen if it still crashes

closed 2018-08-08 by arichardson