CTSRD-CHERI/cheribsd #215: crash in kqueue(2) when running hybrid nginx

trasz, 2017-04-06:
When running nginx hybrid benchmark CheriBSD crashes about one minute with the following backtrace:

```
Trap cause = 3 (TLB miss (store) - kernel mode)
[ thread pid 597 tid 100044 ]
Stopped at      0
db> where
Tracing pid 597 tid 100044 td 0x980000000220b5a0
free+0x118 (?,?,?,?) ra ffffffff802e6780 sp c000000029d37100 sz 64
knote+0x7e8 (?,?,?,?) ra ffffffff802e7344 sp c000000029d37140 sz 32
sys_kqueue+0x374 (?,?,?,?) ra ffffffff802e73a0 sp c000000029d37160 sz 64
sys_kqueue+0x3d0 (?,?,?,?) ra ffffffff802e74fc sp c000000029d371a0 sz 32
knote_fdclose+0x144 (?,?,?,?) ra ffffffff802dba3c sp c000000029d371c0 sz 64
closef+0x354 (?,?,?,?) ra ffffffff802dbd74 sp c000000029d37200 sz 64
kern_close+0x224 (?,?,?,?) ra ffffffff802dbef8 sp c000000029d37240 sz 96
sys_close+0x10 (?,?,?,?) ra ffffffff805abe80 sp c000000029d372a0 sz 32
trap+0x1068 (?,?,?,?) ra ffffffff80598c4c sp c000000029d372c0 sz 416
MipsUserGenException+0x1b4 (?,?,?,12013a5f4) ra 0 sp c000000029d37460 sz 0

```
The problem doesn't occur when nginx is compiled as a CheriABI binary - only with hybrid.

trasz, 2017-04-06:
Is there a clash between EV_FREEUDATA and EV_EOF?

trasz, 2017-04-07:
It fixes the hybrid case, but breaks CheriABI one.  The symptom looks like this:

fetch: http://127.0.0.1/: No error: 0

trasz, 2017-04-07:
Bisection suggests the failure with CheriABI was introduced with adc12632a01e730c47cf7769870917530f9e950b.

brooksdavis, 2017-04-07:
Did you rebuild after adc12632a01e730c47cf7769870917530f9e950b?  It is an ABI breaking change.

trasz, 2017-04-08:
I've just rm-rf-ed the object tree and rebuilt from scratch, the issue still persists:

./nginx-benchmark.sh: nginx binary details:
/usr/local/nginx/sbin/nginx: ELF 64-bit MSB executable, MIPS, MIPS-IV with CHERI-256 (CheriABI) version 1 (FreeBSD), statically linked, for FreeBSD 12.0 (1200024), FreeBSD-style, not stripped
./nginx-benchmark.sh: starting nginx...
./nginx-benchmark.sh: running benchmark 10 times...
fetch: http://127.0.0.1/: No error: 0
fetch: http://127.0.0.1/: No error: 0
fetch: http://127.0.0.1/: No error: 0

brooksdavis, 2017-04-10:
Reverted in 3d696406e15404df11545c104f173fdfa80e871a.  Should work again after a rebuild.

closed 2017-04-10 by brooksdavis