CTSRD-CHERI/cheribsd #220: installworld tries to install libc++ even though it isn't built

kgudka, 2017-05-19:
After building world (using Alex's cheri-build script) as follows:

cd /home/kg365/workspace/cheri/cheri-build/cheribsd && nice make CHERI=256 CHERI_CC=/home/kg365/workspace/cheri/cheri-build/output-latest/sdk256/bin/clang CHERI_CXX=/home/kg365/workspace/cheri/cheri-build/output-
latest/sdk256/bin/clang++ -DDB_FROM_SRC -DNO_WERROR -DNO_CLEAN -DNO_ROOT -DWITHOUT_GDB DEBUG_FLAGS=-g -DWITHOUT_TESTS -s DEBUG_FLAGS=-g -DWITHOUT_TESTS -DWITHOUT_HTML -DWITHOUT_SENDMAIL -DWITHOUT_MAIL -DWITHOUT_S
VNLITE -DWITH_CHERI_PURE -j16 buildworld

installworld then fails due to not finding libc++.a. The installworld command is:

cd /home/kg365/workspace/cheri/cheri-build/cheribsd && env DESTDIR=/home/kg365/workspace/cheri/cheri-build/output-latest/rootfs256-cheriabi nice make CHERI=256 CHERI_CC=/home/kg365/workspace/cheri/cheri-build/output-latest/sdk256/bin/clang CHERI_CXX=/home/kg365/workspace/cheri/cheri-build/output-latest/sdk256/bin/clang++ -DDB_FROM_SRC -DNO_WERROR -DNO_CLEAN -DNO_ROOT -DWITHOUT_GDB DEBUG_FLAGS=-g -DWITHOUT_TESTS -s DEBUG_FLAGS=-g -DWITHOUT_TESTS -DWITHOUT_HTML -DWITHOUT_SENDMAIL -DWITHOUT_MAIL -DWITHOUT_SVNLITE -DWITH_CHERI_PURE -DWITHOUT_LIBCPLUSPLUS installworld

The resulting error is:

===> lib/libc++ (install)
install: libc++.a: No such file or directory

A temporary workaround is to pass -DWITHOUT_LIBCPLUSPLUS to installworld but I'm not sure why this error has now started to occur?

RichardsonAlex, 2017-05-19:
This appears to be caused by my recent change to build the kernel with clang instead of GCC. It seems like installworld reuses the compiler information from the buildkernel step and therefore assumes libc++ needs to be installed because compiler == clang instead of GCC. Another workaround could be to run a target that updates the cached compiler info (no sure which is the fastest one).

RichardsonAlex, 2017-06-12:
I believe this can be closed now that we build libc++

closed 2017-06-12 by RichardsonAlex