CTSRD-CHERI/cheribsd #224: Qt unittests block forever (probably due to libthr/_umtx_op) issues

RichardsonAlex, 2017-06-23:
I was trying to run the unittests for QtBase on QEMU and both MIPS as well as CheriABI all loop forever on startup.

I get the following system call trace for CheriABI (it freezes before producing any output):
```
   636 ktrace   RET   ktrace 0
   636 ktrace   CALL  execve(0x7fffffecc7,0x7fffffe320,0x7fffffe360)
   636 ktrace   NAMI  "/local/scratch/alr48/cheri/build-release/qtbase-256-build/tests/auto/corelib/global/qtendian/tst_qtendian"
   636 tst_qtendian RET   execve 0
   636 tst_qtendian CALL  readlink(0x12001d786,0x7fffffd69c,0x400)
   636 tst_qtendian NAMI  "/etc/malloc.conf"
   636 tst_qtendian RET   readlink -1 errno 2 No such file or directory
   636 tst_qtendian CALL  issetugid
   636 tst_qtendian RET   issetugid 0
   636 tst_qtendian CALL  __sysctl(0x7fffffd958,0x2,0x7fffffda38,0x7fffffda30,0x12001ba20,0xd)
   636 tst_qtendian SCTL  "sysctl.name2oid"
   636 tst_qtendian RET   __sysctl 0
   636 tst_qtendian CALL  __sysctl(0x7fffffda38,0x2,0x7fffffdb9c,0x7fffffdb90,0,0)
   636 tst_qtendian SCTL  "vm.overcommit"
   636 tst_qtendian RET   __sysctl 0
   636 tst_qtendian CALL  mmap(0,0x200000,0x3<PROT_READ|PROT_WRITE>,0x1002<MAP_PRIVATE|MAP_ANON>,0xffffffff,0)
   636 tst_qtendian RET   mmap 4096/0x1000
   636 tst_qtendian CALL  munmap(0x1000,0x200000)
   636 tst_qtendian RET   munmap 0
   636 tst_qtendian CALL  mmap(0,0x3ff000,0x3<PROT_READ|PROT_WRITE>,0x1002<MAP_PRIVATE|MAP_ANON>,0xffffffff,0)
   636 tst_qtendian RET   mmap 4096/0x1000
   636 tst_qtendian CALL  munmap(0x1000,0x1ff000)
   636 tst_qtendian RET   munmap 0
   636 tst_qtendian CALL  mmap(0,0x200000,0x3<PROT_READ|PROT_WRITE>,0x1002<MAP_PRIVATE|MAP_ANON>,0xffffffff,0)
   636 tst_qtendian RET   mmap 4194304/0x400000
   636 tst_qtendian CALL  sysarch(0x1,0)
   636 tst_qtendian RET   sysarch 0
   636 tst_qtendian CALL  sched_yield
   636 tst_qtendian RET   sched_yield 0
   636 tst_qtendian CALL  sched_yield
   636 tst_qtendian RET   sched_yield 0
   636 tst_qtendian CALL  sched_yield
   636 tst_qtendian RET   sched_yield 0
   636 tst_qtendian CALL  sched_yield
   636 tst_qtendian RET   sched_yield 0
   636 tst_qtendian CALL  sched_yield
   636 tst_qtendian RET   sched_yield 0
   636 tst_qtendian CALL  sched_yield
   636 tst_qtendian RET   sched_yield 0
   636 tst_qtendian CALL  sched_yield
   636 tst_qtendian RET   sched_yield 0
   636 tst_qtendian CALL  sched_yield
   636 tst_qtendian RET   sched_yield 0
   636 tst_qtendian CALL  sched_yield
   636 tst_qtendian RET   sched_yield 0
   636 tst_qtendian CALL  sched_yield
   636 tst_qtendian RET   sched_yield 0
   636 tst_qtendian CALL  sched_yield
   636 tst_qtendian RET   sched_yield 0
   636 tst_qtendian CALL  sched_yield
   636 tst_qtendian RET   sched_yield 0
   636 tst_qtendian CALL  sched_yield
   636 tst_qtendian RET   sched_yield 0
```
MIPS get's a bit further but then freezes on _umtx_op:
```
675 tst_qtendian GIO   fd 1 wrote 50 bytes
       "********* Start testing of tst_QtEndian *********
       "
   675 tst_qtendian RET   write 50/0x32
   675 tst_qtendian CALL  write(0x1,0x40e39000,0x70)
   675 tst_qtendian GIO   fd 1 wrote 112 bytes
       "Config: Using QtTest library 5.9.1, Qt 5.9.1 (mips64-big_endian-lp64-n64 static release build; by Clang 5.0.0 )
       "
   675 tst_qtendian RET   write 112/0x70
   675 tst_qtendian CALL  clock_gettime(0x4,0x7fffffe6b0)
   675 tst_qtendian RET   clock_gettime 0
   675 tst_qtendian CALL  mmap(0,0x1000,0x3<PROT_READ|PROT_WRITE>,0x1002<MAP_PRIVATE|MAP_ANON>,0xffffffffffffffff,0)
   675 tst_qtendian RET   mmap 1079685120/0x405ab000
   675 tst_qtendian CALL  mmap(0x7ffbdfd000,0x201000,0x7<PROT_READ|PROT_WRITE|PROT_EXEC>,0x400<MAP_STACK>,0xffffffffffffffff,0)
   675 tst_qtendian RET   mmap 549686595584/0x7ffbdfd000
   675 tst_qtendian CALL  mprotect(0x7ffbdfd000,0x1000,0<PROT_NONE>)
   675 tst_qtendian RET   mprotect 0
   675 tst_qtendian CALL  thr_new(0x7fffffe580,0x68)
   675 tst_qtendian RET   thr_new 0
   675 tst_qtendian CALL  _umtx_op(0x40814770,UMTX_OP_WAIT_UINT_PRIVATE,0,0,0)
   675 tst_qtendian RET   fork 0
   675 tst_qtendian CALL  mmap(0,0x1000,0x3<PROT_READ|PROT_WRITE>,0x1002<MAP_PRIVATE|MAP_ANON>,0xffffffffffffffff,0)
   675 tst_qtendian RET   mmap 1079689216/0x405ac000
   675 tst_qtendian CALL  pipe2(0x40e627f8,0x100004)
   675 tst_qtendian RET   pipe2 0
   675 tst_qtendian CALL  _umtx_op(0x40e177b8,UMTX_OP_NWAKE_PRIVATE,0x1,0,0)
   675 tst_qtendian RET   _umtx_op -1 errno 45 Operation not supported
   675 tst_qtendian CALL  _umtx_op(0x405ab008,UMTX_OP_WAIT_UINT_PRIVATE,0,0,0)
   675 tst_qtendian RET   _umtx_op -1 errno 4 Interrupted system call
   675 tst_qtendian PSIG  SIGINT caught handler=0x407e5508 mask=0x0 code=SI_KERNEL
   675 tst_qtendian CALL  sigprocmask(SIG_SETMASK,0x59b8ac,0)
   675 tst_qtendian RET   sigprocmask 0
   675 tst_qtendian CALL  clock_gettime(0x4,0x59b4c0)
   675 tst_qtendian RET   clock_gettime 0
   675 tst_qtendian CALL  clock_gettime(0x4,0x59b4c0)
   675 tst_qtendian RET   clock_gettime 0
   675 tst_qtendian CALL  clock_gettime(0x4,0x59ae70)
   675 tst_qtendian RET   clock_gettime 0
   675 tst_qtendian CALL  write(0x1,0x40e39000,0x38)
   675 tst_qtendian GIO   fd 1 wrote 56 bytes
       "QFATAL : tst_QtEndian::initTestCase() Received signal 2
```

RichardsonAlex, 2017-06-23:
After a bit of debugging it turns out the CheriABI case has the following backtrace which could indicate a bug in libcxxrt instead:

```
#0  __cxa_guard_acquire (guard_object=<optimized out>) at cxxrt_guard.cc:134
#1  0x00000001206bf794 in (anonymous namespace)::Q_QGS_qtDefaultCategory::innerFunction () at /local/scratch/alr48/cheri/qt5/qtbase/src/corelib/io/qloggingcategory.cpp:47
#2  0x00000001206bf544 in QGlobalStatic<QLoggingCategory, (anonymous namespace)::Q_QGS_qtDefaultCategory::innerFunction(), (anonymous namespace)::Q_QGS_qtDefaultCategory::guard>::operator()() (this=<optimized out>)
    at /local/scratch/alr48/cheri/build-release/qtbase-256-build/src/corelib/../../include/QtCore/../../../../qt5/qtbase/src/corelib/global/qglobalstatic.h:135
#3  0x00000001206bf46c in QLoggingCategory::defaultCategory () at /local/scratch/alr48/cheri/qt5/qtbase/src/corelib/io/qloggingcategory.cpp:387
#4  0x000000012035ae24 in qt_message_print (msgType=<optimized out>, context=..., message=...) at /local/scratch/alr48/cheri/qt5/qtbase/src/corelib/global/qlogging.cpp:1643
#5  0x000000012034d3d8 in qt_message (msgType=<optimized out>, context=..., msg=<optimized out>, ap=<optimized out>) at /local/scratch/alr48/cheri/qt5/qtbase/src/corelib/global/qlogging.cpp:281
#6  0x00000001203511f8 in QMessageLogger::fatal (this=<optimized out>, msg=<optimized out>) at /local/scratch/alr48/cheri/qt5/qtbase/src/corelib/global/qlogging.cpp:793

q
#7  0x000000012033935c in qt_assert_x (where=<optimized out>, what=<optimized out>, file=<optimized out>, line=<optimized out>) at /local/scratch/alr48/cheri/qt5/qtbase/src/corelib/global/qglobal.cpp:3091
#8  0x00000001202e1448 in QMutexLocker::QMutexLocker (this=<optimized out>, m=<optimized out>) at ../../include/QtCore/../../../../qt5/qtbase/src/corelib/thread/qmutex.h:195
#9  0x00000001206c62ec in QLoggingRegistry::registerCategory (this=<optimized out>, cat=<optimized out>, enableForLevel=<optimized out>) at /local/scratch/alr48/cheri/qt5/qtbase/src/corelib/io/qloggingregistry.cpp:336
#10 0x00000001206bed98 in QLoggingCategory::init (this=<optimized out>, category=<optimized out>, severityLevel=<optimized out>) at /local/scratch/alr48/cheri/qt5/qtbase/src/corelib/io/qloggingcategory.cpp:248
#11 0x00000001206bebbc in QLoggingCategory::QLoggingCategory (this=<optimized out>, category=<optimized out>) at /local/scratch/alr48/cheri/qt5/qtbase/src/corelib/io/qloggingcategory.cpp:220
#12 0x00000001206bfa00 in (anonymous namespace)::Q_QGS_qtDefaultCategory::innerFunction()::Holder::Holder() (this=<optimized out>) at /local/scratch/alr48/cheri/qt5/qtbase/src/corelib/io/qloggingcategory.cpp:47
#13 0x00000001206bf7d4 in (anonymous namespace)::Q_QGS_qtDefaultCategory::innerFunction () at /local/scratch/alr48/cheri/qt5/qtbase/src/corelib/io/qloggingcategory.cpp:47
#14 0x00000001206bf544 in QGlobalStatic<QLoggingCategory, (anonymous namespace)::Q_QGS_qtDefaultCategory::innerFunction(), (anonymous namespace)::Q_QGS_qtDefaultCategory::guard>::operator()() (this=<optimized out>)
    at /local/scratch/alr48/cheri/build-release/qtbase-256-build/src/corelib/../../include/QtCore/../../../../qt5/qtbase/src/corelib/global/qglobalstatic.h:135
#15 0x00000001206bf46c in QLoggingCategory::defaultCategory () at /local/scratch/alr48/cheri/qt5/qtbase/src/corelib/io/qloggingcategory.cpp:387
#16 0x000000012035ae24 in qt_message_print (msgType=<optimized out>, context=..., message=...) at /local/scratch/alr48/cheri/qt5/qtbase/src/corelib/global/qlogging.cpp:1643
#17 0x000000012034d3d8 in qt_message (msgType=<optimized out>, context=..., msg=<optimized out>, ap=<optimized out>) at /local/scratch/alr48/cheri/qt5/qtbase/src/corelib/global/qlogging.cpp:281
#18 0x00000001203511f8 in QMessageLogger::fatal (this=<optimized out>, msg=<optimized out>) at /local/scratch/alr48/cheri/qt5/qtbase/src/corelib/global/qlogging.cpp:793
#19 0x000000012033935c in qt_assert_x (where=<optimized out>, what=<optimized out>, file=<optimized out>, line=<optimized out>) at /local/scratch/alr48/cheri/qt5/qtbase/src/corelib/global/qglobal.cpp:3091
#20 0x00000001202e1448 in QMutexLocker::QMutexLocker (this=<optimized out>, m=<optimized out>) at ../../include/QtCore/../../../../qt5/qtbase/src/corelib/thread/qmutex.h:195
#21 0x000000012033a51c in qEnvironmentVariableIntValue (varName=<optimized out>, ok=<optimized out>) at /local/scratch/alr48/cheri/qt5/qtbase/src/corelib/global/qglobal.cpp:3318
#22 0x00000001202d53c8 in disableCoreDump () at /local/scratch/alr48/cheri/qt5/qtbase/src/testlib/qtestcase.cpp:188
#23 0x00000001202ba774 in (anonymous namespace)::disableCoreDump_ctor_class_::disableCoreDump_ctor_class_ (this=<optimized out>) at /local/scratch/alr48/cheri/qt5/qtbase/src/testlib/qtestcase.cpp:199
#24 0x00000001202ba6fc in __cxx_global_var_init () at /local/scratch/alr48/cheri/qt5/qtbase/src/testlib/qtestcase.cpp:199
#25 0x00000001202d5714 in _GLOBAL__sub_I_qtestcase.cpp ()
#26 0x00000001202b06f8 in crt_call_constructors ()
```

brooksdavis, 2017-12-08:
I fixed some obviously broken things with _umtx_op during our last attempt at a paper.  Unless we've abandoned this effort it might be worth trying again.

arichardson, 2017-12-11:
Qt 5.10 was releases last week. I will rebase my changes on top of that and give it another try

jrtc27, 2018-03-01:
Assuming this is the same as what I've been encountering in my work (and fixed in the pull request visible above), this is because a condition variable is being signalled whilst that thread is still holding the lock used when waiting on the cv in another thread. In this case, the waiting thread gets removed from the cv's sleep queue and added to the lock's deferred list, all members of which are woken up with a `UMTX_OP_NWAKE_PRIVATE` once the lock is unlocked. However, if `COMPAT_CHERIABI` is defined, the kernel's `__umtx_op_nwake_private` just delegates to `__umtx_op_unimpl`, regardless of whether the binary in question is a CheriABI binary or not. Whilst the array-of-capabilities to array-of-addresses conversion is not yet implemented, we can still use the standard implementation when the current binary is a normal MIPS/hybrid binary (i.e. not `SV_CHERI`). What doesn't help is that the return value of `_umtx_op_err` isn't checked in many cases, including this one, so this is a silent failure.

(If you signal a condition variable *without* holding the lock, it gets woken immediately with a single `UMTX_OP_WAKE_PRIVATE`, that *is* implemented for CheriABI)

arichardson, 2018-03-19:
Now that gdb can unwind out of system call stubs and the applications no longer fault on startup (https://github.com/CTSRD-CHERI/clang/commit/b7d18c83c52836a1a03c7ce7abbd5f518759fc2a) I've managed to get a backtrace:

```
  4239 tst_qtendian CALL  mmap(0,0x2000,0x3<PROT_READ|PROT_WRITE>,0xd001002<MAP_PRIVATE|MAP_ANON|MAP_ALIGNED(13)>,0xffffffff,0)
  4239 tst_qtendian RET   mmap 5933113344/0x161a42000
  4239 tst_qtendian CALL  _umtx_op(0x161a04320,UMTX_OP_SEM2_WAIT,0,0,0)
  4239 tst_qtendian RET   _umtx_op RESTART
```

```
#0  _umtx_op () at _umtx_op.S:3
#1  0x000000012061ecbc in usem_wait (sem=<optimized out>, clock_id=<optimized out>, rmtp=<optimized out>, flags=<optimized out>,
    rqtp=<optimized out>) at cancelpoints_sem_new.c:365
#2  _sem_clockwait_np (sem=<optimized out>, clock_id=0, flags=<optimized out>, rqtp=<optimized out>, rmtp=<optimized out>)
    at cancelpoints_sem_new.c:425
#3  0x000000012061ee34 in _sem_timedwait (sem=0x161a04320, abstime=0x0) at cancelpoints_sem_new.c:436
#4  _sem_wait (sem=0x161a04320) at cancelpoints_sem_new.c:444
#5  0x00000001201f3f64 in QMutexPrivate::wait (this=<optimized out>, timeout=23)
    at /exports/users/alr48/sources/qt5/qtbase/src/corelib/thread/qmutex_unix.cpp:84
#6  0x00000001201f3284 in QBasicMutex::lockInternal (this=<optimized out>, timeout=-1)
    at /exports/users/alr48/sources/qt5/qtbase/src/corelib/thread/qmutex.cpp:575
#7  0x00000001201f2d5c in QBasicMutex::lockInternal (this=<optimized out>)
    at /exports/users/alr48/sources/qt5/qtbase/src/corelib/thread/qmutex.cpp:491
#8  QMutex::lock (this=<optimized out>) at /exports/users/alr48/sources/qt5/qtbase/src/corelib/thread/qmutex.cpp:229
#9  0x00000001201df210 in QMutexLocker::QMutexLocker(QBasicMutex* cap) (this=<optimized out>, m=<optimized out>)
    at ../../include/QtCore/../../../../../../../exports/users/alr48/sources/qt5/qtbase/src/corelib/thread/qmutex.h:206
#10 qgetenv(char const* cap) (varName=0x0) at /exports/users/alr48/sources/qt5/qtbase/src/corelib/global/qglobal.cpp:3223
#11 0x00000001204d47dc in QJpUnicodeConv::newConverter (rule=23)
    at /exports/users/alr48/sources/qt5/qtbase/src/corelib/codecs/qjpunicode.cpp:766
#12 0x00000001204d1f98 in QEucJpCodec::QEucJpCodec (this=<optimized out>)
    at /exports/users/alr48/sources/qt5/qtbase/src/corelib/codecs/qeucjpcodec.cpp:92
#13 0x00000001204c35f0 in setup () at /exports/users/alr48/sources/qt5/qtbase/src/corelib/codecs/qtextcodec.cpp:282
#14 0x00000001204c529c in setupLocaleMapper () at /exports/users/alr48/sources/qt5/qtbase/src/corelib/codecs/qtextcodec.cpp:164
#15 QTextCodec::codecForLocale () at /exports/users/alr48/sources/qt5/qtbase/src/corelib/codecs/qtextcodec.cpp:698
#16 0x000000012027137c in QString::fromLocal8Bit_helper(char const* cap, int) (str=<optimized out>, size=12)
    at /exports/users/alr48/sources/qt5/qtbase/src/corelib/tools/qstring.cpp:5056
#17 0x000000012038b4c4 in QString::fromLocal8Bit(char const* cap, int) (str=0x7fffffed08 "tst_qtendian", size=-1)
    at ../../include/QtCore/../../../../../../../exports/users/alr48/sources/qt5/qtbase/src/corelib/tools/qstring.h:566
#18 QCoreApplicationPrivate::appName (this=<optimized out>)
    at /exports/users/alr48/sources/qt5/qtbase/src/corelib/kernel/qcoreapplication.cpp:172
#19 0x000000012038f158 in QCoreApplicationPrivate::init (this=0x161a04000)
    at /exports/users/alr48/sources/qt5/qtbase/src/corelib/kernel/qcoreapplication.cpp:800
#20 0x000000012038ef60 in QCoreApplication::QCoreApplication(int& cap, char* cap* cap, int) (this=<optimized out>, argc=<optimized out>,
    argv=<optimized out>, _internal=<optimized out>) at /exports/users/alr48/sources/qt5/qtbase/src/corelib/kernel/qcoreapplication.cpp:778
#21 0x00000001201a4888 in main (argc=-5, argv=<optimized out>)
    at /exports/users/alr48/sources/qt5/qtbase/tests/auto/corelib/global/qtendian/tst_qtendian.cpp:216
```

bsdjhb, 2018-03-19:
Were you hung in _umtx_op before attaching gdb and the ERESTART is due to attaching the debugger?  If so, can you pop up to frame 3 and 'p *sem'?

arichardson, 2018-03-21:
Turns out the mutexes were never unlocked due to CTSRD-CHERI/clang#189 so waiting for ever was actually the correct behaviour. I am now running into an unrelated crash so I guess this can be closed for now.

arichardson, 2018-04-06:
I have now fixed lots of issues in Qt and now it's back to the deadlock on shutdown (during `cleanupTestCase()`).

There are two threads both blocking in this line:
```
        return _umtx_op_err(__DEVOLATILE(void *, mtx), shared ?
            UMTX_OP_WAIT_UINT : UMTX_OP_WAIT_UINT_PRIVATE, id,
            (void *)(uintptr_t)tm_size, __DECONST(void *, tm_p));
```

Thread 1 backtrace:
```
#0  _umtx_op () at _umtx_op.S:3
#1  0x0000000120a32310 in _umtx_op_err (obj=0x121334ea0 <default_wake_addr+32>, op=<optimized out>,
    val=0, uaddr=0x0, uaddr2=0x0) at /exports/users/alr48/sources/cheribsd/lib/libthr/thread/thr_umtx.c:37
#2  _thr_umtx_timedwait_uint (mtx=0x121334ea0 <default_wake_addr+32>, id=15, clockid=<optimized out>,
    abstime=<optimized out>, shared=0)
    at /exports/users/alr48/sources/cheribsd/lib/libthr/thread/thr_umtx.c:234
#3  0x0000000120a3bf50 in _thr_sleep (curthread=<optimized out>, clockid=<optimized out>, abstime=0x0)
    at /exports/users/alr48/sources/cheribsd/lib/libthr/thread/thr_kern.c:201
#4  0x0000000120a46a00 in cond_wait_user (cvp=0x161e342e0, mp=0x161df20c0, abstime=0x0, cancel=1)
    at /exports/users/alr48/sources/cheribsd/lib/libthr/thread/thr_cond.c:304
#5  0x0000000120a45a84 in cond_wait_common (cond=<optimized out>, mutex=0x161e24ea0,
    abstime=<optimized out>, cancel=1)
    at /exports/users/alr48/sources/cheribsd/lib/libthr/thread/thr_cond.c:364
#6  0x0000000120a45b6c in __pthread_cond_wait (cond=0x121334e80 <default_wake_addr>,
    mutex=0x121336020 <sc_table>)
    at /exports/users/alr48/sources/cheribsd/lib/libthr/thread/thr_cond.c:378
#7  0x0000000120281308 in QWaitConditionPrivate::wait (this=<optimized out>, time=<optimized out>)
    at /exports/users/alr48/sources/qt5/qtbase/src/corelib/thread/qwaitcondition_unix.cpp:143
#8  0x0000000120280e24 in QWaitCondition::wait(QMutex* cap, unsigned long) (this=<optimized out>,
    mutex=<optimized out>, time=<optimized out>)
    at /exports/users/alr48/sources/qt5/qtbase/src/corelib/thread/qwaitcondition_unix.cpp:215
#9  0x00000001202795b8 in QThread::wait (this=<optimized out>, time=<optimized out>)
    at /exports/users/alr48/sources/qt5/qtbase/src/corelib/thread/qthread_unix.cpp:741
#10 0x00000001201f0e30 in QTest::WatchDog::~WatchDog (this=<optimized out>)
    at /exports/users/alr48/sources/qt5/qtbase/src/testlib/qtestcase.cpp:971
#11 0x00000001201f1058 in QTest::WatchDog::~WatchDog (this=<optimized out>)
    at /exports/users/alr48/sources/qt5/qtbase/src/testlib/qtestcase.cpp:965
#12 0x00000001201fbb00 in QScopedPointerDeleter<QTest::WatchDog>::cleanup(QTest::WatchDog* cap) (
    pointer=<optimized out>)
    at ../../include/QtCore/../../../../../../../exports/users/alr48/sources/qt5/qtbase/src/corelib/tools/qscopedpointer.h:60
#13 0x00000001201e9a2c in QScopedPointer<QTest::WatchDog, QScopedPointerDeleter<QTest::WatchDog> >::~QScopedPointer (this=<optimized out>)
    at ../../include/QtCore/../../../../../../../exports/users/alr48/sources/qt5/qtbase/src/corelib/tools/qscopedpointer.h:107
#14 0x00000001201d9d98 in QTest::TestMethods::invokeTests(QObject* cap) const (this=<optimized out>,
    testObject=<optimized out>) at /exports/users/alr48/sources/qt5/qtbase/src/testlib/qtestcase.cpp:1408
#15 0x00000001201dbb5c in QTest::qRun ()
    at /exports/users/alr48/sources/qt5/qtbase/src/testlib/qtestcase.cpp:1831
#16 0x00000001201da83c in QTest::qExec(QObject* cap, int, char* cap* cap) (testObject=<optimized out>,
    argc=<optimized out>, argv=<optimized out>)
    at /exports/users/alr48/sources/qt5/qtbase/src/testlib/qtestcase.cpp:1718
#17 0x00000001201c5724 in main (argc=<optimized out>, argv=<optimized out>)
    at /exports/users/alr48/sources/qt5/qtbase/tests/auto/corelib/global/qtendian/tst_qtendian.cpp:216
```
Thread 2 backtrace:
```
#0  _umtx_op () at _umtx_op.S:3
#1  0x0000000120a32310 in _umtx_op_err (obj=0x160ba1020, op=<optimized out>, val=0, uaddr=0x0, uaddr2=0x0)
    at /exports/users/alr48/sources/cheribsd/lib/libthr/thread/thr_umtx.c:37
#2  _thr_umtx_timedwait_uint (mtx=0x160ba1020, id=15, clockid=<optimized out>, abstime=<optimized out>,
    shared=0) at /exports/users/alr48/sources/cheribsd/lib/libthr/thread/thr_umtx.c:234
#3  0x0000000120a3bf50 in _thr_sleep (curthread=<optimized out>, clockid=<optimized out>, abstime=0x0)
    at /exports/users/alr48/sources/cheribsd/lib/libthr/thread/thr_kern.c:201
#4  0x0000000120a46a00 in cond_wait_user (cvp=0x161e34300, mp=0x161df2200, abstime=0x0, cancel=1)
    at /exports/users/alr48/sources/cheribsd/lib/libthr/thread/thr_cond.c:304
#5  0x0000000120a45a84 in cond_wait_common (cond=<optimized out>, mutex=0x161e24e40,
    abstime=<optimized out>, cancel=1)
    at /exports/users/alr48/sources/cheribsd/lib/libthr/thread/thr_cond.c:364
#6  0x0000000120a45b6c in __pthread_cond_wait (cond=0x160ba1000, mutex=0x161dfe120)
    at /exports/users/alr48/sources/cheribsd/lib/libthr/thread/thr_cond.c:378
#7  0x0000000120281308 in QWaitConditionPrivate::wait (this=<optimized out>, time=<optimized out>)
    at /exports/users/alr48/sources/qt5/qtbase/src/corelib/thread/qwaitcondition_unix.cpp:143
#8  0x0000000120280e24 in QWaitCondition::wait(QMutex* cap, unsigned long) (this=<optimized out>,
    mutex=<optimized out>, time=<optimized out>)
    at /exports/users/alr48/sources/qt5/qtbase/src/corelib/thread/qwaitcondition_unix.cpp:215
#9  0x00000001201f1258 in QTest::WatchDog::run (this=<optimized out>)
    at /exports/users/alr48/sources/qt5/qtbase/src/testlib/qtestcase.cpp:993
#10 0x0000000120276b74 in QThreadPrivate::start(void* cap) (arg=<optimized out>)
    at /exports/users/alr48/sources/qt5/qtbase/src/corelib/thread/qthread_unix.cpp:376
#11 0x0000000120a455a8 in thread_start (curthread=<optimized out>)
    at /exports/users/alr48/sources/cheribsd/lib/libthr/thread/thr_create.c:295
#12 0x0000000000000000 in ?? ()
```

commit 8cc870af (2018-04-09): Work around #255 (UMTX_OP_NWAKE_PRIVATE unimplemented)

With this change I can finally run the tst_qtendian QtBase unit test:
```
tst_qtendian: _rtld_thread_init is not available when statically linked.
********* Start testing of tst_QtEndian *********
Config: Using QtTest library 5.10.0, Qt 5.10.0 (mips64-big_endian-ilp32-n64 static debug build; by Clang 7.0.0 (https://github.com/llvm-mirror/clang.git c730fc93d43653089931927b78bb04c585155be6) (https://github.com/llvm-mirror/llvm.git 2460f6c1e6da961d5349a294772b8d619e81a094))
PASS   : tst_QtEndian::initTestCase()
PASS   : tst_QtEndian::fromBig [... cut]
closed 2018-04-09 by arichardson