CTSRD-CHERI/cheribsd #225: mprotect / mmap should allow clearing CHERI-specific permissions

davidchisnall, 2017-07-04:
Currently, we have store-capability and load-capability permissions in the TLB, exposed from pmap as `PMAP_ENTER_NOSTORETAGS` and `PMAP_ENTER_NOLOADTAGS`, but we do not have a way of setting them from userspace.  These are useful for various forms of memory management, where we want to ensure that capabilities are not stored in a region.

brooksdavis, 2017-07-06:
There's plenty of room in the `prot` arguments for more `PROT_` bits.  It does look like `vm_prot_t` is more constrained (it's a `uchar` and 4 bits are used).  At a glance (and with a defiantly incomplete understanding of `pmap_enter()` it looks `PROT_LOADCAP` and `PROT_STORECAP` would be fairly straightforward.

davidchisnall, 2017-07-06:
That would be very helpful, as it would mean that a concurrent GC could observe any updates to pointers.