CTSRD-CHERI/cheribsd #228: MIPS crt1.o is wrong when compiled with clang

RichardsonAlex, 2017-07-17:
I have been trying to get static binaries to work with clang but but have so far not succeeded in getting it to emit relocations that match GCC in crt1.o.

GCC generated (works): https://gist.github.com/RichardsonAlex/81e741bdc14f63162c65fde8546f436d
Clang generated (won't work): https://gist.github.com/RichardsonAlex/0be4f973e3e037d0241d49856e54bc2d

I think one big problem is that the file lib/csu/mips/crt1_c.c gets compiled to an assembly file first. This means that the LLVM asm parser is used for code gen and it often disagrees with clang on what certain flags mean.
It seems to me like the only reason that this is done is because crtbrand.c needs to be processed change the section type and crtbrand.c is #included into crt1_c.c.

Compiling a separate crtbrand.s -> crtbrand.o and using `ld -r` to add it to crt1.o should hopefully make the situation a bit better.

bsdjhb, 2017-07-17:
Are you sure you have the DLA patch?  I get the same relocations as GCC in <_start> in crt1.o.  In particular, the third relocation is a GP-relative CALL16 as it is for binutils instead of the non-PIC HIGHEST/HIGHER/HI/LO:

```
0000000000000000 <__start>:
   0:   240cffe0        li      t0,-32
   4:   03ace824        and     sp,sp,t0
   8:   67bdffe0        daddiu  sp,sp,-32
   c:   ffbf0018        sd      ra,24(sp)
  10:   ffbc0008        sd      gp,8(sp)
  14:   3c1c0000        lui     gp,0x0
                        14: R_MIPS_GPREL16      __start
                        14: R_MIPS_SUB  *ABS*
                        14: R_MIPS_HI16 *ABS*
  18:   279c0000        addiu   gp,gp,0
                        18: R_MIPS_GPREL16      __start
                        18: R_MIPS_SUB  *ABS*
                        18: R_MIPS_LO16 *ABS*
  1c:   0399e02d        daddu   gp,gp,t9
  20:   df990000        ld      t9,0(gp)
                        20: R_MIPS_CALL16       _start1
                        20: R_MIPS_NONE *ABS*
                        20: R_MIPS_NONE *ABS*
  24:   03200008        jr      t9
  28:   00000000        nop
  2c:   00000000        nop
```
I would be happy to get rid of the C -> ASM dance for the CSU files in general mind though.

Looking further in the files, the version I have compiled with clang here more closely matches the GCC one (e.g. using GOT_DISP instead of HI16/LO16, etc.)

RichardsonAlex, 2017-07-17:
I think one problem could be that I compiled with `-mxgot` which can sometimes cause the MIPS backend to do weird stuff. I'll try again tomorrow without mxgot. I am using the master branch of LLVM where have cherry picked your three submitted patches. Or did I possibly miss one?

bsdjhb, 2017-07-18:
Hmm, I think the DLA fix is in 'master'.  I am not using -mxgot for most of CheriBSD (I've tried to use it for rescue but rescue still fails to link).  Did you look at the generated assembly from clang?  I see it using %got_disp(), etc. before the sed.  Would be interesting to see if that was true in your case as well?

RichardsonAlex, 2017-08-11:
Seems like this is only an issue if compiled with -mxgot

closed 2017-08-11 by RichardsonAlex