CTSRD-CHERI/cheribsd #232: pw useradd broken after -mxgot removal

brooksdavis, 2017-09-02:
Kyua fails to install (and thus tests fail to run) on 128-bit CHERI after d3f7ec2ce5ef14a571d33091ab8e0093c02e52a2 which removes -mxgot.  The actual failure is due to the `pw` command failing like this:
```
# pw useradd tests -u 977
pw: uid `0' has already been allocated
```
I've verified this commit is the problem by git bisect.

RichardsonAlex, 2017-09-04:
Does it work on 256?

Not quite sure how to figure out which library/object file needs to be compiled with -mxgot other than just iteratively adding it.

brooksdavis, 2017-09-05:
It does work on 256-bit.  I tried compiling pw with -mxgot and that didn't work.

brooksdavis, 2017-09-05:
Building libc with -mxgot fixes pw.  The failing call is a call to getpwuid which is a horrible mess of indirections for nss.  I'll try and see if I can reduce the set.

Jerryxia32, 2017-09-05:
I recently experienced a bug after removing the -mxgot flag in CheriOS, and opened an issue at [https://github.com/CTSRD-CHERI/llvm/issues/234](url). I wonder if there is a slight chance that these two are related. My bug was about wrong instruction scheduling for writing local static variables, and changing them to global variables worked around the problem, although it was purecap with -mllvm -cheri-no-global-bounds.

commit a9c53eb0 (2017-09-05): Compile getpwent.c with -mxgox.

Without this, getpwuid() returns spurious results on 128-bit CHERI.

Fixes #232
closed 2017-09-05 by brooksdavis