CTSRD-CHERI/cheribsd #235: Second argument to cheri_fd_write/read() is not passed correctly

RichardsonAlex, 2017-09-12:
We weren't noticing because the call to read/write passes `min(nbytes, cheri_getlen(buf_c) - cheri_getoffset(buf_c)` as the size argument so it actually works as expected.
However, when I added a check that buf_c contains at least nbytes in #234 these functions start failing.
For some reason they are receiving a virtual address to somewhere in the executables data segment in `a0` instead of the passed value of 128. I have not looked into the libcheri ccall details so I am not sure whether this is a problem with the CheriBSD code or whether the compiler is generating something incorrect.

arichardson, 2020-04-21:
I think this has been fixed a long time ago.

closed 2020-04-21 by arichardson

jrtc27, 2020-04-21:
This was probably another case of https://github.com/CTSRD-CHERI/llvm-project/commit/fd27e340b9c0a0bf193cedd8d41a88d1eda555fc