CTSRD-CHERI/cheribsd #238: printf crashes on pcc-derived capabilities

davidchisnall, 2017-10-09:
I'm not sure exactly what the root cause of this is, but in this test program:

```c

void caller(void);
__attribute__((noinline))
void callee(void)
{
    void *ret = __builtin_return_address(0);
    // This is fine:
    printf("0x%lx\n", __builtin_cheri_offset_get(ret));
    // This is fine:
    printf("0x%lx\n", __builtin_cheri_offset_get(caller));
    // This is also fine and prints a capability:
    printf("%#p\n", &x);
    // The next two both crash in the same way:
    printf("%#p\n", ret);
    printf("%#p\n", (void*)caller);
}
```

The crash appears to be in the line that prints the tag:

```
; *--cp = ujval ? '1' : '0';
1200a6d6c:  00 02 10 2b     sltu    $2, $zero, $2
1200a6d70:  34 42 00 30     ori $2, $2, 48
1200a6d74:  4a 61 17 f8     cincoffset  $c1, $c2, -8
1200a6d78:  e8 42 07 c0     csb $2, $zero, -8($c2) ; <-- Here
```

The value in $c2 has an offset of 7, so this is trying to write one before the length of the string.

I suspect that this might be caused by the change to `cgettype` returning -1 as the type for unsealed capabilities.  This means that the maximum length of a type is now 16 hex digits and so the value of [`buf`](https://github.com/CTSRD-CHERI/cheribsd/blob/nosp_csp/lib/libc/stdio/vfprintf.c#L308) `#define` is too small.

RichardsonAlex, 2017-10-09:
I assume we will also have to update the printf implementations in rtld, libthr, etc.

davidchisnall, 2017-10-09:
I'm not sure if I am correctly diagnosing the root cause, because printf of a null capability does work correctly, printing:
```
v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:ffffffffffffffff
```

As does a printf of the address of a global:

```
v:1 s:0 p:0007807d b:00000001200ff73c l:0000000000000004 o:0 t:ffffffffffffffff
```

So I'm not sure what is making pcc-derrived capabilities special.  Perhaps the permissions field is longer?

arichardson, 2018-04-09:
```
/* For the CHERI sandbox ABI we need enough space to print a capability dump */
#define BUF	84
```

Should hopefully be fixed by 06d1923d (printing -1 instead of fffffffffff)

`v:1 s:0 p:0007807d b:00000001200ff73c l:0000000000000004 o:0123456789abcdef t:ffffffffffffffff` has 94 chars but the buffer for printing is only 84
The longest possible string I can think of (`v:1 s:0 p:0007807d b:00000001200ff73c l:0000000000000004 o:0123456789abcdef t:-1`) is 80 bytes so will fit in that buffer.
I wonder if we should just bump the size to 128 to avoid issues like this in case we decide to print more information in the future?

closed 2018-04-09 by arichardson

brooksdavis, 2018-04-09:
Increasing to 128 seems reasonable.  We're not short on stack space in this context.