CTSRD-CHERI/cheribsd #239: memcpy failing to sufficiently align 256-bit caps

brooksdavis, 2017-10-19:
I've got a crash deep in the pkg command due to an unaligned `clc`.   It looks like the code failed to sufficiently align before trying capability operations.  The disassembly at the crash site is:

```
                t = length / psize;
                if (t) {
  1ea904:       2cc10020        sltiu   at,a2,32
  1ea908:       1420000a        bnez    at,1ea934 <memmove+0x264>
  1ea90c:       00000000        nop
/home/bed22/git/cheribsd/lib/libc/mips/string/jdw57_memcpy.c:253
                        src -= t*psize;
  1ea910:       6401ffe0        daddiu  at,zero,-32
  1ea914:       00c11024        and     v0,a2,at
  1ea918:       0002082f        dnegu   at,v0
  1ea91c:       49a21040        cincoffset      c2,c2,at
/home/bed22/git/cheribsd/lib/libc/mips/string/jdw57_memcpy.c:254
                        dst -= t*psize;
  1ea920:       49a10840        cincoffset      c1,c1,at
/home/bed22/git/cheribsd/lib/libc/mips/string/jdw57_memcpy.c:263
                            *((ptr * CAPABILITY)(src+t));, -8/*sizeof(ptr)*/);
#elif _MIPS_SZCAP==128
                        MIPSLOOP(t, 0, *((ptr * CAPABILITY)(dst+t)) =
                            *((ptr * CAPABILITY)(src+t));, -16/*sizeof(ptr)*/);
#elif _MIPS_SZCAP==256
                        MIPSLOOP(t, 0, *((ptr * CAPABILITY)(dst+t)) =
  1ea924:       6442ffe0        daddiu  v0,v0,-32
  1ea928:       d8621000        clc     c3,v0,0(c2)
  1ea92c:       1440fffd        bnez    v0,1ea924 <memmove+0x254>
  1ea930:       f8611000        csc     c3,v0,0(c1)
/home/bed22/git/cheribsd/lib/libc/mips/string/jdw57_memcpy.c:268
                            *((ptr * CAPABILITY)(src+t));, -32/*sizeof(ptr)*/);
#endif

                }
```

The error occurs at `0x1ea928` and the values of the relevant registers are:
```
$c01: v:1 s:0 p:7fff817d b:00000001614cf66c l:000000000000003c o:4 t:0
$c02: v:1 s:0 p:7fff817d b:00000001613ccc0c l:000000000000003c o:4 t:0          
v0: 0
```
It looks like more tests with different offsets are clearly required.

jonwoodruff, 2017-10-20:
It looks like this is the memmove (not memmove_c?), and the "copy backwards case".

brooksdavis, 2017-12-12:
Fixed in c917c058b5fe9ba314f4cbf2ae9eecb5d3826e72

closed 2017-12-12 by brooksdavis