CTSRD-CHERI/cheribsd #241: .cfi directives missing for system call stubs

RichardsonAlex, 2017-10-30:
GDB can't print backtraces when interrupted in system call stubs:

```
Program received signal SIGINT, Interrupt.
_wait4 () at _wait4.S:3
3	_wait4.S: No such file or directory.
(gdb) bt
#0  _wait4 () at _wait4.S:3
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)
```

bsdjhb, 2018-01-18:
I looked and there isn't much to add by way of CFI directives aside from startproc/endproc perhaps.  That said, I did have some patches lying around from when I was trying to debug stack traces that have a manual unwinder for CheriABI (for when DWARF info isn't present).  They aren't perfect, but using them does seem to permit unwinding out of system calls (though it usually dies before unwinding all the way back to main() for some reason).  I've pushed that change to the GDB 8.0.1 branch if you want to try it out.

arichardson, 2018-03-02:
Backtraces are working now.

closed 2018-03-02 by arichardson