CTSRD-CHERI/cheribsd #242: fabs(-0) returns -0

kgudka, 2017-11-13:
fabs(3) for MIPS is defined in lib/libc/mips/gen/fabs.c as:

double
fabs(double x)
{
  if (x < 0)
    x = -x; 
  return(x);
}

This obviously doesn't work if x is -0 because the if-test fails. One fix could be to make the if-test: (x < 0 || x == -0), which seems to work.

davidchisnall, 2017-11-13:
This looks like an upstream issue and is unlikely to matter for anything real.

arichardson, 2017-11-17:
We should probably just change the check to use `signbit()`. Looking at the Linux manpage for [fabs()](https://linux.die.net/man/3/fabs) it explicitly mentions that -0 returns +0. And the [signbit()](http://man7.org/linux/man-pages/man3/signbit.3.html) manpage states that this is not the same as `x < 0`. I can submit an upstream patch if we want to fix this.

arichardson, 2017-11-17:
Actually I just noticed clang supports a __builtin_fabs() which is probably better. It looks like glibc uses [__builtin_fabs()](https://code.woboq.org/userspace/glibc/sysdeps/x86_64/fpu/s_fabs.c.html#24) as well so possibly even the ancient GCC supports it.

brooksdavis, 2017-12-08:
I've cherry-picked @arichardson's fix in 1e84b653b2e858c3a9db9d2940fccf250c866cf0.

closed 2017-12-08 by brooksdavis