CTSRD-CHERI/cheribsd #245: CHERI BSD requires 19 bits of precision to run

jonwoodruff, 2017-12-11:
When trying to boot with CHERICC, where non-zero exponents steal the lower bits of the Top and Bottom, reducing precision to 17 bits.  During the bootstrap of jemalloc in mibench basicmatch, the kernel fails the below assertion.  Any binary I have tried fails in a similar way.

panic: cheri_capability_set: length 0xcc000000 rather than 0x7ecbe00000
cpuid = 0
time = 1511115046
KDB: enter: panic
[ thread pid 713 tid 100065 ]
Stopped at      kdb_enter+0x90: break   0

A few notes:
Looking at a trace, I believe the length being printed by the message is truncated to 32-bits and the actual length that was achieved was 0x7ecc00000, the correct rounded-up value for CHERIcc with 17 bits of precision.

The kernel should ideally allow for less precise allocations.  8-bit precision should be sufficient for any 128-bit format we have, and 4-bit precision should also work for any 64-bit format.  Is there any reason we can't align kernel allocations at 4-bit precision, for a maximum loss of 1/16th to fragmentation?

rwatson, 2017-12-11:
A patch such as the following (untested) should fix up the assertion's output; awaiting a toolchain rebuild here before I can test with it:
```
--- a/sys/mips/cheri/cheri.c
+++ b/sys/mips/cheri/cheri.c
@@ -157,14 +157,14 @@ cheri_capability_set(void * __capability *cp, uint32_t perms, vaddr_t basep,
 #ifdef INVARIANTS
        KASSERT(cheri_gettag(*cp) != 0, ("%s: capability untagged", __func__));
        KASSERT(cheri_getperm(*cp) == (register_t)perms,
-           ("%s: permissions 0x%x rather than 0x%x", __func__,
-           (unsigned int)cheri_getperm(*cp), perms));
+           ("%s: permissions 0x%lx rather than 0x%x", __func__,
+           (unsigned long)cheri_getperm(*cp), perms));
        KASSERT(cheri_getbase(*cp) == (register_t)basep,
            ("%s: base %p rather than %lx", __func__,
             (void *)cheri_getbase(*cp), basep));
        KASSERT(cheri_getlen(*cp) == (register_t)length,
-           ("%s: length 0x%x rather than %p", __func__,
-           (unsigned int)cheri_getlen(*cp), (void *)length));
+           ("%s: length 0x%lx rather than %p", __func__,
+           (unsigned long)cheri_getlen(*cp), (void *)length));
        KASSERT(cheri_getoffset(*cp) == (register_t)off,
            ("%s: offset %p rather than %p", __func__,
            (void *)cheri_getoffset(*cp), (void *)off));
```

The kernel provides precise bounds for user space memory mappings where possible (which it looks like this is), and assertions are used to detect kernel bugs and enforce invariants. The kernel's alignment logic during allocation likely simply needs some updating to reflect the new precision rules. The kernel could be changed to tolerate less precise bounds, but in order to provide spatial safety on mappings, would need to over-allocate mappings in those situations, which is something we are exploring via the new cmmap(2) API, which differentiates "reservations" from specific mappings, but this work is not yet complete. I.e., this is not just an issue of lack fragmentation but also of safety -- and the historic mmap(2) API is not able to differentiate those ideas because it knows only that it allocates what you ask for (or doesn't), rather than that an over-allocation may take place.

(Tagging @brooksdavis as an FYI.)

brooksdavis, 2017-12-12:
It would be good to know where this was triggered.  What do you get if you type `bt` at the panic prompt?

If it is mmap, then there is a bug in that mmap should fail on invalid mappings (if you allow less precision, you must modify `CHERI_BASELEN_BITS` in `sys/mips/include/cherireg.h` accordingly).  There may also be a jemalloc bug in that it isn't yet aware of the need to round up large, odd-sized allocations.

arichardson, 2019-06-13:
See https://github.com/CTSRD-CHERI/cheribsd/pull/310

brooksdavis, 2020-04-30:
This seems to be sufficiently resolved.

closed 2020-04-30 by brooksdavis