CTSRD-CHERI/cheribsd #249: Jemalloc assertion when running GDB

arichardson, 2018-01-16:
```
root@qemu-cheri256-alr48:~ # gdb-mips -ex=r --args /root/tst_qtendian
uchar *in;GNU gdb (GDB) 8.0.1
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "mips64-unknown-freebsd".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /root/tst_qtendian...<jemalloc>: jemalloc_arena.c:821: Failed assertion: "nstime_compare(&decay->epoch, &time) <= 0"
```

This sounds like time is going backwards again: 
Probably related to 
https://github.com/CTSRD-CHERI/qemu/issues/44
https://github.com/CTSRD-CHERI/qemu/issues/54
https://github.com/CTSRD-CHERI/qemu/issues/55

arichardson, 2018-01-19:
Pinning QEMU to a single CPU didn't help here so it is not related to QEMU using different counts across CPUs.

I think the issue here might be that if I understand sys/mips/tick.c correctly the timer is only 32 bits since all it does is read the low 32 bits of the count register.
This suggest we should se a wraparound after `0xffffffff / (50× 1000 × 1000) = 85.5 seconds`. Probably because of https://github.com/CTSRD-CHERI/qemu/issues/55 it actually happens after about 1 minute and 9 seconds.

arichardson, 2018-04-09:
Closing this since with workaround I added to QEMU (https://github.com/CTSRD-CHERI/qemu/commit/fd6f754bc5a51bcfdd752ca33426a6c963366180) I can now run gdb without this assertion triggering.

arichardson, 2018-04-26:
Seems like the workaround is not sufficient: Just saw this when running the testsuite (in 
lib.libc.db.db_test.btree_tricky_page_split):
```
Assertion failed: (nstime_compare(&decay->epoch, &time) <= 0), function arena_maybe_decay, file jemalloc_arena.c, line 821.
```

brooksdavis, 2018-07-27:
Worked around in 52ec1852d716d5639425cd7ed3112e2db9079818.

closed 2018-07-27 by brooksdavis

trasz, 2020-09-16:
FWIW, it's breaking MIPS runs on FreeBSD CI: https://ci.freebsd.org/job/FreeBSD-head-mips64-test/14/console.  Would it make sense to upstream it, or perhaps add as a port-local patch?

jrtc27, 2020-09-16:
Or we fix CLOCK_MONOTONIC to not go backwards? Having QEMU's time go backwards is one thing, but having CLOCK_MONOTONIC go backwards as a result is another and a true FreeBSD bug.