CTSRD-CHERI/cheribsd #253: _get_curthread() returns NULL during crt_call_constructors

arichardson, 2018-03-21:
While debugging some crashes in Qt I ran into the following error:

```
#0  0x0000000120b23030 in testcancel (curthread=0x0) at /exports/users/alr48/sources/cheribsd/lib/libthr/thread/thr_cancel.c:44         
#1  _pthread_testcancel () at /exports/users/alr48/sources/cheribsd/lib/libthr/thread/thr_cancel.c:135
#2  0x0000000120b2a0b4 in _sem_clockwait_np (sem=0x161f2001c, clock_id=0, flags=1, rqtp=0x0, rmtp=0x0) at cancelpoints_sem_new.c:401
#3  0x0000000120b2a384 in _sem_timedwait (sem=0x161f2001c, abstime=0x0) at cancelpoints_sem_new.c:436
#4  _sem_wait (sem=0x161f2001c) at cancelpoints_sem_new.c:444                    
#5  0x000000012027bc7c in QMutexPrivate::wait (this=<optimized out>, timeout=<optimized out>)
    at /exports/users/alr48/sources/qt5/qtbase/src/corelib/thread/qmutex_unix.cpp:84
#6  0x000000012027ace0 in QBasicMutex::lockInternal (this=<optimized out>, timeout=<optimized out>)
    at /exports/users/alr48/sources/qt5/qtbase/src/corelib/thread/qmutex.cpp:575 
#7  0x000000012027a35c in QBasicMutex::lockInternal (this=<optimized out>)       
    at /exports/users/alr48/sources/qt5/qtbase/src/corelib/thread/qmutex.cpp:491 
#8  0x000000012027a200 in QMutex::lock (this=<optimized out>) at /exports/users/alr48/sources/qt5/qtbase/src/corelib/thread/qmutex.cpp:229
#9  0x00000001201f6570 in QMutexLocker::QMutexLocker(QBasicMutex* cap) (this=<optimized out>, m=<optimized out>)
    at ../../include/QtCore/../../../../../../../exports/users/alr48/sources/qt5/qtbase/src/corelib/thread/qmutex.h:206
#10 0x00000001205e5774 in (anonymous namespace)::Q_QGS_qtDefaultCategory::innerFunction ()
    at /exports/users/alr48/sources/qt5/qtbase/src/corelib/io/qloggingcategory.cpp:47
#11 0x00000001205e54c4 in QGlobalStatic<QLoggingCategory, (anonymous namespace)::Q_QGS_qtDefaultCategory::innerFunction(), (anonymous namespace)::Q_QGS_qtDefaultCategory::guard>::operator()() (this=<optimized out>)
    at ../../include/QtCore/../../../../../../../exports/users/alr48/sources/qt5/qtbase/src/corelib/global/qglobalstatic.h:139
#12 0x00000001205e53e4 in QLoggingCategory::defaultCategory ()                   
    at /exports/users/alr48/sources/qt5/qtbase/src/corelib/io/qloggingcategory.cpp:390
#13 0x000000012026dc58 in qt_message_print(QtMsgType, QMessageLogContext const& cap, QString const& cap) (msgType=<optimized out>, context=...,
    message=...) at /exports/users/alr48/sources/qt5/qtbase/src/corelib/global/qlogging.cpp:1663
#14 0x000000012025fb84 in qt_message(QtMsgType, QMessageLogContext const& cap, char const* cap, void* cap) (msgType=<optimized out>,
    context=..., msg=<optimized out>, ap=<optimized out>) at /exports/users/alr48/sources/qt5/qtbase/src/corelib/global/qlogging.cpp:301
#15 0x0000000120263db0 in QMessageLogger::fatal(char const* cap, ...) const (this=<optimized out>, msg=<optimized out>)
    at /exports/users/alr48/sources/qt5/qtbase/src/corelib/global/qlogging.cpp:813
#16 0x000000012024c240 in qt_assert(char const* cap, char const*, int) (assertion=<optimized out>, file=<optimized out>, line=<optimized out>)
    at /exports/users/alr48/sources/qt5/qtbase/src/corelib/global/qglobal.cpp:3152
#17 0x00000001201f6ba8 in qClearLowPointerBits (ptr=<optimized out>, negatedmask=<optimized out>)
    at ../../include/QtCore/../../../../../../../exports/users/alr48/sources/qt5/qtbase/src/corelib/global/qglobal.h:1191
#18 0x00000001201f68dc in QMutexLocker::unlock (this=<optimized out>)            
    at ../../include/QtCore/../../../../../../../exports/users/alr48/sources/qt5/qtbase/src/corelib/thread/qmutex.h:218
#19 0x00000001201f66c0 in QMutexLocker::~QMutexLocker (this=<optimized out>)     
    at ../../include/QtCore/../../../../../../../exports/users/alr48/sources/qt5/qtbase/src/corelib/thread/qmutex.h:213
#20 0x00000001205ed9a4 in (anonymous namespace)::Q_QGS_qtLoggingRegistry::innerFunction ()
    at /exports/users/alr48/sources/qt5/qtbase/src/corelib/io/qloggingregistry.cpp:56
#21 0x00000001205ed584 in QGlobalStatic<QLoggingRegistry, (anonymous namespace)::Q_QGS_qtLoggingRegistry::innerFunction(), (anonymous namespace)::Q_QGS_qtLoggingRegistry::guard>::operator()() (this=<optimized out>)
    at ../../include/QtCore/../../../../../../../exports/users/alr48/sources/qt5/qtbase/src/corelib/global/qglobalstatic.h:139
#22 0x00000001205ed4a4 in QLoggingRegistry::instance () at /exports/users/alr48/sources/qt5/qtbase/src/corelib/io/qloggingregistry.cpp:408
#23 0x00000001205e4cf4 in QLoggingCategory::init(char const* cap, QtMsgType) (this=<optimized out>, category=<optimized out>,
    severityLevel=<optimized out>) at /exports/users/alr48/sources/qt5/qtbase/src/corelib/io/qloggingcategory.cpp:251
#24 0x00000001205e4ba0 in QLoggingCategory::QLoggingCategory(char const* cap) (this=<optimized out>, category=<optimized out>)
    at /exports/users/alr48/sources/qt5/qtbase/src/corelib/io/qloggingcategory.cpp:224
#25 0x00000001205e583c in (anonymous namespace)::Q_QGS_qtDefaultCategory::innerFunction ()
    at /exports/users/alr48/sources/qt5/qtbase/src/corelib/io/qloggingcategory.cpp:47
#26 0x00000001205e54c4 in QGlobalStatic<QLoggingCategory, (anonymous namespace)::Q_QGS_qtDefaultCategory::innerFunction(), (anonymous namespace)::Q_QGS_qtDefaultCategory::guard>::operator()() (this=<optimized out>)
    at ../../include/QtCore/../../../../../../../exports/users/alr48/sources/qt5/qtbase/src/corelib/global/qglobalstatic.h:139
#27 0x00000001205e53e4 in QLoggingCategory::defaultCategory ()                   
    at /exports/users/alr48/sources/qt5/qtbase/src/corelib/io/qloggingcategory.cpp:390
#28 0x000000012026dc58 in qt_message_print(QtMsgType, QMessageLogContext const& cap, QString const& cap) (msgType=<optimized out>, context=...,
    message=...) at /exports/users/alr48/sources/qt5/qtbase/src/corelib/global/qlogging.cpp:1663
#29 0x000000012025fb84 in qt_message(QtMsgType, QMessageLogContext const& cap, char const* cap, void* cap) (msgType=<optimized out>,
    context=..., msg=<optimized out>, ap=<optimized out>) at /exports/users/alr48/sources/qt5/qtbase/src/corelib/global/qlogging.cpp:301
#30 0x0000000120263db0 in QMessageLogger::fatal(char const* cap, ...) const (this=<optimized out>, msg=<optimized out>)
    at /exports/users/alr48/sources/qt5/qtbase/src/corelib/global/qlogging.cpp:813
#31 0x000000012024c240 in qt_assert(char const* cap, char const*, int) (assertion=<optimized out>, file=<optimized out>, line=<optimized out>)
    at /exports/users/alr48/sources/qt5/qtbase/src/corelib/global/qglobal.cpp:3152
#32 0x00000001201f6ba8 in qClearLowPointerBits (ptr=<optimized out>, negatedmask=<optimized out>)
    at ../../include/QtCore/../../../../../../../exports/users/alr48/sources/qt5/qtbase/src/corelib/global/qglobal.h:1191
#33 0x00000001201f68dc in QMutexLocker::unlock (this=<optimized out>)            
    at ../../include/QtCore/../../../../../../../exports/users/alr48/sources/qt5/qtbase/src/corelib/thread/qmutex.h:218
#34 0x00000001201f66c0 in QMutexLocker::~QMutexLocker (this=<optimized out>)     
    at ../../include/QtCore/../../../../../../../exports/users/alr48/sources/qt5/qtbase/src/corelib/thread/qmutex.h:213
#35 0x000000012024d520 in qEnvironmentVariableIntValue(char const* cap, bool* cap) (varName=<optimized out>, ok=<optimized out>)
    at /exports/users/alr48/sources/qt5/qtbase/src/corelib/global/qglobal.cpp:3408
#36 0x00000001201eaf98 in disableCoreDump () at /exports/users/alr48/sources/qt5/qtbase/src/testlib/qtestcase.cpp:188
---Type <return> to continue, or q <return> to quit---                           
#37 0x00000001201ced7c in (anonymous namespace)::disableCoreDump_ctor_class_::disableCoreDump_ctor_class_ (this=<optimized out>)
    at /exports/users/alr48/sources/qt5/qtbase/src/testlib/qtestcase.cpp:199     
#38 0x00000001201ced04 in __cxx_global_var_init () at /exports/users/alr48/sources/qt5/qtbase/src/testlib/qtestcase.cpp:199
#39 0x00000001201eb3d8 in _GLOBAL__sub_I_qtestcase.cpp ()                        
#40 0x00000001201c06f4 in crt_call_constructors () at /exports/users/alr48/sources/cheribsd/lib/csu/cheri/crtbeginC.c:107
#41 0x00000001201c02dc in __start (auxv=<optimized out>, cleanup=<optimized out>, obj=<optimized out>)
    at /exports/users/alr48/sources/cheribsd/lib/csu/cheri/crt1.c:163            
```

_get_curthread() looks like this:
```
static __inline struct pthread *
_get_curthread(void)
{
	if (_thr_initial)
		return (_tcb_get()->tcb_thread);
	return (NULL);
}
```

I think the fix for this should be having the sem_ functions call `_libpthread_init()` before using `_pthread_cancel()`

arichardson, 2018-03-21:
It seems like `_libpthread_init()` gets called using a constructor function:
`void	_thread_init_hack(void) __attribute__ ((constructor));` but apparently the C++ ctors run first.

bsdjhb, 2018-03-21:
On Wednesday, March 21, 2018 04:35:11 PM Alexander Richardson wrote:
> It seems like `_libpthread_init()` gets called using a constructor function:
> `void	_thread_init_hack(void) __attribute__ ((constructor));` but apparently the C++ ctors run first.

The pattern is that various thread library routines call _thr_check_init()
which is a wrapper around _libpthread_init().  (e.g. pthread_mutex_lock/unlock
call this).  These checks are all contained in libthr rather than in libc, so
I think what we should do is call _thr_check_init() in pthread_cancel().

I'll work on a test case for this along with a fix.  sem_open() uses a pthread
mutex, so only sem_init() semaphores can trip over this.

-- 
John Baldwin

commit c201a726 (2018-03-21): Invoke _thr_check_init() in pthread_testcancel() before reading curthread.

Fixes #253.
closed 2018-03-21 by bsdjhb