CTSRD-CHERI/cheribsd #255: __umtx_op_nwake_private is unimplemented for CHERIABI

arichardson, 2018-04-09:
The following is the ktrace log of running a QT test binary:

```
 2121 tst_qtendian CALL  mmap(0,0x11000,0x3<PROT_READ|PROT_WRITE>,0x1040<MAP_ANON|MAP_CHERI_DDC>,0xffffffff,0)                          
  2121 tst_qtendian RET   mmap 45056/0xb000                                      
  2121 tst_qtendian CALL  mmap(0,0x800000,0x7<PROT_READ|PROT_WRITE|PROT_EXEC>,0x400<MAP_STACK>,0xffffffff,0)
  2121 tst_qtendian RET   mmap 5937651712/0x161e96000                            
  2121 tst_qtendian CALL  thr_new(0x7fffff9f80,0x1a0)                            
  2121 tst_qtendian RET   thr_new 0                                              
  2121 tst_qtendian CALL  _umtx_op(0x121334ea0,UMTX_OP_WAIT_UINT_PRIVATE,0,0,0)  
  2121 tst_qtendian RET   fork 0                                                 
  2121 tst_qtendian CALL  mmap(0,0x4000,0x3<PROT_READ|PROT_WRITE>,0x1002<MAP_PRIVATE|MAP_ANON>,0xffffffff,0)
  2121 tst_qtendian RET   mmap 5937651712/0x161e96000                            
  2121 tst_qtendian CALL  mmap(0,0x1e000,0x3<PROT_READ|PROT_WRITE>,0xd001002<MAP_PRIVATE|MAP_ANON|MAP_ALIGNED(13)>,0xffffffff,0)
  2121 tst_qtendian RET   mmap 5937668096/0x161e9a000                            
  2121 tst_qtendian CALL  mmap(0,0x2000,0x3<PROT_READ|PROT_WRITE>,0xd001002<MAP_PRIVATE|MAP_ANON|MAP_ALIGNED(13)>,0xffffffff,0)
  2121 tst_qtendian RET   mmap 5937790976/0x161eb8000                            
  2121 tst_qtendian CALL  mmap(0,0x6000,0x3<PROT_READ|PROT_WRITE>,0xd001002<MAP_PRIVATE|MAP_ANON|MAP_ALIGNED(13)>,0xffffffff,0)
  2121 tst_qtendian RET   mmap 5937799168/0x161eba000                            
  2121 tst_qtendian CALL  pipe2(0x161eba1e0,0x100004)                            
  2121 tst_qtendian RET   pipe2 0                                                
  2121 tst_qtendian CALL  _umtx_op(0x161df6800,UMTX_OP_NWAKE_PRIVATE,0x1,0,0)    
  2121 tst_qtendian RET   _umtx_op -1 errno 45 Operation not supported           
  2121 tst_qtendian CALL  _umtx_op(0x160ba1020,UMTX_OP_WAIT_UINT_PRIVATE,0,0,0)  
  2121 tst_qtendian RET   _umtx_op -1 errno 4 Interrupted system call            
  2121 tst_qtendian PSIG  SIGINT caught handler=0x120a34dd8 mask=0x0 code=SI_KERNEL
  2121 tst_qtendian CALL  write(0x2,0x120bdf17c,0x1)                             
  2121 tst_qtendian GIO   fd 2 wrote 1 byte        
```
Becaues NWAKE_PRIVATE is unimplemented this deadlocks. Qt doesn't use umtx directly on FreeBSD so this is due to libthr.

arichardson, 2018-04-09:
Backtrace for the UMTX_OP_NWAKE_PRIVATE call:
```
(gdb) bt                        
#0  _thr_wake_all (waddrs=0x161df6800, count=1) at /exports/users/alr48/sources/cheribsd/lib/libthr/thread/thr_kern.c:210          
#1  0x0000000120a399bc in mutex_unlock_common (m=<optimized out>, cv=false, mtx_defer=0x0)                                         
    at /exports/users/alr48/sources/cheribsd/lib/libthr/thread/thr_mutex.c:981                                                     
#2  0x0000000120a3969c in _pthread_mutex_unlock (mutex=<optimized out>)                                                            
    at /exports/users/alr48/sources/cheribsd/lib/libthr/thread/thr_mutex.c:775                                                     
#3  0x0000000120280b90 in QWaitCondition::wakeAll (this=<optimized out>)                                                           
    at /exports/users/alr48/sources/qt5/qtbase/src/corelib/thread/qwaitcondition_unix.cpp:199                                      
:#4  0x00000001201f1170 in QTest::WatchDog::run (this=<optimized out>)                                                                  
    at /exports/users/alr48/sources/qt5/qtbase/src/testlib/qtestcase.cpp:988                                                            
#5  0x0000000120276b74 in QThreadPrivate::start(void* cap) (arg=<optimized out>)                                                        
    at /exports/users/alr48/sources/qt5/qtbase/src/corelib/thread/qthread_unix.cpp:376                                                  
#6  0x0000000120a455a8 in thread_start (curthread=<optimized out>)  
    at /exports/users/alr48/sources/cheribsd/lib/libthr/thread/thr_create.c:295                                                         
#7  0x0000000000000000 in ?? ()                                  
```

brooksdavis, 2018-05-29:
This should have been fixed in 798b4b7d4999ee9ba697557c6a465bf349b55c4d.  If it's broken there is a separate issue.

closed 2018-05-29 by brooksdavis