CTSRD-CHERI/cheribsd #260: Crash during jemalloc thread shutdown

arichardson, 2018-05-01:
I am getting the following crash on thread exit (if it has called malloc):
```
Trapframe Register Dump:
$0: 0                  at: 0x1                v0: 0x140              v1: 0x8
a0: 0x400              a1: 0x162f6efa0        a2: 0                  a3: 0
a4: 0x12c0             a5: 0x1                a6: 0                  a7: 0
t0: 0x40               t1: 0                  t2: 0                  t3: 0
s0: 0x161d60328        s1: 0x140000000        s2: 0x5                s3: 0x140
s4: 0x600              s5: 0                  s6: 0                  s7: 0
t8: 0                  t9: 0x78               k0: 0                  k1: 0
gp: 0                  sp: 0                  s8: 0                  ra: 0
status: 0x408084b3 mullo: 0x4; mulhi: 0; badvaddr: 0x120a2a4a0
cause: 0x48; pc: 0x120a2a4a0
CHERI cause: ExcCode: 0x02 RegNum: $c01 (tag violation)
$c00: v:1 s:0 p:7fff817d b:0000000000000000 l:00000001212e9000 o:0 t:-1
$c01: v:0 s:0 p:00000000 b:0000000000000000 l:ffffffffffffffff o:0 t:-1
$c02: v:0 s:1 p:00000000 b:0000000000000000 l:ffffffffffffffff o:ffffffffffffffa0 t:0
$c03: v:1 s:0 p:7fff817d b:0000000161360000 l:0000000000a00000 o:43ae00 t:-1
$c04: v:0 s:0 p:00000000 b:0000000000000000 l:ffffffffffffffff o:0 t:-1
$c05: v:0 s:1 p:00000000 b:0000000161e32000 l:0000000000800000 o:7fe080 t:0
$c06: v:1 s:0 p:7fff817f b:000000016262ff74 l:0000000000000004 o:0 t:-1
$c07: v:1 s:0 p:7fff817f b:000000016262ff70 l:0000000000000004 o:0 t:-1
$c08: v:1 s:0 p:7fff817f b:000000016262ff40 l:0000000000000020 o:0 t:-1
$c09: v:1 s:0 p:7fff817f b:000000016262ff20 l:0000000000000020 o:0 t:-1
$c10: v:1 s:0 p:7ffe8117 b:0000000000000000 l:00000001212e9000 o:120bcd1a0 t:-1
$c11: v:1 s:0 p:7fff817f b:0000000161e32000 l:0000000000800000 o:7fcfc0 t:-1
$c12: v:1 s:0 p:7ffe8117 b:0000000000000000 l:00000001212e9000 o:120b3bc00 t:-1
$c13: v:0 s:0 p:00000000 b:0000000000000000 l:ffffffffffffffff o:0 t:-1
$c14: v:1 s:0 p:7ffe8117 b:0000000000000000 l:00000001212e9000 o:1209b3b5c t:-1
$c15: v:1 s:0 p:7fff817f b:0000000162630a78 l:0000000000000004 o:0 t:-1
$c16: v:1 s:0 p:7fff817f b:0000000162630a40 l:0000000000000020 o:0 t:-1
$c17: v:1 s:0 p:7ffe8117 b:0000000000000000 l:00000001212e9000 o:120a2adbc t:-1
$c18: v:1 s:0 p:7fff817f b:000000016262fc60 l:0000000000000600 o:0 t:-1
$c19: v:1 s:0 p:7fff817d b:0000000160b42000 l:0000000000200000 o:20c0 t:-1
$c20: v:0 s:0 p:00000000 b:0000000000000000 l:ffffffffffffffff o:0 t:-1
$c21: v:1 s:0 p:7ffe8117 b:0000000000000000 l:00000001212e9000 o:120bcd1a0 t:-1
$c22: v:1 s:0 p:7fff817f b:00000001626304dc l:0000000000000002 o:0 t:-1
$c23: v:1 s:0 p:7fff817f b:0000000162630be0 l:0000000000000120 o:0 t:-1
$c24: v:1 s:0 p:7fff817f b:0000000161e32000 l:0000000000800000 o:7fcfc0 t:-1
$c26: v:1 s:0 p:7ffe8117 b:0000000000000000 l:00000001212e9000 o:120bcd1a0 t:-1
$c31: v:1 s:0 p:7ffe8117 b:0000000000000000 l:00000001212e9000 o:120a2a4a0 t:-1
Jan  2 06:52:47 qemu-cheri256-alr48 kernel: USER_CHERI_EXCEPTION: pid 695 tid 100062 (tst_qtendian), uid 0: CP2 fault (type 0x32)
```
Backtrace:
```
#0  0x0000000120a2a4a0 in rtree_leaf_elm_lookup (tsdn=0x0, rtree=<optimized out>, rtree_ctx=<optimized out>,
    dependent=<error reading variable: Cannot access memory at address 0xffffffffffffffff>,
    init_missing=<error reading variable: Cannot access memory at address 0x0>, key=<optimized out>)
    at /exports/users/alr48/sources/cheribsd/contrib/jemalloc/include/jemalloc/internal/rtree.h:327
#1  rtree_read (dependent=<error reading variable: Cannot access memory at address 0xffffffffffffffff>, tsdn=<optimized out>,
    rtree=<optimized out>, rtree_ctx=<optimized out>, key=<optimized out>)
    at /exports/users/alr48/sources/cheribsd/contrib/jemalloc/include/jemalloc/internal/rtree.h:396
#2  rtree_extent_read (dependent=<error reading variable: Cannot access memory at address 0xffffffffffffffff>, tsdn=<optimized out>,
    rtree=<optimized out>, rtree_ctx=<optimized out>, key=<optimized out>)
    at /exports/users/alr48/sources/cheribsd/contrib/jemalloc/include/jemalloc/internal/rtree.h:408
#3  unbound_ptr (tsdn=<optimized out>, ptr=<optimized out>)
    at /exports/users/alr48/sources/cheribsd/contrib/jemalloc/include/jemalloc/internal/jemalloc_internal_inlines_c.h:19
#4  idalloctm (ptr=<optimized out>, is_internal=<optimized out>, slow_path=255, tsdn=<optimized out>, tcache=<optimized out>,
    alloc_ctx=<optimized out>)
    at /exports/users/alr48/sources/cheribsd/contrib/jemalloc/include/jemalloc/internal/jemalloc_internal_inlines_c.h:120
#5  a0idalloc (ptr=<optimized out>, is_internal=<optimized out>) at jemalloc_jemalloc.c:259
#6  0x0000000120a2c150 in __je_a0dalloc (ptr=0x0) at jemalloc_jemalloc.c:269
#7  __je_arenas_tdata_cleanup (tsd=0x16179ae00) at jemalloc_jemalloc.c:647
#8  0x00000001209e2eec in tsd_do_data_cleanup (tsd=<optimized out>) at jemalloc_tsd.c:187
#9  __je_tsd_cleanup (arg=<optimized out>) at jemalloc_tsd.c:210
#10 0x00000001209e32a0 in tsd_cleanup_wrapper ()
    at /exports/users/alr48/sources/cheribsd/contrib/jemalloc/include/jemalloc/internal/tsd_malloc_thread_cleanup.h:15
#11 0x00000001209e2d58 in _malloc_thread_cleanup () at jemalloc_tsd.c:165
#12 0x00000001209cca48 in exit_thread () at /exports/users/alr48/sources/cheribsd/lib/libthr/thread/thr_exit.c:334
#13 0x00000001209ccde8 in thread_unwind_stop (version=<optimized out>, actions=<optimized out>, exc_class=<optimized out>,
    exc_obj=<optimized out>, context=<optimized out>, stop_parameter=<optimized out>)
    at /exports/users/alr48/sources/cheribsd/lib/libthr/thread/thr_exit.c:197
#14 0x00000001209aab34 in unwind_phase2_forced (uc=<optimized out>, cursor=0x162630ea0, exception_object=0x161d96660,
    stop=0x1209cccc8 <thread_unwind_stop>, stop_parameter=0x0)
    at /exports/users/alr48/sources/cheribsd/contrib/cheri-libunwind/src/UnwindLevel1.c:274
#15 0x00000001209aaed8 in _Unwind_ForcedUnwind (exception_object=<optimized out>, stop=<optimized out>, stop_parameter=<optimized out>)
    at /exports/users/alr48/sources/cheribsd/contrib/cheri-libunwind/src/UnwindLevel1.c:416
#16 0x00000001209cc944 in thread_unwind () at /exports/users/alr48/sources/cheribsd/lib/libthr/thread/thr_exit.c:210
#17 0x00000001209cc884 in _pthread_exit_mask (status=0x0, mask=0x0)
    at /exports/users/alr48/sources/cheribsd/lib/libthr/thread/thr_exit.c:292
#18 0x00000001209cc6d4 in _pthread_exit (status=0x16179ae00) at /exports/users/alr48/sources/cheribsd/lib/libthr/thread/thr_exit.c:244
#19 0x00000001209c8fd4 in thread_start (curthread=<optimized out>)
    at /exports/users/alr48/sources/cheribsd/lib/libthr/thread/thr_create.c:295
#20 0x0000000000000000 in ?? ()
```

It seems like we probably never ran into this before because we weren't calling thread destructors?

I added some debug information to the individual jemalloc calls but I am quite unfamilar with the code so I am not sure what unbound_ptr should be doing in this case.
```
tsd_cleanup_wrapper: v:1 s:0 p:7fff817d b:0000000000000000 l:00000001205bb000 o:b0a0 t:-1
__je_arenas_tdata_cleanup: v:1 s:0 p:7fff817d b:0000000000000000 l:00000001205bb000 o:b0a0 t:-1
__je_arenas_tdata_cleanup: arenas_tdata = v:1 s:0 p:7fff817d b:0000000161336000 l:0000000000002000 o:8 t:-1
__je_a0dalloc: v:1 s:0 p:7fff817d b:0000000161336000 l:0000000000002000 o:8 t:-1
a0idalloc: v:1 s:0 p:7fff817d b:0000000161336000 l:0000000000002000 o:8 t:-1
unbound_ptr(v:0 s:0 p:00000000 b:0000000000000000 l:ffffffffffffffff o:0 t:-1, v:1 s:0 p:7fff817d b:0000000161336000 l:0000000000002000 o:8 t:-1)
```

brooksdavis, 2018-05-01:
The issue is this code from `unbound_ptr()`:
```
        rtree_ctx = tsd_rtree_ctx(tsdn_tsd(tsdn));
        extent = rtree_extent_read(tsdn, &extents_rtree,
            rtree_ctx, (vaddr_t)ptr, true);
```
This fails to handle the case that tsdn==NULL.  I don't entirely understand how it gets as far as `rtree_extent_read()` because `tsdn_tsd()` dereferences the NULL pointer.  We need to figure out how to find the correct TSDN in `a0idalloc` or find another way to locate the extent associated with `arenas_tdata`.

arichardson, 2018-05-01:
If I'm reading the source correctly it seems to me like arenas_tdata_cleanup (jemalloc.c:645) sets tsd to NULL before calling a0dalloc so maybe just moving that after the dealloc works?

arichardson, 2018-05-01:
I just tried moving the setting to NULL below and it still faults