CTSRD-CHERI/cheribsd #263: Use of cheri_clear_low_ptr_bits breaks non-MIPS build

brooksdavis, 2018-05-04:
Building on amd64 fails due to missing cheri includes due to the include of cheri/cheric.h to get cheri_clear_low_ptr_bits().  These should probably be moved to a non-cheri header file.

brooksdavis, 2020-04-21:
Do we still need these with address mode?

jrtc27, 2020-04-21:
Aren't they mainly about provenance for binary operators? With the recently-added compiler analysis maybe they can go away.

arichardson, 2020-04-21:
Now that we infer the provenance source and more importantly no longer include the tag bit in comparisons they are no longer needed. 

We can revisit this if we ever want to make exact equals the default.

brooksdavis, 2020-04-21:
Reopening to track removal of these macros.

brooksdavis, 2020-08-11:
We've long since fixed the macros.

closed 2020-08-11 by brooksdavis