CTSRD-CHERI/cheribsd #268: mount_smbfs broken after latest merge

arichardson, 2018-06-29:
I used to be able to use mount_smbfs to get the QEMU smb mount, but with the latest kernel I am getting the following error:
```
mount_smbfs -I 10.0.2.4 -N //10.0.2.4/qemu /mnt
KDB: stack backtrace:
db_trace_self+0x18 (?,?,?,?) ra ffffffff801e7330 sp c000000000317260 sz 16
db_fetch_ksymtab+0x238 (?,?,?,?) ra ffffffff8044319c sp c000000000317270 sz 800
kdb_backtrace+0x7c (?,?,?,?) ra ffffffff804675cc sp c000000000317590 sz 16
copyin+0x104 (?,?,?,?) ra ffffffff80524710 sp c0000000003175a0 sz 64
iconv_vfs_refcount+0x878 (?,?,?,?) ra ffffffff803f80d0 sp c0000000003175e0 sz 208
sbuf_new_for_sysctl+0x290 (?,?,?,?) ra ffffffff803f72a8 sp c0000000003176b0 sz 64
kernel_sysctl+0x5c0 (?,?,?,?) ra 0 sp c0000000003176f0 sz 0
mount_smbfs: can not setup kernel iconv table (ISO8859-1:tolower): syserr = Memory protection violation
```

It seems to be working with a MIPS mount_smbfs, only CheriABI appears to crash

brooksdavis, 2018-06-29:
That code has been broken for months.  `iconv_sysctl_add()` relies on the ability to copy pointers in via sysctl (not currently supported), copies in the wrong struct (the one with integer pointers), and then tries to use the high bits of a capability to copy via an implicit capability which is disallowed even if they were a valid virtual address.

brooksdavis, 2018-06-30:
After fixing `kern.iconv.add` I'm now getting failures in:

```
   631 mount_smbfs CALL  ioctl(0x3,SMBIOC_LOOKUP,0)                             
   631 mount_smbfs RET   ioctl -1 errno 19 Operation not supported by device    
```
`SMBIOC_LOOKUP` will require significant translation to fix.

arichardson, 2018-06-30:
Maybe we should just add `WANT_CHERI=none` for now? I'm not sure it makes sense to spend lots of time on fixing the smb ioctls right now.

brooksdavis, 2018-06-30:
I think that's fine

PR #282 (2018-08-21): Make smb use capabilities for IO.
This makes mount_smbfs work as a CheriABI binary at the cost of
breaking mips64 binaries.  I'll work 32- and 64-bit compat later.

Should fix #275 and half fixes #268.
closed 2018-08-21 by arichardson