CTSRD-CHERI/cheribsd #275: fgets on a smbfs file returns with errno 97

nwf, 2018-08-03:
At least it gives a stack trace, but this is something that doesn't seem like it should be misbehaving.

```
db_trace_self+0x18 (?,?,?,?) ra ffffffff801e7800 sp c000000000397000 sz 16
db_fetch_ksymtab+0x238 (?,?,?,?) ra ffffffff80443c9c sp c000000000397010 sz 800
kdb_backtrace+0x7c (?,?,?,?) ra ffffffff80468364 sp c000000000397330 sz 16
copyout+0x104 (?,?,?,?) ra ffffffff80447c8c sp c000000000397340 sz 64
md_get_mem+0x15c (?,?,?,?) ra ffffffff80448c50 sp c000000000397380 sz 80
md_get_uio+0x120 (?,?,?,?) ra ffffffff805ce008 sp c0000000003973d0 sz 96
smb_read+0x450 (?,?,?,?) ra ffffffff802d536c sp c000000000397430 sz 192
smbfs_readvnode+0x36c (?,?,?,?) ra ffffffff802dcf7c sp c0000000003974f0 sz 496
smbfs_free_scred+0x267c (?,?,?,?) ra ffffffff806ed0f4 sp c0000000003976e0 sz 16
VOP_READ_APV+0x10c (?,?,?,?) ra ffffffff8052120c sp c0000000003976f0 sz 48
vn_fsid+0x394 (?,?,?,?) ra ffffffff8051bbd0 sp c000000000397720 sz 128
sys_posix_fadvise+0x1e8 (?,?,?,?) ra ffffffff804741f8 sp c0000000003977a0 sz 144
sys_readv+0x140 (?,?,?,?) ra ffffffff80473d20 sp c000000000397830 sz 80
kern_readv+0x80 (?,?,?,?) ra ffffffff80152354 sp c000000000397880 sz 80
cheriabi_read+0x74 (?,?,?,?) ra ffffffff806b7d38 sp c0000000003978d0 sz 128
trap+0xf68 (?,?,?,?) ra ffffffff806a3608 sp c000000000397950 sz 272
MipsUserGenException+0x1ac (?,?,?,7c002a20d4) ra 0 sp c000000000397a60 sz 0

fgets returns 0x0, errno is 97
```

brooksdavis, 2018-08-03:
This is a gap in CheriABI.  The stack trace is because the system attempted to use an implicit capability (via bare `copyout`) in a CheriABI process.  The fix is to make `md_get_mem()`'s `target` argument be a `char * __capability` and then start pulling on the string of changes the fall out from there.

arichardson, 2018-08-04:
I am also seeing this when running the libcxx test suite:
```
# KDB: stack backtrace:
db_trace_self+0x18 (?,?,?,?) ra ffffffff801e7710 sp c0000000002ff000 sz 16
db_fetch_ksymtab+0x238 (?,?,?,?) ra ffffffff80443a3c sp c0000000002ff010 sz 800
kdb_backtrace+0x7c (?,?,?,?) ra ffffffff804680c4 sp c0000000002ff330 sz 16
copyout+0x104 (?,?,?,?) ra ffffffff80447a2c sp c0000000002ff340 sz 64
md_get_mem+0x15c (?,?,?,?) ra ffffffff804489f0 sp c0000000002ff380 sz 80
md_get_uio+0x120 (?,?,?,?) ra ffffffff805cdbe8 sp c0000000002ff3d0 sz 96
smb_read+0x450 (?,?,?,?) ra ffffffff802d541c sp c0000000002ff430 sz 192
smbfs_readvnode+0x36c (?,?,?,?) ra ffffffff802dd02c sp c0000000002ff4f0 sz 496
smbfs_free_scred+0x267c (?,?,?,?) ra ffffffff806ec254 sp c0000000002ff6e0 sz 16
VOP_READ_APV+0x10c (?,?,?,?) ra ffffffff80520dec sp c0000000002ff6f0 sz 48
vn_fsid+0x394 (?,?,?,?) ra ffffffff8051b7b0 sp c0000000002ff720 sz 128
sys_posix_fadvise+0x1e8 (?,?,?,?) ra ffffffff80473f58 sp c0000000002ff7a0 sz 144
sys_readv+0x140 (?,?,?,?) ra ffffffff80473a80 sp c0000000002ff830 sz 80
kern_readv+0x80 (?,?,?,?) ra ffffffff80152354 sp c0000000002ff880 sz 80
cheriabi_read+0x74 (?,?,?,?) ra ffffffff806b7298 sp c0000000002ff8d0 sz 128
trap+0xf68 (?,?,?,?) ra ffffffff806a2d88 sp c0000000002ff950 sz 272
MipsUserGenException+0x1ac (?,?,?,12019e3c4) ra 0 sp c0000000002ffa60 sz 0
pid 427 (member_swap.pass.cp), uid 0: exited on signal 6
KDB: stack backtrace:
db_trace_self+0x18 (?,?,?,?) ra ffffffff801e7710 sp c0000000002ff060 sz 16
db_fetch_ksymtab+0x238 (?,?,?,?) ra ffffffff80443a3c sp c0000000002ff070 sz 800
kdb_backtrace+0x7c (?,?,?,?) ra ffffffff80467e6c sp c0000000002ff390 sz 16
copyin+0x104 (?,?,?,?) ra ffffffff80446fb0 sp c0000000002ff3a0 sz 64
mb_put_mem+0x198 (?,?,?,?) ra ffffffff804474e8 sp c0000000002ff3e0 sz 112
mb_put_uio+0x120 (?,?,?,?) ra ffffffff805cdf7c sp c0000000002ff450 sz 96
smb_write+0x2f4 (?,?,?,?) ra ffffffff802d5a00 sp c0000000002ff4b0 sz 240
smbfs_writevnode+0x100 (?,?,?,?) ra ffffffff802dd07c sp c0000000002ff5a0 sz 80
smbfs_free_scred+0x26cc (?,?,?,?) ra ffffffff806ec4e4 sp c0000000002ff5f0 sz 16
VOP_WRITE_APV+0x19c (?,?,?,?) ra ffffffff805212a4 sp c0000000002ff600 sz 288
vn_fsid+0x84c (?,?,?,?) ra ffffffff8051b7b0 sp c0000000002ff720 sz 128
sys_posix_fadvise+0x1e8 (?,?,?,?) ra ffffffff80474734 sp c0000000002ff7a0 sz 144
sys_writev+0x164 (?,?,?,?) ra ffffffff80474238 sp c0000000002ff830 sz 80
kern_writev+0x80 (?,?,?,?) ra ffffffff80152484 sp c0000000002ff880 sz 80
cheriabi_write+0x74 (?,?,?,?) ra ffffffff806b7298 sp c0000000002ff8d0 sz 128
trap+0xf68 (?,?,?,?) ra ffffffff806a2d88 sp c0000000002ff950 sz 272
MipsUserGenException+0x1ac (?,?,?,12019d764) ra 0 sp c0000000002ffa60 sz 0
pid 587 (seekoff.pass.cpp.ex), uid 0: exited on signal 6
```
And then later on the kernel panics with the error from #273 . I doubt these are related but it is possible.

PR #282 (2018-08-21): Make smb use capabilities for IO.
This makes mount_smbfs work as a CheriABI binary at the cost of
breaking mips64 binaries.  I'll work 32- and 64-bit compat later.

Should fix #275 and half fixes #268.
closed 2018-08-21 by arichardson